phifogg/ioBroker.sainlogic #11: TCP Log

latzi0710, 2020-05-01:
Hi Daniel!

Anbei die Daten des traffic für die Wetterstation, mein Handy hat die IP 10.0.0.26, die Wetterstation 10.0.0.35. Man erkennt gut die Kommunikation mit meinem Raspi (10.0.0.210) und WU. Ich hab im WS View den custom port editiert (Leerzeichen dazu und wieder weg); also keine Änderung. Die angehängte Datei hab ich um ".txt" erweitert, damit der upload klappt.
Ich bin gespannt, ob du was rausfindest!

Vielen Dank für deine Mühe!

LG Latzi
[Wetterstation_Traffic.pcap.txt](https://github.com/phifogg/ioBroker.sainlogic/files/4564019/Wetterstation_Traffic.pcap.txt)

phifogg, 2020-05-01:
Kannst du mir den genauen Typ der Wetterstation noch hier eintragen? Und welche Werte gerade angezeigt werden in der App (Temperatur, Windstärke...). Irgendwas davon dass ich mal suchen kann.

phifogg, 2020-05-01:
Werte passt, sind ja die Weatherundergrund updates drin :D 
Irgendwie ist aber wenig Kommunikation mit dem Handy - zumindest geht wenig von der Wetterstation zum Handy.

phifogg, 2020-05-01:
Hehe... deine Wetterstation redet neben WU auch mit irgendjemand in China:
https://whatismyipaddress.com/ip/47.102.253.116
Das würde ich ihr mal verbieten.

latzi0710, 2020-05-01:
Das ist Ecowitt

phifogg, 2020-05-01:
Ich sehe einige Befehle die von Handy zu der WS gehen, aber leider keine Daten zurück kommen :/ Sorry.

phifogg, 2020-05-03:
Da die WS leider nicht auf die Scheduler Befehle reagiert kann ich da keine Daten auslesen. Sorry.

closed 2020-05-03 by phifogg