phifogg/ioBroker.sainlogic #43: Also support Foggit WH3000?

Apollon77, 2020-06-20:
My Weatherstation Foggit WH3000 lao has the port 45000 open and Sso I tried the adapter :-)

Result is that he can read out the software but is not reading more:-( Any chance to?

Debug Log:
```
2020-06-20 21:43:18.692  - info: sainlogic.0 (9208) starting. Version 0.6.2 in /opt/iobroker/node_modules/iobroker.sainlogic, node: v14.4.0, js-controller: 3.1.4
2020-06-20 21:43:18.708  - info: sainlogic.0 (9208) Starting Scheduler
2020-06-20 21:43:18.709  - info: sainlogic.0 (9208) Scheduler call ws_getfirmware activated
2020-06-20 21:43:18.709  - info: sainlogic.0 (9208) Scheduler call ws_getcurrent activated
2020-06-20 21:43:18.709  - info: sainlogic.0 (9208) Scheduler call ws_getmax activated
2020-06-20 21:43:18.709  - info: sainlogic.0 (9208) Scheduler call ws_getmin activated
2020-06-20 21:43:18.709  - info: sainlogic.0 (9208) Scheduler call ws_getmaxdaily activated
2020-06-20 21:43:18.710  - info: sainlogic.0 (9208) Scheduler call ws_getmindaily activated
2020-06-20 21:43:18.710  - debug: sainlogic.0 (9208) Weatherstation IP: 192.168.178.58
2020-06-20 21:43:18.710  - debug: sainlogic.0 (9208) Weatherstation port: 45000
2020-06-20 21:43:18.710  - debug: sainlogic.0 (9208) Scheduler Interval: 15

2020-06-20 21:43:33.711  - info: sainlogic.0 (9208) Scheduler pulling for new data
2020-06-20 21:43:33.736  - debug: sainlogic.0 (9208) Scheduler connected to weather station run 255,255,80,3,83
2020-06-20 21:43:33.743  - debug: sainlogic.0 (9208) FW Scheduler Received data string: ffff501511456173795765617468657256312e312e34
2020-06-20 21:43:33.743  - debug: sainlogic.0 (9208) Data Command received: 80 subcommand 69
2020-06-20 21:43:33.744  - debug: sainlogic.0 (9208) Data object: {"softwaretype":"EasyWeatherV1.1.4"}
2020-06-20 21:43:33.744  - info: sainlogic.0 (9208) Scheduler updating IOBroker states
2020-06-20 21:43:33.745  - debug: sainlogic.0 (9208) Setting value from data for info.softwaretype to EasyWeatherV1.1.4
2020-06-20 21:43:33.746  - debug: sainlogic.0 (9208) Scheduler connected to weather station run 255,255,11,0,6,4,4,25

2020-06-20 21:43:48.711  - info: sainlogic.0 (9208) Scheduler pulling for new data
2020-06-20 21:43:48.711  - debug: sainlogic.0 (9208) Scheduler connected to weather station run 255,255,80,3,83
2020-06-20 21:43:48.722  - debug: sainlogic.0 (9208) FW Scheduler Received data string: ffff501511456173795765617468657256312e312e34
2020-06-20 21:43:48.722  - debug: sainlogic.0 (9208) Data Command received: 80 subcommand 69
2020-06-20 21:43:48.722  - debug: sainlogic.0 (9208) Data object: {"softwaretype":"EasyWeatherV1.1.4"}
2020-06-20 21:43:48.722  - info: sainlogic.0 (9208) Scheduler updating IOBroker states
2020-06-20 21:43:48.723  - debug: sainlogic.0 (9208) Setting value from data for info.softwaretype to EasyWeatherV1.1.4
2020-06-20 21:43:48.723  - debug: sainlogic.0 (9208) Scheduler connected to weather station run 255,255,11,0,6,4,4,25

2020-06-20 21:44:03.711  - info: sainlogic.0 (9208) Scheduler pulling for new data
2020-06-20 21:44:03.712  - error: sainlogic.0 (9208) Connection error on 192.168.178.58:45000: Error: connect EISCONN 192.168.178.58:45000 - Local (192.168.178.79:46968)
2020-06-20 21:44:03.713  - debug: sainlogic.0 (9208) FW Scheduler Connection closed

2020-06-20 21:44:18.711  - info: sainlogic.0 (9208) Scheduler pulling for new data
2020-06-20 21:44:18.721  - debug: sainlogic.0 (9208) Scheduler connected to weather station run 255,255,80,3,83
2020-06-20 21:44:18.722  - debug: sainlogic.0 (9208) Scheduler connected to weather station run 255,255,80,3,83
2020-06-20 21:44:18.737  - debug: sainlogic.0 (9208) FW Scheduler Received data string: ffff501511456173795765617468657256312e312e34
2020-06-20 21:44:18.737  - debug: sainlogic.0 (9208) Data Command received: 80 subcommand 69
2020-06-20 21:44:18.737  - debug: sainlogic.0 (9208) Data object: {"softwaretype":"EasyWeatherV1.1.4"}
2020-06-20 21:44:18.737  - info: sainlogic.0 (9208) Scheduler updating IOBroker states
2020-06-20 21:44:18.738  - debug: sainlogic.0 (9208) Setting value from data for info.softwaretype to EasyWeatherV1.1.4
2020-06-20 21:44:18.738  - debug: sainlogic.0 (9208) Scheduler connected to weather station run 255,255,11,0,6,4,4,25
2020-06-20 21:44:18.790  - debug: sainlogic.0 (9208) FW Scheduler Received data string: ffff501511456173795765617468657256312e312e34
2020-06-20 21:44:18.790  - debug: sainlogic.0 (9208) Data Command received: 80 subcommand 69
2020-06-20 21:44:18.790  - debug: sainlogic.0 (9208) Data object: {"softwaretype":"EasyWeatherV1.1.4"}
2020-06-20 21:44:18.791  - info: sainlogic.0 (9208) Scheduler updating IOBroker states
2020-06-20 21:44:18.791  - debug: sainlogic.0 (9208) Setting value from data for info.softwaretype to EasyWeatherV1.1.4
2020-06-20 21:44:18.791  - debug: sainlogic.0 (9208) Scheduler connected to weather station run 255,255,11,0,6,5,5,27
2020-06-20 21:44:19.175  - debug: host.io-test.compactgroup1 object change system.adapter.sainlogic.0 (from: system.adapter.admin.0)
2020-06-20 21:44:19.175  - debug: host.io-test object change system.adapter.sainlogic.0 (from: system.adapter.admin.0)
```

BTW: When I see that log correct then something is strange because sometimes the connection not seems to be closed  (at least log missing).

phifogg, 2020-06-21:
This seems a bit weird and indeed a similar pattern as with the Eurochron EFWS2900. The get-firmware command returns just fine, other commands do not return anything, not even an error. Hence i keep the connection open waiting on an answer :/

If someone finds our which commands these stations need I can surely adjust this in the adapter. But at this point it is difficult for me.

Does the WeatherSmartIP app on windows for your station? I would expect this to have similar issues. Might also be that your firmware is just to old. Mine lists EasyWeatherV1.5.0. Can you check if an update is possible using WSView on your phone?

Apollon77, 2020-06-21:
wstool is finding the device but not offers any fimrware updates. it say firmware 1.1.4 is current one. wsTool is the "official" tool Frogitt tells about. It sensd my data to WU at the moment. configured via wsTool (and I also can add Weathercloud or weatherobservations

WsView (i tested now) and offers me to update firmware to 1.5.0 ... hm... :-) Should I do it? ;-)

Apollon77, 2020-06-21:
Ok, I do the update :-) fingers crossed :-)

Apollon77, 2020-06-21:
Hm no luck ... it says "Downloading ..." for 2-3 mins on the phone, and then "update fail" and thats all.
I also tried restarting the display ... no luck ... same again. I think wstool is doing something different ten WSView :-(

Apollon77, 2020-06-21:
Do you have a download URL for the weather station PC software

Apollon77, 2020-06-21:
I have written to Froggit and asked for an update ... lets see

phifogg, 2020-06-22:
The WeatherStation IP is available via ELV: https://files2.elv.com/public/25/2504/250408/Internet/ws980wifi_weathersmartip_v1_1_0426.zip

Let me know what Froggit's answer is.

Apollon77, 2020-06-22:
Ok, Software installed, find the device but also shows no values ...

phifogg, 2020-06-22:
Alright, kind of expected.
So let's wait for Froggit - otherwise it seems your station is more similar to the Eurocron and does not reply to the commands :/

Maybe it is different command strings that it requires. If you are curious and adventurious enough you can check the lib/constants.js, all commands are listed their in hex code. :D

Apollon77, 2020-07-03:
Very Interesting. No answer from them but today I just tried again and it downloaded and updated successfully via WSTool. Now it is listed there. Will try adapter later ;-) seems that the download of firmware did not worked.

Apollon77, 2020-07-03:
Ok, so it now has the 1.5.0FW but still:

```
2020-07-03 22:08:14.850  - debug: sainlogic.0 (3088) Weatherstation IP: 192.168.178.58
2020-07-03 22:08:14.850  - debug: sainlogic.0 (3088) Weatherstation port: 45000
2020-07-03 22:08:14.850  - debug: sainlogic.0 (3088) Scheduler Interval: 15
2020-07-03 22:08:28.895  - info: influxdb.0 (2108) Store 301 buffered influxDB history points
2020-07-03 22:08:29.851  - info: sainlogic.0 (3088) Scheduler pulling for new data
2020-07-03 22:08:29.927  - debug: sainlogic.0 (3088) Scheduler connected to weather station run 255,255,80,3,83
2020-07-03 22:08:29.932  - debug: sainlogic.0 (3088) FW Scheduler Received data string: ffff501511456173795765617468657256312e352e30
2020-07-03 22:08:29.933  - debug: sainlogic.0 (3088) Data Command received: 80 subcommand 69
2020-07-03 22:08:29.934  - debug: sainlogic.0 (3088) Data object: {"softwaretype":"EasyWeatherV1.5.0"}
2020-07-03 22:08:29.934  - info: sainlogic.0 (3088) Scheduler updating IOBroker states
2020-07-03 22:08:29.935  - debug: sainlogic.0 (3088) Setting value from data for info.softwaretype to EasyWeatherV1.5.0
2020-07-03 22:08:29.936  - debug: sainlogic.0 (3088) Scheduler connected to weather station run 255,255,11,0,6,4,4,25
2020-07-03 22:08:44.851  - info: sainlogic.0 (3088) Scheduler pulling for new data
2020-07-03 22:08:44.852  - debug: sainlogic.0 (3088) Scheduler connected to weather station run 255,255,80,3,83
2020-07-03 22:08:44.855  - debug: sainlogic.0 (3088) FW Scheduler Received data string: ffff501511456173795765617468657256312e352e30
2020-07-03 22:08:44.855  - debug: sainlogic.0 (3088) Data Command received: 80 subcommand 69
2020-07-03 22:08:44.855  - debug: sainlogic.0 (3088) Data object: {"softwaretype":"EasyWeatherV1.5.0"}
2020-07-03 22:08:44.855  - info: sainlogic.0 (3088) Scheduler updating IOBroker states
2020-07-03 22:08:44.856  - debug: sainlogic.0 (3088) Setting value from data for info.softwaretype to EasyWeatherV1.5.0
2020-07-03 22:08:44.856  - debug: sainlogic.0 (3088) Scheduler connected to weather station run 255,255,11,0,6,4,4,25
2020-07-03 22:08:59.852  - info: sainlogic.0 (3088) Scheduler pulling for new data
2020-07-03 22:08:59.853  - error: sainlogic.0 (3088) Connection error on 192.168.178.58:45000: Error: connect EISCONN 192.168.178.58:45000 - Local (192.168.178.79:37174)
2020-07-03 22:08:59.854  - debug: sainlogic.0 (3088) FW Scheduler Connection closed
```

BTW: EISCONN means that the socket is alkready connected so you might have cases where your code is not optimal! Comes only sometimes ... but still no answer :-(

Apollon77, 2020-07-03:
ANd the windows program also do not work ... so it seems to be something different

phifogg, 2020-07-04:
The connection issue is due to the fact the station does not send back a result. I should probably add a timeout as my code is waiting on the response.

Apollon77, 2020-07-04:
Timeout is a good idea ;-)

But in fact I now could after the fw Update think about using the „Webserver“ Mode. ;-)

phifogg, 2021-03-11:
Closing w/o update.
Station will only work in Listener mode.

closed 2021-03-11 by phifogg