phifogg/ioBroker.sainlogic #44: Installation funktioniert nicht

lhoem, 2020-07-03:
Hallo!
Ich habe gerade versucht die github version zu installieren. Keine Reaktion...
Hab dann den npm install über beliebige Quellen versucht. Wenigsten eine Reaktion mit abbruch. Das sagt das Log:

host.iobroker1 | 2020-07-03 16:11:56.651 | error | iobroker host.iobroker1 Cannot install https://www.npmjs.com/package/iobroker.sainlogic: 1
-- | -- | -- | --
host.iobroker1 | 2020-07-03 16:11:56.636 | info | iobroker npm ERR! A complete log of this run can be found in:npm ERR! /home/iobroker/.npm/_logs/2020-07-03T14_11_56_622Z-debug.log
host.iobroker1 | 2020-07-03 16:11:56.620 | info | iobroker npm ERR! package.json Non-registry package missing package.json: https://www.npmjs.com/package/iobroker.sainlogic.npm ERR! package.json npm can't find a package.json file in your current dir
host.iobroker1 | 2020-07-03 16:11:56.619 | info | iobroker npm ERR! code ENOPACKAGEJSON

node: v10.16.3, js-controller: 3.1.6

Was mache ich falsch?
Groß
Lothar

lhoem, 2020-07-03:
Das wird bei der Installation angezeigt:

$ ./iobroker url "https://github.com/phifogg/ioBroker.sainlogic/issues" --debug
install https://github.com/phifogg/ioBroker.sainlogic/issues/tarball/b35495235881dd622dc64d6a0aad4c7522d3ab1c
NPM version: 6.9.0
npm install https://github.com/phifogg/ioBroker.sainlogic/issues/tarball/b35495235881dd622dc64d6a0aad4c7522d3ab1c --loglevel error --prefix "/opt/iobroker" (System call)
npm 
ERR! code E404
npm 
ERR! 404 Not Found - GET https://github.com/phifogg/ioBroker.sainlogic/issues/tarball/b35495235881dd622dc64d6a0aad4c7522d3ab1cnpm ERR! 404 npm ERR! 404  'https://github.com/phifogg/ioBroker.sainlogic/issues/tarball/b35495235881dd622dc64d6a0aad4c7522d3ab1c' is not in the npm registry.npm ERR! 404 Your package name is not valid, because 
npm ERR! 404  1. name can only contain URL-friendly charactersnpm ERR! 404  2. name can no longer contain capital lettersnpm ERR! 404 npm
 ERR! 404 Note that you can also install from anpm ERR! 404 tarball, folder, http url, or git url.

npm ERR! A complete log of this run can be found in:npm ERR!     /home/iobroker/.npm/_logs/2020-07-03T18_42_18_989Z-debug.log
ERROR: host.iobroker1 Cannot install https://github.com/phifogg/ioBroker.sainlogic/issues/tarball/b35495235881dd622dc64d6a0aad4c7522d3ab1c: 1
ERROR: process exited with code 25

Apollon77, 2020-07-03:
Wrong path. Remove the /issues.  And why you install it that complicated way? Use admin

lhoem, 2020-07-03:
What do mean with use admin???

sainlogic is not listed in latest...

without /issues the  sameproblem...

this is the log:
host.iobroker1 | 2020-07-03 21:50:28.219 | info | iobroker exit 25
-- | -- | -- | --
host.iobroker1 | 2020-07-03 21:50:27.212 | error | iobroker host.iobroker1 Cannot install http://github.com/phifogg/ioBroker.sainlogic: 1
host.iobroker1 | 2020-07-03 21:50:27.189 | info | iobroker npm ERR! A complete log of this run can be found in:npm ERR! /home/iobroker/.npm/_logs/2020-07-03T19_50_27_173Z-debug.log
host.iobroker1 | 2020-07-03 21:50:27.172 | info | iobroker ERR! Error while executing:npm ERR! /usr/bin/git ls-remote -h -t ssh://git@github.com/phifogg/ioBroker.sainlogic.gitnpm ERR! npm ERR! Host key verification failed.npm ERR! fatal: Could not
host.iobroker1 | 2020-07-03 21:50:27.171 | info | iobroker npm
host.iobroker1 | 2020-07-03 21:50:22.110 | info | iobroker npm install http://github.com/phifogg/ioBroker.sainlogic --loglevel error --prefix "/opt/iobroker" (System call)
host.iobroker1 | 2020-07-03 21:50:22.109 | info | iobroker NPM version: 6.9.0
host.iobroker1 | 2020-07-03 21:50:21.867 | info | iobroker install http://github.com/phifogg/ioBroker.sainlogic
host.iobroker1 | 2020-07-03 21:50:20.776 | info | iobroker url "http://github.com/phifogg/ioBroker.sainlogic"

Apollon77, 2020-07-03:
Now you wrote http:// in the url which is also wrong. Needs to be https://

lhoem, 2020-07-04:
Wow... i think i'm confused... now it works... thanks!

Nice adapter 

Regards Lothar

phifogg, 2020-07-04:
Thanks @Apollon77, closing this issue as it works.

closed 2020-07-04 by phifogg