phifogg/ioBroker.sainlogic #47: Only create datapoints which are supported by the weather station

adx-git, 2020-08-05:
While using "ecowitt" as protocol for my weather station I recognized that some datapoints are created even if there is no string defined for this datapoint.
For example UVraw and dewpointtemp.
The fastest, to implement, solution would be to check if there is a string definition for the protocol and if not, do not create the datapoint.
Since some weather stations might support different datapoints, even if the protocol is the same, the better solution would be an automatic detection and giving the user the possibility to select the desired datapoints.
A mid term solution could be to pre-define datapoint for known weather stations and fall back to default if the station is not recognized.

phifogg, 2020-08-08:
I believe a first version was part of you pull request. I like the idea of marking the data points for validity of specific weather stations. need to think about it...

phifogg, 2022-01-20:
I blieve this is added quite some time by now. Closing issue.

closed 2022-01-20 by phifogg