phifogg/ioBroker.sainlogic #53: Warning after Upgrade to jscontroller 3.2.16

Boronsbruder, 2021-02-20:
Following warning is display in log:
`State "sainlogic.0.info.last_listener_update" has no existing object, this might lead to an error in future versions`

Adapterversion 0.6.4

phifogg, 2021-03-14:
Reason for this is in main.js around line 213:
```
            this.verify_datapoint(attr, this, my_attr_def[0], my_attr_def[0].channels[0].name); // allways channel 0 as primary attribute name

            this.setStateAsync(attr, { val: display_val, ack: true });

```

verify_datapoint will create the data point if needed, but this happens asynchronously and the datapoint might not be created fast enough before the first setState is called. Need to think how to rework this....

phifogg, 2021-03-14:
This should be solved now in 0.6.6 - can you confirm?

phifogg, 2021-03-17:
closed as confirmed in other thread.

closed 2021-03-17 by phifogg