phifogg/ioBroker.sainlogic #54: Adapter does not want to work with Sainlogic WS3500

FranzKillig, 2021-03-11:
Hi,

i installed the adapter to read the data from Sainlogic WS3500.  I followed the instructions for the configuration, but the instance cannot be started. The following log is output: 

host.raspberrypi | 2021-03-11 14:25:35.653 | info | Restart adapter system.adapter.sainlogic.0 because enabled
-- | -- | -- | --
host.raspberrypi | 2021-03-11 14:25:35.652 | error | instance system.adapter.sainlogic.0 terminated with code 6 (UNCAUGHT_EXCEPTION)
sainlogic.0 | 2021-03-11 14:25:35.099 | warn | (4457) Terminated (UNCAUGHT_EXCEPTION): Without reason
sainlogic.0 | 2021-03-11 14:25:35.097 | info | (4457) terminating
sainlogic.0 | 2021-03-11 14:25:35.034 | info | (4457) Starting Listener
sainlogic.0 | 2021-03-11 14:25:35.001 | info | (4457) starting. Version 0.6.4 in /opt/iobroker/node_modules/iobroker.sainlogic, node: v12.20.2, js-controller: 3.2.16
sainlogic.0 | 2021-03-11 14:25:34.595 | error | (4457) Cannot read property 'bind' of undefined
sainlogic.0 | 2021-03-11 14:25:34.594 | error | at internal/main/run_main_module.js:17:47
sainlogic.0 | 2021-03-11 14:25:34.594 | error | at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:60:12)
sainlogic.0 | 2021-03-11 14:25:34.594 | error | at Function.Module._load (internal/modules/cjs/loader.js:708:14)
sainlogic.0 | 2021-03-11 14:25:34.594 | error | at Module.load (internal/modules/cjs/loader.js:863:32)
sainlogic.0 | 2021-03-11 14:25:34.594 | error | at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
sainlogic.0 | 2021-03-11 14:25:34.594 | error | at Module._compile (internal/modules/cjs/loader.js:999:30)
sainlogic.0 | 2021-03-11 14:25:34.594 | error | at Object.<anonymous> (/opt/iobroker/node_modules/iobroker.sainlogic/main.js:281:5)
sainlogic.0 | 2021-03-11 14:25:34.594 | error | at new Sainlogic (/opt/iobroker/node_modules/iobroker.sainlogic/main.js:30:53)
sainlogic.0 | 2021-03-11 14:25:34.594 | error | (4457) TypeError: Cannot read property 'bind' of undefined
sainlogic.0 | 2021-03-11 14:25:34.591 | error | (4457) uncaught exception: Cannot read property 'bind' of undefined
host.raspberrypi | 2021-03-11 14:25:33.416 | info | instance system.adapter.sainlogic.0 started with pid 4457

My system:

raspberrypi
Platform linux
Betriebssystem linux
Architektur arm
CPUs 4
Geschwindigkeit 700 MHz
Modell ARMv7 Processor rev 3 (v7l)
RAM 7.69 GB
Node.js v12.20.2 (Es gibt eine neuere Version: v12.21.0)
NPM 6.14.11
Festplatte Größe 58.25 GB
Festplatte frei 54.28 GB
Anzahl der Adapter 334
Aktive Instanzen 10

Can someone help me find the cause? 
Thank you!

greeting
Franz

phifogg, 2021-03-11:
Hallo Franz,
das ist spannend, hatte ich noch nicht gesehen. Kannst Du den Adapter mal auf 'debug' stellen und schauen was dann ausgegeben wird?
Danke, 
Daniel

FranzKillig, 2021-03-11:
Hallo Daniel,

da hätte ich auch gleich drandenken können:

host.raspberrypi | 2021-03-11 18:53:54.694 | info | Restart adapter system.adapter.sainlogic.0 because enabled
-- | -- | -- | --
host.raspberrypi | 2021-03-11 18:53:54.692 | error | instance system.adapter.sainlogic.0 terminated with code 6 (UNCAUGHT_EXCEPTION)
sainlogic.0 | 2021-03-11 18:53:54.105 | warn | (4793) Terminated (UNCAUGHT_EXCEPTION): Without reason
sainlogic.0 | 2021-03-11 18:53:54.104 | info | (4793) terminating
sainlogic.0 | 2021-03-11 18:53:53.959 | info | (4793) Starting Listener
sainlogic.0 | 2021-03-11 18:53:53.958 | debug | (4793) Forward URL:
sainlogic.0 | 2021-03-11 18:53:53.957 | debug | (4793) Listener protocol: wunderground
sainlogic.0 | 2021-03-11 18:53:53.957 | debug | (4793) Listener path: /weatherstation/updateweatherstation.php
sainlogic.0 | 2021-03-11 18:53:53.957 | debug | (4793) Listener port: 45000
sainlogic.0 | 2021-03-11 18:53:53.956 | debug | (4793) Listener IP: 192.168.153.111
sainlogic.0 | 2021-03-11 18:53:53.923 | info | (4793) starting. Version 0.6.4 in /opt/iobroker/node_modules/iobroker.sainlogic, node: v12.20.2, js-controller: 3.2.16
host.raspberrypi | 2021-03-11 18:53:52.417 | info | instance system.adapter.sainlogic.0 started with pid 4793
host.raspberrypi | 2021-03-11 18:53:22.381 | info | Restart adapter system.adapter.sainlogic.0 because enabled
host.raspberrypi | 2021-03-11 18:53:22.379 | error | instance system.adapter.sainlogic.0 terminated with code 6 (UNCAUGHT_EXCEPTION)
sainlogic.0 | 2021-03-11 18:53:21.779 | warn | (4778) Terminated (UNCAUGHT_EXCEPTION): Without reason
sainlogic.0 | 2021-03-11 18:53:21.778 | info | (4778) terminating
sainlogic.0 | 2021-03-11 18:53:21.658 | info | (4778) Starting Listener
sainlogic.0 | 2021-03-11 18:53:21.657 | debug | (4778) Forward URL:
sainlogic.0 | 2021-03-11 18:53:21.657 | debug | (4778) Listener protocol: wunderground
sainlogic.0 | 2021-03-11 18:53:21.656 | debug | (4778) Listener path: /weatherstation/updateweatherstation.php
sainlogic.0 | 2021-03-11 18:53:21.656 | debug | (4778) Listener port: 45000
sainlogic.0 | 2021-03-11 18:53:21.655 | debug | (4778) Listener IP: 192.168.153.111
sainlogic.0 | 2021-03-11 18:53:21.623 | info | (4778) starting. Version 0.6.4 in /opt/iobroker/node_modules/iobroker.sainlogic, node: v12.20.2, js-controller: 3.2.16
sainlogic.0 | 2021-03-11 18:53:21.382 | debug | (4778) statesDB connected
sainlogic.0 | 2021-03-11 18:53:21.381 | debug | (4778) States connected to redis: 127.0.0.1:9000
sainlogic.0 | 2021-03-11 18:53:21.373 | debug | (4778) States create User PubSub Client
sainlogic.0 | 2021-03-11 18:53:21.371 | debug | (4778) States create System PubSub Client
sainlogic.0 | 2021-03-11 18:53:21.363 | debug | (4778) Redis States: Use Redis connection: 127.0.0.1:9000
sainlogic.0 | 2021-03-11 18:53:21.360 | debug | (4778) objectDB connected
sainlogic.0 | 2021-03-11 18:53:21.351 | debug | (4778) Objects connected to redis: 127.0.0.1:9001
sainlogic.0 | 2021-03-11 18:53:21.336 | debug | (4778) Objects client initialize lua scripts
sainlogic.0 | 2021-03-11 18:53:21.335 | debug | (4778) Objects create User PubSub Client
sainlogic.0 | 2021-03-11 18:53:21.334 | debug | (4778) Objects create System PubSub Client
sainlogic.0 | 2021-03-11 18:53:21.323 | debug | (4778) Objects client ready ... initialize now
sainlogic.0 | 2021-03-11 18:53:21.276 | error | (4778) Cannot read property 'bind' of undefined
sainlogic.0 | 2021-03-11 18:53:21.275 | error | at internal/main/run_main_module.js:17:47
sainlogic.0 | 2021-03-11 18:53:21.275 | error | at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:60:12)
sainlogic.0 | 2021-03-11 18:53:21.275 | error | at Function.Module._load (internal/modules/cjs/loader.js:708:14)
sainlogic.0 | 2021-03-11 18:53:21.275 | error | at Module.load (internal/modules/cjs/loader.js:863:32)
sainlogic.0 | 2021-03-11 18:53:21.275 | error | at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
sainlogic.0 | 2021-03-11 18:53:21.275 | error | at Module._compile (internal/modules/cjs/loader.js:999:30)
sainlogic.0 | 2021-03-11 18:53:21.275 | error | at Object.<anonymous> (/opt/iobroker/node_modules/iobroker.sainlogic/main.js:281:5)
sainlogic.0 | 2021-03-11 18:53:21.275 | error | at new Sainlogic (/opt/iobroker/node_modules/iobroker.sainlogic/main.js:30:53)
sainlogic.0 | 2021-03-11 18:53:21.275 | error | (4778) TypeError: Cannot read property 'bind' of undefined
sainlogic.0 | 2021-03-11 18:53:21.274 | error | (4778) uncaught exception: Cannot read property 'bind' of undefined
sainlogic.0 | 2021-03-11 18:53:21.262 | debug | (4778) Redis Objects: Use Redis connection: 127.0.0.1:9001
host.raspberrypi | 2021-03-11 18:53:20.094 | info | instance system.adapter.sainlogic.0 started with pid 4778
sainlogic.0 | 2021-03-11 18:53:21.779 | warn | (4778) Terminated (UNCAUGHT_EXCEPTION): Without reason
sainlogic.0 | 2021-03-11 18:53:21.778 | info | (4778) terminating

Nachgefragt, 2021-03-11:
@Froog
ELV 980 WIFI gleiches Problem, würde mich echt freunden wenn dus lösen könntest, hab mir die Station deinem Adapter gekauft weil ich dachte dann bekomme ich das als Laie auch hin, scheitere aber seit 2 Tagen. 
Gruß
Norbert

Nachgefragt, 2021-03-11:
sainlogic.0 | 2021-03-11 22:12:26.470 | warn | (19113) Terminated (UNCAUGHT_EXCEPTION): Without reason
-- | -- | -- | --
sainlogic.0 | 2021-03-11 22:12:26.469 | info | (19113) terminating
sainlogic.0 | 2021-03-11 22:12:26.269 | info | (19113) Starting Listener
sainlogic.0 | 2021-03-11 22:12:26.266 | info | (19113) Scheduler call ws_getmindaily activated
sainlogic.0 | 2021-03-11 22:12:26.266 | info | (19113) Scheduler call ws_getmaxdaily activated
sainlogic.0 | 2021-03-11 22:12:26.266 | info | (19113) Scheduler call ws_getmin activated
sainlogic.0 | 2021-03-11 22:12:26.265 | info | (19113) Scheduler call ws_getmax activated
sainlogic.0 | 2021-03-11 22:12:26.265 | info | (19113) Scheduler call ws_getcurrent activated
sainlogic.0 | 2021-03-11 22:12:26.264 | info | (19113) Scheduler call ws_getfirmware activated
sainlogic.0 | 2021-03-11 22:12:26.263 | info | (19113) Starting Scheduler
sainlogic.0 | 2021-03-11 22:12:26.230 | info | (19113) starting. Version 0.6.4 in /opt/iobroker/node_modules/iobroker.sainlogic, node: v12.21.0, js-controller: 3.2.16
sainlogic.0 | 2021-03-11 22:12:25.967 | error | (19113) Cannot read property 'bind' of undefined
sainlogic.0 | 2021-03-11 22:12:25.966 | error | at internal/main/run_main_module.js:17:47
sainlogic.0 | 2021-03-11 22:12:25.966 | error | at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:60:12)
sainlogic.0 | 2021-03-11 22:12:25.966 | error | at Function.Module._load (internal/modules/cjs/loader.js:708:14)
sainlogic.0 | 2021-03-11 22:12:25.966 | error | at Module.load (internal/modules/cjs/loader.js:863:32)
sainlogic.0 | 2021-03-11 22:12:25.966 | error | at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
sainlogic.0 | 2021-03-11 22:12:25.966 | error | at Module._compile (internal/modules/cjs/loader.js:999:30)
sainlogic.0 | 2021-03-11 22:12:25.966 | error | at Object.<anonymous> (/opt/iobroker/node_modules/iobroker.sainlogic/main.js:281:5)
sainlogic.0 | 2021-03-11 22:12:25.966 | error | at new Sainlogic (/opt/iobroker/node_modules/iobroker.sainlogic/main.js:30:53)
sainlogic.0 | 2021-03-11 22:12:25.966 | error | (19113) TypeError: Cannot read property 'bind' of undefined
sainlogic.0 | 2021-03-11 22:12:25.963 | error | (19113) uncaught exception: Cannot read property 'bind' of undefined
sainlogic.0 | 2021-03-11 22:12:21.577 | warn | (19098) Terminated (UNCAUGHT_EXCEPTION): Without reason
sainlogic.0 | 2021-03-11 22:12:21.576 | info | (19098) terminating
sainlogic.0 | 2021-03-11 22:12:21.411 | info | (19098) Starting Listener
sainlogic.0 | 2021-03-11 22:12:21.408 | info | (19098) Scheduler call ws_getmindaily activated
sainlogic.0 | 2021-03-11 22:12:21.408 | info | (19098) Scheduler call ws_getmaxdaily activated
sainlogic.0 | 2021-03-11 22:12:21.407 | info | (19098) Scheduler call ws_getmin activated
sainlogic.0 | 2021-03-11 22:12:21.407 | info | (19098) Scheduler call ws_getmax activated
sainlogic.0 | 2021-03-11 22:12:21.406 | info | (19098) Scheduler call ws_getcurrent activated
sainlogic.0 | 2021-03-11 22:12:21.406 | info | (19098) Scheduler call ws_getfirmware activated
sainlogic.0 | 2021-03-11 22:12:21.405 | info | (19098) Starting Scheduler
sainlogic.0 | 2021-03-11 22:12:21.371 | info | (19098) starting. Version 0.6.4 in /opt/iobroker/node_modules/iobroker.sainlogic, node: v12.21.0, js-controller: 3.2.16
sainlogic.0 | 2021-03-11 22:12:21.074 | error | (19098) Cannot read property 'bind' of undefined
sainlogic.0 | 2021-03-11 22:12:21.073 | error | at internal/main/run_main_module.js:17:47
sainlogic.0 | 2021-03-11 22:12:21.073 | error | at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:60:12)
sainlogic.0 | 2021-03-11 22:12:21.073 | error | at Function.Module._load (internal/modules/cjs/loader.js:708:14)
sainlogic.0 | 2021-03-11 22:12:21.073 | error | at Module.load (internal/modules/cjs/loader.js:863:32)
sainlogic.0 | 2021-03-11 22:12:21.073 | error | at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
sainlogic.0 | 2021-03-11 22:12:21.073 | error | at Module._compile (internal/modules/cjs/loader.js:999:30)
sainlogic.0 | 2021-03-11 22:12:21.073 | error | at Object.<anonymous> (/opt/iobroker/node_modules/iobroker.sainlogic/main.js:281:5)
sainlogic.0 | 2021-03-11 22:12:21.073 | error | at new Sainlogic (/opt/iobroker/node_modules/iobroker.sainlogic/main.js:30:53)
sainlogic.0 | 2021-03-11 22:12:21.073 | error | (19098) TypeError: Cannot read property 'bind' of undefined
sainlogic.0 | 2021-03-11 22:12:21.070 | error | (19098) uncaught exception: Cannot read property 'bind' of undefined
sainlogic.0 | 2021-03-11 22:12:03.744 | warn | (19083) Terminated (UNCAUGHT_EXCEPTION): Without reason
sainlogic.0 | 2021-03-11 22:12:03.743 | info | (19083) terminating
sainlogic.0 | 2021-03-11 22:12:03.583 | info | (19083) Starting Listener
sainlogic.0 | 2021-03-11 22:12:03.580 | info | (19083) Scheduler call ws_getcurrent activated
sainlogic.0 | 2021-03-11 22:12:03.579 | info | (19083) Scheduler call ws_getfirmware activated
sainlogic.0 | 2021-03-11 22:12:03.578 | info | (19083) Starting Scheduler
sainlogic.0 | 2021-03-11 22:12:03.544 | info | (19083) starting. Version 0.6.4 in /opt/iobroker/node_modules/iobroker.sainlogic, node: v12.21.0, js-controller: 3.2.16
sainlogic.0 | 2021-03-11 22:12:03.241 | error | (19083) Cannot read property 'bind' of undefined
sainlogic.0 | 2021-03-11 22:12:03.240 | error | at internal/main/run_main_module.js:17:47
[... cut ...]
sainlogic.0 | 2021-03-11 21:43:18.542 | error | at Object.<anonymous> (/opt/iobroker/node_modules/iobroker.sainlogic/main.js:281:5)
sainlogic.0 | 2021-03-11 21:43:18.542 | error | at new Sainlogic (/opt/iobroker/node_modules/iobroker.sainlogic/main.js:30:53)
sainlogic.0 | 2021-03-11 21:43:18.542 | error | (18917) TypeError: Cannot read property 'bind' of undefined
sainlogic.0 | 2021-03-11 21:43:18.540 | error | (18917) uncaught exception: Cannot read property 'bind' of undefined
sainlogic.0 | 2021-03-11 21:42:46.746 | warn | (18901) Terminated (UNCAUGHT_EXCEPTION): Without reason

Nachgefragt, 2021-03-11:
Node.js
v12.21.0
NPM
7.6.1
js-controller
Version
3.2.16

I3imbel, 2021-03-12:
Hi,
mein Log sieht genauso aus...

Node.js
v12.16.1
NPM
6.13.4
js-controller 
3.2.16

phifogg, 2021-03-13:
Ich kann den Fehler hier nachvollziehen. Schaue mir das an.

phifogg, 2021-03-13:
Ich glaub ich hab die Ursache gefunden. Probiert nochmal eine neue Installation, sollte nun 0.6.5 sein und wieder gehen.

Nachgefragt, 2021-03-13:
Hallo Daniel, vielen Dank hat geklappt.

ainlogic.0 | 2021-03-13 13:27:17.066 | info | (4699) Unit changed for sainlogic.0.weather.maxvalues.daily.solarradiation from lx to W/m2, updating data point
-- | -- | -- | --
sainlogic.0 | 2021-03-13 13:27:17.056 | info | (4699) Scheduler updating IOBroker states
sainlogic.0 | 2021-03-13 13:27:17.036 | info | (4699) Scheduler updating IOBroker states
sainlogic.0 | 2021-03-13 13:27:17.024 | info | (4699) Unit changed for sainlogic.0.weather.maxvalues.absolut.solarradiation from lx to W/m2, updating data point
sainlogic.0 | 2021-03-13 13:27:17.009 | info | (4699) Scheduler updating IOBroker states
sainlogic.0 | 2021-03-13 13:27:16.979 | info | (4699) Scheduler updating IOBroker states
sainlogic.0 | 2021-03-13 13:27:16.975 | warn | (4699) State "sainlogic.0.weather.current.windheading" has no existing object, this might lead to an error in future versions
sainlogic.0 | 2021-03-13 13:27:16.970 | info | (4699) Unit changed for sainlogic.0.weather.current.solarradiation from lx to W/m2, updating data point
sainlogic.0 | 2021-03-13 13:27:16.947 | info | (4699) Scheduler updating IOBroker states
sainlogic.0 | 2021-03-13 13:27:16.932 | info | (4699) Scheduler updating IOBroker states
sainlogic.0 | 2021-03-13 13:27:16.915 | info | (4699) Scheduler pulling for new data
sainlogic.0 | 2021-03-13 13:27:02.097 | info | (4699) Unit changed for sainlogic.0.weather.maxvalues.daily.solarradiation from lx to W/m2, updating data point
sainlogic.0 | 2021-03-13 13:27:02.077 | info | (4699) Scheduler updating IOBroker states
sainlogic.0 | 2021-03-13 13:27:02.048 | info | (4699) Scheduler updating IOBroker states
sainlogic.0 | 2021-03-13 13:27:02.025 | info | (4699) Scheduler updating IOBroker states
sainlogic.0 | 2021-03-13 13:27:02.015 | info | (4699) Unit changed for sainlogic.0.weather.maxvalues.absolut.solarradiation from lx to W/m2, updating data point
sainlogic.0 | 2021-03-13 13:27:01.987 | info | (4699) Scheduler updating IOBroker states
sainlogic.0 | 2021-03-13 13:27:01.982 | warn | (4699) State "sainlogic.0.weather.current.windheading" has no existing object, this might lead to an error in future versions
sainlogic.0 | 2021-03-13 13:27:01.976 | info | (4699) Unit changed for sainlogic.0.weather.current.solarradiation from lx to W/m2, updating data point
sainlogic.0 | 2021-03-13 13:27:01.950 | info | (4699) Scheduler updating IOBroker states
sainlogic.0 | 2021-03-13 13:27:01.927 | info | (4699) Scheduler updating IOBroker states
sainlogic.0 | 2021-03-13 13:27:01.914 | info | (4699) Scheduler pulling for new data
sainlogic.0 | 2021-03-13 13:26:47.083 | info | (4699) Unit changed for sainlogic.0.weather.maxvalues.daily.solarradiation from lx to W/m2, updating data point
sainlogic.0 | 2021-03-13 13:26:47.076 | info | (4699) Scheduler updating IOBroker states
sainlogic.0 | 2021-03-13 13:26:47.047 | info | (4699) Scheduler updating IOBroker states
sainlogic.0 | 2021-03-13 13:26:47.035 | info | (4699) Unit changed for sainlogic.0.weather.maxvalues.absolut.solarradiation from lx to W/m2, updating data point
sainlogic.0 | 2021-03-13 13:26:47.006 | info | (4699) Scheduler updating IOBroker states
sainlogic.0 | 2021-03-13 13:26:46.977 | info | (4699) Scheduler updating IOBroker states

Nachgefragt, 2021-03-13:
Kannst du evtl. sagen woher der Fehler kommt? (4699) State "sainlogic.0.weather.current.windheading" has no existing object, this might lead to an error in future versions

phifogg, 2021-03-13:
Mach doch bitte ein neues Issue auf dafür, dann schaue ich danach. Habe zumindest schon eine Idee. Das Problem hier haben wir ja gelöst.

closed 2021-03-13 by phifogg