phifogg/ioBroker.sainlogic #56: has no existing object, this might lead to an error in future versions

Nachgefragt, 2021-03-13:
Hallo Daniel, kannst du evtl. sagen woher der Fehler kommt?

(4699) State "sainlogic.0.weather.current.windheading" has no existing object, this might lead to an error in future versions
(4699) Scheduler received data for unkown command 11 subcommand 1

phifogg, 2021-03-14:
Die erste Ausgabe 'has no existing object' kommt von #53.

Die zweite ist etwas spannender. Hier kommt bei mir was an was ich nicht kenne. Welche Wetterstation ist das? Kannst Du mal ein Log auf Debug schicken? Dann sehe ich zumindest was ich dabei angefragt hab.

Nachgefragt, 2021-03-14:
Hallo Daniel, danke erstmal für deine schnelle Umsetzungen. Den neuen Adapter habe ich installiert, läuft gut, die Fehlermeldung mit command 11 subcommand 1 ist nun auch weg. Die Wetterstation ist von ELV 980 WIFI, hab ich mir gekauft weil du in deinem Adapter geschrieben hattest das sie funktioniert. Stimmt ja auch und ich bin voll zufrieden. Anbei noch das Debug Log nach dem Adapter Update:

sainlogic.0 | 2021-03-14 14:13:16.257 | info | (20847) Scheduler updating IOBroker states
-- | -- | -- | --
sainlogic.0 | 2021-03-14 14:13:16.250 | info | (20847) Unit changed for sainlogic.0.weather.maxvalues.daily.solarradiation from lx to W/m2, updating data point
sainlogic.0 | 2021-03-14 14:13:16.230 | info | (20847) Scheduler updating IOBroker states
sainlogic.0 | 2021-03-14 14:13:16.212 | info | (20847) Scheduler updating IOBroker states
sainlogic.0 | 2021-03-14 14:13:16.210 | info | (20847) Unit changed for sainlogic.0.weather.maxvalues.absolut.solarradiation from lx to W/m2, updating data point
sainlogic.0 | 2021-03-14 14:13:16.192 | info | (20847) Scheduler updating IOBroker states
sainlogic.0 | 2021-03-14 14:13:16.189 | info | (20847) Unit changed for sainlogic.0.weather.current.solarradiation from lx to W/m2, updating data point
sainlogic.0 | 2021-03-14 14:13:16.174 | info | (20847) Scheduler updating IOBroker states
sainlogic.0 | 2021-03-14 14:13:16.156 | info | (20847) Scheduler updating IOBroker states
sainlogic.0 | 2021-03-14 14:13:16.120 | info | (20847) Scheduler pulling for new data
sainlogic.0 | 2021-03-14 14:13:01.393 | info | (20847) Unit changed for sainlogic.0.weather.maxvalues.daily.solarradiation from lx to W/m2, updating data point
sainlogic.0 | 2021-03-14 14:13:01.375 | info | (20847) Scheduler updating IOBroker states
sainlogic.0 | 2021-03-14 14:13:01.332 | info | (20847) Scheduler updating IOBroker states
sainlogic.0 | 2021-03-14 14:13:01.325 | info | (20847) Unit changed for sainlogic.0.weather.maxvalues.absolut.solarradiation from lx to W/m2, updating data point
sainlogic.0 | 2021-03-14 14:13:01.298 | info | (20847) Scheduler updating IOBroker states
sainlogic.0 | 2021-03-14 14:13:01.253 | info | (20847) Scheduler updating IOBroker states
sainlogic.0 | 2021-03-14 14:13:01.248 | info | (20847) Unit changed for sainlogic.0.weather.current.solarradiation from lx to W/m2, updating data point
sainlogic.0 | 2021-03-14 14:13:01.222 | info | (20847) Scheduler updating IOBroker states
sainlogic.0 | 2021-03-14 14:13:01.186 | info | (20847) Scheduler updating IOBroker states
sainlogic.0 | 2021-03-14 14:13:01.120 | info | (20847) Scheduler pulling for new data
sainlogic.0 | 2021-03-14 14:12:46.377 | info | (20847) Unit changed for sainlogic.0.weather.maxvalues.daily.solarradiation from lx to W/m2, updating data point
sainlogic.0 | 2021-03-14 14:12:46.371 | info | (20847) Scheduler updating IOBroker states
sainlogic.0 | 2021-03-14 14:12:46.345 | info | (20847) Scheduler updating IOBroker states
sainlogic.0 | 2021-03-14 14:12:46.340 | info | (20847) Unit changed for sainlogic.0.weather.maxvalues.absolut.solarradiation from lx to W/m2, updating data point
sainlogic.0 | 2021-03-14 14:12:46.313 | info | (20847) Scheduler updating IOBroker states
sainlogic.0 | 2021-03-14 14:12:46.299 | info | (20847) Unit changed for sainlogic.0.weather.current.solarradiation from lx to W/m2, updating data point
sainlogic.0 | 2021-03-14 14:12:46.283 | info | (20847) Scheduler updating IOBroker states
sainlogic.0 | 2021-03-14 14:12:46.248 | info | (20847) Scheduler updating IOBroker states
sainlogic.0 | 2021-03-14 14:12:46.230 | info | (20847) Scheduler updating IOBroker states
sainlogic.0 | 2021-03-14 14:12:46.120 | info | (20847) Scheduler pulling for new data

Denke du kannst den Issue schliessen und als mit dem Update als erledigt setzen. Danke für die Hilfe.
Gruß
Norbert

phifogg, 2021-03-14:
Ah.. super. dann mach ich hier zu.

Ja, die 980 Wifi muss funktionieren, die hab ich ja auch auf dem Dach :D

closed 2021-03-14 by phifogg