phifogg/ioBroker.sainlogic #58: Reset stale values (daily, rain, ...)

phifogg, 2021-03-14:
Some values make no sense if they get not updated regularly. Like rain, dailyrain, solarradiation, ... 

Add feature to automatically reset them to 0 if not updated after X rounds of updates.

phifogg, 2021-03-14:
https://forum.iobroker.net/topic/33611/test-adapter-iobroker-sainlogic-v0-6-x-github-latest/127?_=1615717499254

phifogg, 2021-03-14:
[image]

Nachgefragt, 2021-03-14:
Hallo Daniel, sehe gerade die Funktion verstehe sie aber nicht ganz. Erstmal was bedeutet daily, weekly, monthly ich dachte da kumuliert die Wetterstation selber die erfassten Werte und berechnet die Tages, Wochen, Monats und Jahresdurchschnittswerte. Somit müsste ich die nicht löschen sondern können bleiben. Die aktuelle Regenrate ok, die könnten nach einigen Updates tatsächlich gelöscht und neu erfasst werden um zu messen ob es wieder regnet. So verstehe ich die Funktion aber nicht. Außerdem werden die Zeitstempel der jeweiligen Einzelwerte abgefragt die, zumindest bei mir garnicht im Adapter vorhanden sind. Verstehe ich hier was komplett falsch?
Gruß
Norbert

phifogg, 2021-03-14:
Korrekt, die aktuellen Datenpunkte daily, weekly und monthly kommen so direkt aus der Station - zumindest bei mir. Von daher muss ich da eigentlich auch nichts machen. Es gibt wohl Stationen die die Werte wenn sie auf '0' sind nicht schicken. Dann kriegt der Adapter kein Update und kann folglich auch keine '0' notieren. Für diese bräuchte es die Reset-Funktion.

phifogg, 2025-01-21:
Ich mach mal zu da keine weiteren Aktivitäten. Wenn hier noch was zu tun ist bitte ein neues Issue aufmachen.

closed 2025-01-21 by phifogg