phifogg/ioBroker.sainlogic #61: Adapter (Version 0.6.6) läuft bei mir jetzt auch, aber es kommen keine Werte an.

FranzKillig, 2021-03-14:
Hallo,

der Adapter starte nun ohne Fehlermeldung, jedoch kommen auch nach längerem warten keine Fehler an:
Ich kann aber auf dem Raspi das Verzeichnis /weatherstation/updateweatherstation.php nicht finden. Muss dies von Hand angelegt werden bzw. sollte diese im Stammverzeichnis oder unter /iobroker.sainlogic/ sein?

  |   |   |  
-- | -- | -- | --
sainlogic.0 | 2021-03-14 17:42:01.449 | info | (11541) Starting Listener
sainlogic.0 | 2021-03-14 17:42:01.448 | debug | (11541) Forward URL:
sainlogic.0 | 2021-03-14 17:42:01.448 | debug | (11541) Listener protocol: wunderground
sainlogic.0 | 2021-03-14 17:42:01.448 | debug | (11541) Listener path: /weatherstation/updateweatherstation.php
sainlogic.0 | 2021-03-14 17:42:01.447 | debug | (11541) Listener port: 45000
sainlogic.0 | 2021-03-14 17:42:01.446 | debug | (11541) Listener IP: 0.0.0.0
sainlogic.0 | 2021-03-14 17:42:01.413 | info | (11541) starting. Version 0.6.6 in /opt/iobroker/node_modules/iobroker.sainlogic, node: v12.20.2, js-controller: 3.2.16
sainlogic.0 | 2021-03-14 17:42:01.103 | debug | (11541) statesDB connected
sainlogic.0 | 2021-03-14 17:42:01.102 | debug | (11541) States connected to redis: 127.0.0.1:9000
sainlogic.0 | 2021-03-14 17:42:01.093 | debug | (11541) States create User PubSub Client
sainlogic.0 | 2021-03-14 17:42:01.092 | debug | (11541) States create System PubSub Client
sainlogic.0 | 2021-03-14 17:42:01.083 | debug | (11541) Redis States: Use Redis connection: 127.0.0.1:9000
sainlogic.0 | 2021-03-14 17:42:01.080 | debug | (11541) objectDB connected
sainlogic.0 | 2021-03-14 17:42:01.071 | debug | (11541) Objects connected to redis: 127.0.0.1:9001
sainlogic.0 | 2021-03-14 17:42:01.047 | debug | (11541) Objects client initialize lua scripts
sainlogic.0 | 2021-03-14 17:42:01.045 | debug | (11541) Objects create User PubSub Client
sainlogic.0 | 2021-03-14 17:42:01.044 | debug | (11541) Objects create System PubSub Client
sainlogic.0 | 2021-03-14 17:42:01.041 | debug | (11541) Objects client ready ... initialize now
sainlogic.0 | 2021-03-14 17:42:00.998 | debug | (11541) Redis Objects: Use Redis connection: 127.0.0.1:9001
host.raspberrypi | 2021-03-14 17:41:59.781 | info | instance system.adapter.sainlogic.0 started with pid 11541
sainlogic.0 | 2021-03-14 17:42:01.449 | info | (11541) Starting Listener
sainlogic.0 | 2021-03-14 17:42:01.448 | debug | (11541) Forward URL:
sainlogic.0 | 2021-03-14 17:42:01.448 | debug | (11541) Listener protocol: wunderground
sainlogic.0 | 2021-03-14 17:42:01.448 | debug | (11541) Listener path: /weatherstation/updateweatherstation.php
sainlogic.0 | 2021-03-14 17:42:01.447 | debug | (11541) Listener port: 45000
sainlogic.0 | 2021-03-14 17:42:01.446 | debug | (11541) Listener IP: 0.0.0.0
sainlogic.0 | 2021-03-14 17:42:01.413 | info | (11541) starting. Version 0.6.6 in /opt/iobroker/node_modules/iobroker.sainlogic, node: v12.20.2, js-controller: 3.2.16

Gruß
Franz

phifogg, 2021-03-14:
Hallo Franz,

nein, das ' /weatherstation/updateweatherstation.php' braucht es nicht als Verzeichnis. Dass ist nur die URL wo Deine Wetterstation die Daten anliefert sofern sie das kann.

Wenn der Listener läuft verhält sich der Adapeter quasi wie ein Wunderground Server, du musst dann Deiner Wetterstation sagen sie muss als Ziel den ioBroker host benutzen. Gleiche Einstellungen wie Du beim Listener machts (IP, Pfad und Port).

Gruss,
Daniel

FranzKillig, 2021-03-16:
Hallo Daniel,

ich kann machen was ich will. Im ioBroker kommen einfach keine Daten an:

[image]

[image]

[image]

Noch eine Idee? Ach so Originalname des Path und die 45000 als Port wurden natürlich auch probiert.
Könnte es an den Benutzerrechten auf dem Raspi liegen?

Gruß
Franz

phifogg, 2021-03-16:
Hi Franz,
spannend, die Einstellungen sehen gut aus. Sind denn die Wetterstation und der Raspi um gleichen Netzwerk? Sprich können die miteinander reden?
Was für eine Station ist es (sorry wenn ich das schonmal gefragt hab... sind gerade ein paar Chats...), wenn es die ELV 980 Wifi ist kannst Du auch mal den Scheduler probieren, der holt die Daten ab.
Gruss,
Daniel

FranzKillig, 2021-03-16:
Hallo Daniel, ja spannend...............

Die Sainlogic WS3500 ist im gleichen Netzwerk. Die kann halt nur Listener.

phifogg, 2021-03-16:
Ich hab bei mir im Listener als Ip einfach 'All - 0.0.0.0' benutzt. Nehme an das hast Du auch schon probiert? Kannst Du ggf. mit einem anderen Toll wie Wireshark oder so schauen ob die Wetterstation daten senden? Wenn Du mit Linux fit bist kannst auch Netcat (NC) dafür nutzen. Das habe ich bei der Entwicklung benutzt.

FranzKillig, 2021-03-17:
All - 0.0.0.0 habe ich auch versucht. Da Daten bei ecowitt und bei Wunterground ankommen sendet die Station sicher auch.  Ich habe nun Versuchsweise den Wunderground-Adapter installiert und bekomme über diesen Umweg die Daten in den ioBroker.
Da dass aus meiner Sicht aber nicht die Lösung sein kann werde ich mich mit Wireshark bzw. Netcat (mein Linux ist ziemlich eingerostet, muss da viel nachlesen) befassen. Komme aber wahrscheinlich erst am Wochenende dazu.

phifogg, 2021-03-17:
Wenn du an den Berechtigungen im Raspi zweifelst kannst auch auf dem ioBroker host nachdem der Listener läuft schauen ob der Port offen ist:

`netstat -tulpn | grep LISTEN
`

Hier müsste ja der von die gewählte Port (1030) gelistet sein und wenn Du den Adapter stoppst sollte der wieder weg sein.
Wenn das geht kannst auch mit einem Browser einfach mal http://192.168.153.111:1030/wetterdaten/wetterstation1.php?hugo=blubb aufrufen. Dann müsste der Listener zumindest reagieren und eine Ausgabe generieren dass er das nicht versteht :D

FranzKillig, 2021-03-17:
Habe das mal geprüft. Der Port ist da, sobald der Adapter läuft. Eine Rückmeldung im Browser gibt es jedoch nicht.

phifogg, 2021-03-17:
das ist doch schonmal gut. Im Browser gibt es keine Rückmeldung, korrekt. Aber im Debug-Log des Adapters müsste Deine Anfrage kommen.

FranzKillig, 2021-03-17:
Stimmt, da kommt was an:

sainlogic.0 | 2021-03-17 11:01:10.749 | warn | (5417) Listener received illegal request: /favicon.ico
-- | -- | -- | --
sainlogic.0 | 2021-03-17 11:01:10.658 | warn | (5417) State "sainlogic.0.info.last_listener_update" has no existing object, this might lead to an error in future versions
sainlogic.0 | 2021-03-17 11:01:10.638 | debug | (5417) Listener received WU update: {}

phifogg, 2021-03-17:
Das ist doch schonmal gut, dann läuft der Adapter richtig. Nun liegt es an der Wetterstation auch da Daten anzuliefern. Da kann ich leider wenig helfen :/ Sorry.

phifogg, 2021-03-27:
Läuft es nun bei Dir? Kann ich den Issue schliessen?

FranzKillig, 2021-03-28:
Hallo Daniel,

nein leider noch nicht, ich hatte aber auch in den letzten eineinhalb Wochen auch leider keine Zeit  mir die Sache noch näher anzusehen.  Ich habe nur per Netcat mal kurz den Port gelogt und da kam nicht an.  Ich will mal versuchen mit Wireshark zu logen was die Wetterstation überhaupt sendet.  Den bei Wunderground kommen die Daten ja an.  In Wireshark muss ich mich jedoch zuerst einlesen. Melde mich wieder sobald ich irgendwelche Erkenntnisse habe.

Gruß
Franz

AlteEnte, 2021-04-14:
Ich habe leider das gleiche Problem wie Franz. 0.7.er Version installiert, Port ist offen, aber es kommen keine Werte an.

phifogg, 2021-04-14:
Kannst du versuchen mit einem Browser auf den ioBroker was anzuliefern?
Du müsstest mit http://<ip_iobroker>:<port auf der config>/<path der config> gehen, dann sollte im Debug eine Meldung kommen, ähnlich hier https://github.com/phifogg/ioBroker.sainlogic/issues/61#issuecomment-800954821.

AlteEnte, 2021-04-14:
[image]

Meinst du sowas?

AlteEnte, 2021-04-14:
Also ich habe nicht viel Ahnung, aber ich habe die Vermutung, dass nicht die richtigen Pakete gesendet werden. Das spukt mir Wireshark aus:
[image]

phifogg, 2021-04-15:
Ja, das meinte ich. Sieht doch auf Adapter-Seite gut aus. Er reagiert wenn Du ihn ansprichst :D

Das Wireshare-Snippet ist ein Broadcast gesendet von 192.168.178.49 - ist das Deine Wetterstation?

Am einfachsten filterst Du im Wireshare auf Src-IP und gibst die der Wetterstation an. Dann solltest Du sehen was da so alles kommt. Wenn das geht kannst noch auf dst (destinaiton) ip filtern und hier den ioBroker host angeben. Dann siehst was die Station zum Adapter schickt. Wenn da nichts kommt schickt sie nichts.

AlteEnte, 2021-04-15:
Genau, dass ist meine Wetterstation, sprich die Src-IP und diese sendet leider nur die Broadcast Meldung. Daher vermute ich, dass sie nichts schickt :(
Warum ist mir allerdings schleierhaft, habe auch alles so eingestellt wie vorgegeben ...

FranzKillig, 2021-04-15:
Hallo Zusammen,

ich habe nun auch mal ein wenig mit Wireshark geloggt.  Das Ergebnis ist exakt das selbe wie bei AlteEnte und wird alle zwei Sekunden gesendet..  Ich vermute jedoch, dass da wahrscheinlich bei uns beiden etwas am Filter nicht stimmt. Denn ich lasse meine Station die Wetterdaten auch an Wunderground senden und das funktioniert. Allerdings müsste man die Datenpakete ja auch irgendwann mal in Wireshark mitbekommen.  Auf der anderen Seite hat die WS View-APP kürzlich beim Aufruf eine Firmwareupdate für die Wetterstation gemeldet. In den Verbesserungen stand irgendwas drin, das man  bei der Funktion Customized-Daten Senden irgendwelche Fehler behoben hätte. Eventuell sind jetzt noch mehr vorhanden oder es gibt damit generelle Probleme.

AlteEnte, 2021-04-16:
Moin ... ein Update von mir.
Ich habe es in der Wetterstation auf Ecowitt gestellt und nochmal vom Strom getrennt und jetzt kommen auch Werte bei mir an ... läuft jetzt super - Danke an Fogg :)

FranzKillig, 2021-04-16:
Ich habe nun wie AlteEnte, auch das Protokoll auf Ecowitt umgestellt und die Station neu gestartet. nun sin auch bei mir die Werte da. Noch kurz zum Abgleich. Bei mir wird im Adapter WS2900_V2.01.13 und EasyWeatherV1.5.8 angezeigt, obwohl es offiziell eigentlich eine WS 3500 sein sollte. Eventuell hat das ganze ja damit zu tun.
Danke auch von mir an Fogg und an AlteEnde. Wollte das auch schon länger mal ausprobieren, hatte aber keine Hoffnung das dies was bringen soll.

AlteEnte, 2021-04-16:
Ich hab den Adapter WS2900_V2.01.12 und EasyWeatherV1.5.8 obwohl ich ebenfalls die WS 3500 habe. Vielleicht noch als Anregung - es fehlen noch ein paar Werte wie sainlogic.0.weather.current.UVraw, sainlogic.0.weather.current.dewpointtemp, sainlogic.0.weather.current.soilmoisture und sainlogic.0.weather.current.windchilltemp.
Ich glaube den dewpoint und windchill zeigt er aber an ...

phifogg, 2021-04-17:
Ecowitt schickt leider weniger Daten als Wunderground - soweit ich weiss zumindest.
Ihr könnt im Debug ja mal schauen was alles an Daten kommt. Wenn da Werte dabei sind die nicht im ioBroker gespeichert werden kann ich die easy aufnehmen.

Aber schonmal super dass es funktioniert jetzt!

AlteEnte, 2021-04-17:
Die Werte scheinst du noch nicht zu haben:
 No mapping found for dataelement: wh65batt
No mapping found for dataelement: totalrainin
No mapping found for dataelement: eventrainin
No mapping found for dataelement: dateutc
No mapping found for dataelement: PASSKEY

Oder verstehe ich die Meldung falsch?

phifogg, 2021-04-27:
Korrekt. Date und passkey brauch ich auch nicht. Die anderen könnte ich wohl aufnehmen :D

AlteEnte, 2021-04-27:
An die ganzen Max Min Werte komme ich mit dem Ecowitt Protokoll sicherlich nicht ran, oder?!

phifogg, 2021-04-27:
Eher nicht, die kommen typischerweise nur beim Scheduler raus.

phifogg, 2021-05-29:
Geschlossen wegen Inaktivität. Bitte neu aufmachen sollte es immer noch nicht gehen.

closed 2021-05-29 by phifogg

1rfsNet, 2022-05-14:
Ich habe nach einigen Stunden jetzt die Ursache und eine Lösung für das Problem gefunden. Die Station ID und Station Key müssen beide zwingend ausgefüllt sein. Die WS3500 scheint das prüfen und anstatt eine Fehlermeldung auszugeben, wird einfach gar nichts geschickt. 
Ich habe jetzt jeweils einfach ein "x" in das Feld für Station ID und Station Key geschrieben und das reicht aus.

So jetzt schickt die Wetterstation die Daten an den ioBroker. Der Adapter nimmt sie allerdings noch nicht an. Fehlermeldung im Log ist:
`Listener received illegal request: /weatherstation/updateweatherstation.phpID=x&PASSWORD=x&indoortempf=77.5&tempf=75.2&dewptf=51.8&windchillf=75.2&indoorhumidity=40&humidity=44&windspeedmph=0.0&windgustmph=0.0&winddir=318&absbaromin=29.856&baromin=29.915&rainin=0.000&dailyrainin=0.012&weeklyrainin=0.012&monthlyrainin=0.012&solarradiation=5.97&UV=0&dateutc=2022-05-14%2014:44:58&softwaretype=EasyWeatherV1.6.4&action=updateraw&realtime=1&rtfreq=5`

In der App muss jetzt noch zusätzlich ein "?" hinter dem Path ergänzt werden (also z.B. "/weatherstation/updateweatherstation.php?"). Im Adapter muss aber lediglich der Pfad ohne das ? angegeben werden. 

Vielleicht könnte das irgendwo ins Readme aufgenommen werden, das hilft sicherlich auch anderen.

phifogg, 2022-05-18:
Cool, danke für die Info! Nehme ich gerne auf.