phifogg/ioBroker.sainlogic #64: Compatibility with Froggit DP1500?

mcmuller, 2021-03-16:
Any Info if it is compatible with [https://www.froggit.de/product_info.php?info=p410_dp1500-wi-fi-wetterserver-usb-dongle.html](url)
As Froggit DP1500 supports soil moisture through WS View I'm curious if someone tried this yet.

phifogg, 2021-03-17:
I have not tried it as I do not own this type of station. But on this [thread](https://forum.iobroker.net/topic/28384/linux-shell-skript-wlan-wetterstation/2094?_=1615966686340) it is mentioned that the DP1500 works with a shell skript using the same technology like my adapter. So it should basically work.

If you add additional sensors like soil moisture I would most likely need to add some config to read/store the data coming from that sensor. But that is a simple exercise once you have it. I just need the debug log to see what the station is delivering.

mcmuller, 2021-03-17:
...good to know that the output looks the same like from the Stations. I ordered sensors and receiver and will provide you with data for integration asap! Thanks for your work!

mcmuller, 2021-03-22:
Got the Froggit DP1500 today and did a debug log for you. There can be a total of 8 Soil Moisture Sensors (called DP1000) in the system, according to the docs. Would be very nice if you add them to your adapter.
[sainlogic-debug-GW1000A.txt](https://github.com/phifogg/ioBroker.sainlogic/files/6184019/sainlogic-debug-GW1000A.txt)

phifogg, 2021-03-23:
The interessting part ist this one:

`2021-03-22 17:01:16.047 - debug: sainlogic.1 (10353) Listener body is PASSKEY=5166D3F509F4FD30F5EF49B0BF29D5EE&stationtype=GW1000A_V1.6.6&dateutc=2021-03-22+17:01:16&tempinf=71.2&humidityin=49&baromrelin=29.935&baromabsin=29.935&tempf=66.0&humidity=56&winddir=229&windspeedmph=4.70&windgustmph=9.17&maxdailygust=13.65&solarradiation=132.24&uv=1&rainratein=0.000&eventrainin=0.000&hourlyrainin=0.000&dailyrainin=0.000&weeklyrainin=0.000&monthlyrainin=0.000&yearlyrainin=0.000&totalrainin=0.000&soilmoisture1=35&soilmoisture2=31&wh65batt=0&soilbatt1=1.6&soilbatt2=1.5&freq=868M&model=GW1000_Pro
`
In your case the parameters for soil are these:
soilmoisture1=35
soilmoisture2=31

Do you know what unit these are? I can easily add them to the config.

phifogg, 2021-03-27:
Version 0.7.0 should support now these devices. Check it out and let me know.

mcmuller, 2021-03-27:
Thanks! Will try when I'm home next week

Coalado, 2021-04-23:
I tested GW1000 and WH51 (Same from Ecowitt) today, and it works great.

mcmuller, 2021-04-23:
> Version 0.7.0 should support now these devices. Check it out and let me know.

Back from an injury I tested and it worked for my 5 sensors. Only caveat is the battery voltage is only reported for the first sensor. Will provide log later!

phifogg, 2021-04-23:
Awesome, i take the GW100 and WH51 into the supported devices list...

mcmuller, 2021-04-28:
can confirm its pulling soilbatt aka Batterylevel for all sensors now!
Ah, sehe Du bist zu DE gewechselt- Dankeschön für die Anpassung!

closed 2021-04-28 by mcmuller