phifogg/ioBroker.sainlogic #71: Version 0.6.6 wird nicht installiert

F95Markus, 2021-04-23:
Hallo,
ich habe leider das Problem das ich den Adapter nicht updaten kann. 

Anbei hab ich die beiden Logdatein
[image]
[image]

Ich hoffe ihr könnt mir helfen. Soweit läuft der Adapter mit meiner Froggit WH3000 sehr gut mittlerweile.

phifogg, 2021-04-23:
Das ist komisch, die 0.6.6 gibt es sehr wohl:
https://github.com/phifogg/ioBroker.sainlogic/releases/tag/0.6.6

Welche Version hast Du aktuell drauf?

F95Markus, 2021-04-24:
Aktuell ist die 0.6.0 installiert und bekomme auch keine höhere installiert. Da ist dann das selbe Problem.

olafbono, 2021-04-26:
Ich hatte das gleiche Problem, 0.6.6 wird von npm nicht gefunden.

Workaround: Auf der Shell einloggen als user iobroker, dann Version 0.6.6 als Archiv runterladen (z.B. wget https://github.com/phifogg/ioBroker.sainlogic/archive/refs/tags/0.6.6.tar.gz), dann aus dem Archiv installieren:

npm install 0.6.6.tar.gz --prefix "/srv/iobroker"
(oder /opt/iobroker, je nachdem, wo iobroker installiert ist).

F95Markus, 2021-04-27:
Dankeschön so hat es jetzt geklappt.

phifogg, 2021-04-27:
Wow... das sollte nicht so sein. Aber schön wenn es geklappt hat. 0.7.0 kommt bald, dann schauen wir mal ob es den Umweg wieder braucht....

closed 2021-04-27 by phifogg