phifogg/ioBroker.sainlogic #72: Probleme mit Froggit WH2601

Karl1234567890, 2021-04-27:
Hallo, die Einbindung der Wetterstation hat soweit funktioniert, nur werden keine Min- und Max- Werte ausgegeben. In der Log wird dauernd die Fehlermeldung:
sainlogic.0 | 2021-04-27 19:09:18.268 | error | (27304) Scheduler received data for unkown command 85 subcommand 91
ausgegeben. 
Wo liegt hier der Fehler - vielen Dank für die Hilfe

phifogg, 2021-04-27:
Hallo Karl, das klingt spannend. Eine Command 85 subcommand 91 kenn ich so leider nicht, eigentlich schicke ich normal Command 11 Subcommand 4 was die aktuellen Wetterdaten auslesen sollte. Kannst du mal Debug aktivieren? Dann sehen wir den String der da kommt. Weiss nicht ob es mir hilft, aber wir können es versuchen.

Karl1234567890, 2021-04-27:
@phifogg vielen Dank für deine rasche Rückmeldung. Wie beschrieben die aktuellen Messdaten funktionieren super, nur die Max-/Min werden nicht angezeigt. Ich habe auch leider meine Station nicht in der Liste gefunden, deshalb habe ich eine andere Froggit verwendet - ist das vielleicht die Ursache?
Deine rasche Hilfe ist echt toll - 1000 DANK! LG
[sainlogic.txt](https://github.com/phifogg/ioBroker.sainlogic/files/6386699/sainlogic.txt)

phifogg, 2021-04-27:
Ah.. dann kann sein dass deine Wetterstation schlicht den Befehl nicht kennt und daher komisch reagiert. Du kannst im Scheduler Tab den Call auf min/max einfach ausschalten. Wenn mir jemand sagen/zeigen kann wie das bei der Froggit geht kann ich es schon einbauen. Aber so ist es etwas 'jugend forscht'.

Karl1234567890, 2021-04-27:
min und max habe ich schon versucht zu deaktivieren - die Meldungen kommen dennoch. Es ist nur noch "Get Firmware" und "Get Current Data" aktiv. Lt. Rechnung soll die Wetterstation Froggit WH2600 Pro sich nennen. Es ist echt verwirrend, da im Internet es baugleiche Typen gibt, die WH2601 oder WH2650. Deshalb habe ich auch bei der Einstellung WH2650A gewählt.

phifogg, 2021-04-27:
In dem Log welches Du angehängt hast wären noch alle aktiv:
sainlogic.0	2021-04-27 20:08:04.602	info	(11332) Scheduler call ws_getmindaily activated
sainlogic.0	2021-04-27 20:08:04.601	info	(11332) Scheduler call ws_getmaxdaily activated
sainlogic.0	2021-04-27 20:08:04.601	info	(11332) Scheduler call ws_getmin activated
sainlogic.0	2021-04-27 20:08:04.599	info	(11332) Scheduler call ws_getmax activated
sainlogic.0	2021-04-27 20:08:04.599	info	(11332) Scheduler call ws_getcurrent activated
sainlogic.0	2021-04-27 20:08:04.598	info	(11332) Scheduler call ws_getfirmware activated

Ja, das mit den Stationen ist so ne Sache, fallen vermutlich alle vom gleichen Band in China...

Bei welchen Einstellungen meinst Du? Im Adapter kann man da nichts einstellen - wär mir neu ;D

Karl1234567890, 2021-04-27:
Ja beim Log war alles wieder an - vorher hatte ich die Bereiche aus. Bei der Verbindung über WS View kann man die Station einstellen. Dort werden die Messwerte ja auch geholt - oder?

phifogg, 2021-04-27:
Ah.. WSView, da weiss ich wenig. Holt das nicht die Werte von WeatherUnderground? Sonst könnte es doch auch nicht die Graphen zusammenbauen.

Der Adapter holt sich nichts aus WSView.

Karl1234567890, 2021-04-27:
In WSView stellt man ein wo die Daten gesendet werden sollen. Ich habe Ecowitt eingestellt - auch im Adapter. Mit WU hat es nicht funktioniert - mit Ecowitt waren die aktuellen Messdaten gleich im IObroker.

phifogg, 2021-04-27:
ja genau. schon merkwürdig, müsste auch mit WU gehen. Zumindest meine macht das problemlos. Ist ja nur das Format der Daten anders, das Prinzip ist das gleiche.

Karl1234567890, 2021-04-27:
Werde morgen mal alles löschen, die WS zurücksetzen und nochmals beginnen. Vielleicht ändert sich was. Danke für deine Bemühungen

phifogg, 2021-05-29:
Geschlossen wegen Inaktivität. Bitte ein neues Issue aufmachen sollte das immernoch ein Problem sein.

closed 2021-05-29 by phifogg

Apollon77, 2022-09-20:
I had a look here again and the app „ws view“ that I can also use to update firmware and such seems to be able to read data in that binary format. I will try to get wireshark running tinder what they send