phifogg/ioBroker.sainlogic #77: Compatibility check and testing for Node.js 14 and 16

ioBrokerChecker, 2021-05-06:
Dear Adapter developer,

Node.js 14 is now available for a year and Node.js 16 was release just some days ago and will become LTS by October 2021. We plan to update the ioBroker Node.js recommendation (currently 12.x) to 14.x later this year.

Please check your adapter with Node.js 14 especially, and ideally also directly with Node.js 16

Please add both versions to the adapter testing which is executed on commits.

If your adapter requires a certain minimum version of Node.js please set the 'engine' setting in package.json accordingly! Please also do this if the adapter is not able to work in certain Node.js versions, so that ioBroker can prevent users from installing te adapter if not compatible!

On questions please talk to us at https://github.com/ioBroker/ioBroker.js-controller/issues/1138 

Please close the issue after you checked it.

Thank you very much for your support!

phifogg, 2022-01-20:
Tested Nodejs 14 - confirmed working.

phifogg, 2022-01-20:
Tested Nodejs 16 - confirmed working.

closed 2022-01-20 by phifogg