phifogg/ioBroker.sainlogic #83: Sensor availability

Coalado, 2021-05-27:
I have the GW1000 Gateway and 8 Moisture sensors. 
The datapoints of all 8 sensors get updated every ~60 seconds. 
It seems like they get updated even if they have no actual connection to the gateway. (e.g. out of battery, out of range,..)

Is there any information about the connection between the sensors and the gateway?

phifogg, 2021-05-29:
I can only display what is coming back from the stations - sorry. At this point I am not aware of any connection properties.
There is a battery level state in the info-channel. Should be one for each sensor. If this falls below a good value you could alert yourself.

phifogg, 2021-07-27:
Are we ok to close this one?

Coalado, 2021-07-27:
yes, thanks

closed 2021-07-27 by Coalado