phifogg/ioBroker.sainlogic #85: Support JSON as Input

MrDrache333, 2021-06-14:
This is not a Bug. Its a Feature Request.
The WS2900_V2.01.08 allows me to use a custom Server. I use my Server for this. In my Server I got the following JSON, which I would like to relay to this Adapter. When I just Post the Data to the HTTP-Endpoint, nothing happens but I got a 200 StatusCode.

```json
{"PASSKEY":"something","stationtype":"EasyWeatherV1.5.8","dateutc":"2021-06-14 19:28:11","tempinf":"73.9","humidityin":"57","baromrelin":"29.864","baromabsin":"30.142","tempf":"61.7","humidity":"87","winddir":"5","windspeedmph":"0.0","windgustmph":"0.0","maxdailygust":"11.4","rainratein":"0.000","eventrainin":"0.000","hourlyrainin":"0.000","dailyrainin":"0.000","weeklyrainin":"0.000","monthlyrainin":"1.827","totalrainin":"71.799","solarradiation":"11.25","uv":"0","wh65batt":"0","freq":"868M","model":"WS2900_V2.01.08"}
```

phifogg, 2021-06-15:
If your server is reachable from ioBroker you can switch the direction. The adapter has a feature to forward the received data to another URL.

To trouble shoot your issue it would be helpful if you can set the adatper into debug mode and send me the logs produced.

phifogg, 2021-07-04:
Your JSON code actually looks good. Can you put the adapter in debug mode and see what it states?

phifogg, 2022-01-20:
No more feedback since 6 months. closing issue.

closed 2022-01-20 by phifogg