phifogg/ioBroker.sainlogic #89: compact mode not supported

handy80, 2021-07-27:
Hi enabling compact mode on your adapter is failing.
command i used:
iobroker compact sainlogic.0 enable

Kind regards
Holger

phifogg, 2021-07-27:
Uf... never heard of compact mode. Need to investigate what that means.

handy80, 2021-07-27:
maybe you can find valuable information here: https://forum.iobroker.net/topic/32789/anleitung-f%C3%BCr-adapter-entwickler-compact-mode-testen

phifogg, 2021-07-28:
https://forum.iobroker.net/topic/18338/experimentell-js-controller-compact-mode

handy80, 2021-07-29:
This links to a forum thread that was started 2019 and ended somewhen in 2020. Is the "experimental" what you're pointing out or it's just a web reference for this issue?
I can provide more details (system, logs, configuration) for this issue in my case if it helps to solve the issue.

phifogg, 2022-01-19:
finally i try to look into this. I can reproduce the error:
<html>
<body>

sainlogic.1 | 2022-01-19 14:07:46.412 | warn | Terminated (INVALID_ADAPTER_CONFIG): Without reason
-- | -- | -- | --
sainlogic.1 | 2022-01-19 14:07:46.411 | error | sainlogic.1 invalid config

</body>
</html>

but... i have no clue why. i don't know enough about compact mode and what I need to add. needs some research.

phifogg, 2025-01-21:
Closed as too old. If still relevant open a new issue.

closed 2025-01-21 by phifogg