phifogg/ioBroker.sainlogic #91: adapter installation is failing

vdi-cloud, 2021-08-01:
When I try to install the adapter from iobroker admin guy, I receive following error

$ ./iobroker add sainlogic  --host iobroker
NPM version: 6.14.12
npm install iobroker.sainlogic@0.7.2 --loglevel error --prefix "/opt/iobroker" (System call)
npm ERR! code ETARGET
npm ERR! notarget No matching version found for iobroker.sainlogic@0.7.2.npm ERR! notarget In most cases you or one of your dependencies are requestingnpm ERR! notarget a package version that doesn't exist.
npm ERR! A complete log of this run can be found in:npm ERR!     /home/iobroker/.npm/_logs/2021-08-01T16_11_04_120Z-debug.log
ERROR: host.iobroker Cannot install iobroker.sainlogic@0.7.2: 1
ERROR: process exited with code 25

the corresponding log
[2021-08-01T16_11_04_120Z-debug.log](https://github.com/phifogg/ioBroker.sainlogic/files/6912975/2021-08-01T16_11_04_120Z-debug.log)

Version Info 
iobroker.admin V4.2.2
Node.js: v10.24.1
NPM: 6.14.12

MrDrache333, 2021-08-02:
The latest Version in this Repo is 0.7.2 The rpm Version is still on 0.6.6. Therefor the npm Version has to be updated to eliminate the Error.

vdi-cloud, 2021-08-02:
Ok, I did overlook this one. npm version is fixed, now I am getting a different error:
-----------------
$ ./iobroker add sainlogic  --host iobroker
NPM version: 7.20.3
npm install iobroker.sainlogic@0.7.2 --loglevel error --prefix "/opt/iobroker" (System call)
npm ERR! code ETARGET
npm ERR! notarget No matching version found for iobroker.sainlogic@0.7.2.
npm ERR! notarget In most cases you or one of your dependencies are requesting
npm ERR! notarget a package version that doesn't exist.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/iobroker/.npm/_logs/2021-08-02T18_22_12_745Z-debug.log
ERROR: host.iobroker Cannot install iobroker.sainlogic@0.7.2: 1
ERROR: process exited with code 25
-----------------

The mentioned 2021-08-02T18_22_12_745Z-debug.log is empty..

MrDrache333, 2021-08-02:
Yes. The nom Version of this Adapter is outdated. You can simply install this repo with
```npm install https://github.com/phifogg/ioBroker.sainlogic --prefix "/opt/iobroker"```
Then the newest Version is installed

phifogg, 2021-08-03:
Thanks for your help @MrDrache333  - I am working with the ioBroker team to get the npm version updated. However, they ask for some additional work from me with takes extra time.

phifogg, 2021-10-07:
0.7.3 has been pushed to the official repo... should show up shortly in ioBroker.

closed 2021-10-07 by phifogg