phifogg/ioBroker.sainlogic #92: Feature request: Regen Variable

e-i-k-e, 2021-08-04:
Hallo,

könnte vielleicht folgende Kleinigkeit integriert werden? 
Bei Regen einen Variable auf true setzen und bei Trockenheit auf false?

phifogg, 2021-08-07:
Klar, möglich ist das :D
Ab wieviel mm akueller Wassermenge sollte es denn als Regen gelten?....

e-i-k-e, 2021-08-11:
Das muss mal getestet werden. :)
Aber vielleicht für den Anfang direkt ab 1mm.

Eistee82, 2022-07-22:
Bei mir wird rain nicht auf 0 gesetzt sondern bleibt auf dem letzten Wert stehen. Hab mir das auf 0 setzen jetzt mit Blockly programmiert nach 5min aber das könnte der Adapter ja eigentlich auch selbstständig machen. auf wunderground geht der Wert ja auch auf 0 wenn der nicht mehr von der Station gesendet wird.