phifogg/ioBroker.sainlogic #96: adapter installation without  weather station

volkerz12, 2021-09-26:
At first I had problems with the installation of the Sainlogic adapter. It wasn't until I installed via GITHUB that the installation succeeded. I haven't bought a weather station yet. First I wanted to see if the adapter installation worked. It is unusual for the adapter not to show up in the instances. The adapter appears in the objects. Is this behavior normal for this adapter?

phifogg, 2021-09-26:
It should appear in instances as well, at least it does for me :D

As fot the installation, I know, trying to get the repo team to accept the latest version, but they added some extra hurdles which take me extra time to figure it out. Will try to work on this during october...

volkerz12, 2021-09-26:
Thanks for answering. Could the reason for not appearing in  instances, the
fact I have no wether station yet?
What is repo team?

Greetings

Volker Ziegler

volkerz12, 2021-09-29:
> No, that should not be an issue. You need the instance to configure how to
> access the weather station.
>
> Can you check if there is any message in the log while/after installing?
>
> There is a ioBroker repository with know adapters which is maintained by a
> bespoke team. They control what is allowed in the list of adapters where
> you do not need to install from github URLs.

Hello,

there is no message in the log.

I have installed the adapter again via github. Please see attached
installing log.

Grreetings

Volker Ziegler

phifogg, 2021-09-29:
Hi Volker,
can't see any attachment :/
Daniel

volkerz12, 2021-09-29:
Hi Daniel,

sorry, here is attachment.

Greetings

Volker Ziegler

phifogg, 2021-09-29:
Seems the attachment doesn't carry over to github :/

volkerz12, 2021-09-29:
sorry I did not understand. Did you get the  attachment ?

Volker Ziegler

phifogg, 2021-09-29:
No... seems github does not store the attachment in your email. Can you copy/paste the content?

volkerz12, 2021-09-29:
here is it. I hope the words are big enough to read

[image: image.png]
[image: image.png]

Volker Ziegler

phifogg, 2021-09-29:
doesn't work either ... sorry. All I see is:
[image: image.png]
[image: image.png]

You can try to answer to this issue on the github website and attache the files/images there.

volkerz12, 2021-09-29:
Let us try here
[saint.pdf](https://github.com/phifogg/ioBroker.sainlogic/files/7253448/saint.pdf)

volkerz12, 2021-12-11:
Hello,

I need you help again.

I have bought a Sainlogic Weatherstation WS3500. I want to connect to
iobroker and import the weather data in iobroker. I need temperatur,
pressure, humidity, windspeed, winddirection

Do I have to use Wunderground or ecowitt?

Thanks

Greetings

Volker Ziegler

phifogg, 2021-12-13:
Hi Volker,
good question. Usually the ecowitt has more values it transmits. Would recommend you to test both with debug and see what you get in the logs.

Daniel

volkerz12, 2021-12-13:
Hi Daniel,

I have tried ecowitt. But in iobroker is no data displayed.

please see attachement.

greetings

Volker Ziegler

volkerz12, 2021-12-13:
here is the  attachement

Volker Ziegler

volkerz12, 2021-12-16:
HI Daniel,

do you have an answer?

Greetings

Volker

phifogg, 2021-12-17:
Hm.. attachment is missing again.

volkerz12, 2021-12-17:
Hi Daniel,

sorry again. Here is it.

Greetings

Volker

phifogg, 2021-12-20:
no, it's not. not sure if it works via email...

volkerz12, 2021-12-20:
Ok, please try the link below

https://gofile.io/d/qXNNKF

Greetings

Volker Ziegler

volkerz12, 2021-12-21:
Did you get the attachment?

https://gofile.io/d/qXNNKF

Greetings

Volker Ziegler

phifogg, 2021-12-21:
Yes, link works. thank you.

The protocol-section is only relevant for the listener. This would mean your weatherstation needs to be configured to send it's data to ioBroker. In my station I can do that with WSView and custom setup. Not sure this works for you.

This is how it looks on my WSView settings:
[image]

As for the scheduler, it will only be able to pull some information. If you activate debug I might be able to decode it... not fully sure though. So far I was lucky to find a documentation how to parse the content.

volkerz12, 2021-12-21:
Ok,

thanks for helping. This will not work. I send weatherstatoin  back to
amazon.

Greetings

Volker Ziegler

adx-git, 2021-12-22:
Which one did you use, might be interesting for other as well.

phifogg, 2021-12-22:
He mentioned a Sainlogic Weatherstation WS3500 some posts back.

@volkerz12 : Sorry to hear that. Sainlogic has so many of these stations and it seems not all are equal.

Closing this issue for now. Feel free to open a new one when needed.

closed 2021-12-22 by phifogg