phifogg/ioBroker.sainlogic #103: Feature Request: Zeitpunkte der Min- und MaxWerte

paulinchen14, 2022-01-05:
Hallo,
bisher lese ich die WS980WIFI direkt aus.
Ich möchte aber jetzt gerne auf den Adapter umsteigen.

Frage:
Wäre es möglich, die Zeitpunkte der Min- und Maxwerte mit auszulesen?
Bei der WS980WIFI könnten diese ja als getrennte Stunden- und Minutenangabe direkt nach dem eigentlichen Wert aus der API mit ausgelesen werden?

  $format =
    'x7/' .                 # Override first 7 bytes
    'n1Innentemp/' .        # Get the next 2 bytes
    'C1Innentemp_h/' .      # Get the next 1 byte
    'C1Innentemp_m/' .      # Get the next 1 byte
    'x1/' .                 # Override 1 byte
    'n1Aussentemp/' .       # Get the next 2 bytes
    'C1Aussentemp_h/' .     # Get the next 1 byte
    'C1Aussentemp_m/' .     # Get the next 1 byte
    'x1/' .                 # Override 1 byte
    'n1Taupunkt/' .         # Get the next 2 bytes
    'C1Taupunkt_h/' .       # Get the next 1 byte
    'C1Taupunkt_m/' .       # Get the next 1 byte
...

Ansonsten weis ich ja gar nicht, wann am heutigen Tag die höchste Windgeschwindigkeit war usw.

Viele Grüße,
paulinchen

phifogg, 2022-01-05:
Wenn die API das liefert klar, dann sollte ich das auch rauskriegen. Leider wird mir aus dem Code-Snippet von Dir nicht klar wir :D

paulinchen14, 2022-01-06:
Bisher nutze ich u.a. folgendes php-script zum auslesen der Max-Werte.
Damit erhalte ich neben dem Wert auch den entsprechenden Zeitpunkt:

Beispiel:
InnenTemp: maximaler Wert
InnenTemp_h: Stunde des maximalen Wertes
InnenTemp_m: Minute des maximalen Wertes

Vielleicht kannst Du damit was anfangen?
Wäre toll, wenn Du das integrieren könntest.

`<?php

################################################################################
#
# Scriptbezeichnung: ws980_abfrage_maximum.php
# Stand: 2019_07_30
# Author: Heiko Wilknitz
# Änderungen: Thomas Schmidt
# - Fehlerkorrekturen und Anpassungen
# - Ausgabe als Json eingefügt
#
# Abruf der Tageshoechstwerte von einer WS980WiFi Station
#
# Beschreibung des Kommunikationsprotokolls der Wetterstation von
# R. Petzoldt (R.Petzoldt@web.de)
# https://www.elv.de/topic/protokolldefinition-zum-datenaustausch-ws980-zum-pc.html
#
################################################################################

  $ip = "192.168.1.222";
  $port = 45000;
  //Tagesmaximum...
  $str=chr(0xFF).chr(0xFF).chr(0x0B).chr(0x00).chr(0x06).chr(0x07).chr(0x07).chr(0x1F);
  $sock = socket_create(AF_INET, SOCK_STREAM, SOL_TCP);
  if (($sock = socket_create(AF_INET, SOCK_STREAM, SOL_TCP)) === false) {
    echo "socket_create() fehlgeschlagen: Grund: " . socket_strerror(socket_last_error()) . "\n";
  }
  if (socket_connect($sock, $ip, $port) === false) {
    echo "socket_connect() fehlgeschlagen: Grund: " . socket_strerror(socket_last_error($sock)) . "\n";
  }
  $len = socket_write($sock, $str, strlen($str));
  if (false === $len) {
    echo "socket_write() fehlgeschlagen: Grund: " . socket_strerror(socket_last_error($sock)) . "\n";
  }
  if (false === ($buf = socket_read ($sock, 2048, PHP_BINARY_READ))) {
    echo "socket_read() fehlgeschlagen: Grund: " . socket_strerror(socket_last_error($sock)) . "\n";
    return;
  }
  socket_close($sock);
  $format =
    'x7/' .                 # Override first 7 bytes
    'n1Innentemp/' .        # Get the next 2 bytes
    'C1Innentemp_h/' .      # Get the next 1 byte
    'C1Innentemp_m/' .      # Get the next 1 byte
    'x1/' .                 # Override 1 byte
    'n1Aussentemp/' .       # Get the next 2 bytes
    'C1Aussentemp_h/' .     # Get the next 1 byte
    'C1Aussentemp_m/' .     # Get the next 1 byte
    'x1/' .                 # Override 1 byte
    'n1Taupunkt/' .         # Get the next 2 bytes
    'C1Taupunkt_h/' .       # Get the next 1 byte
    'C1Taupunkt_m/' .       # Get the next 1 byte
    'x1/' .                 # Override 1 byte
    'n1Hitze/' .            # Get the next 2 bytes
    'C1Hitze_h/' .          # Get the next 1 byte
    'C1Hitze_m/' .          # Get the next 1 byte
    'x1/' .                 # Override 1 byte
    'C1Innenfeuchte/' .     # Get the next 1 byte
    'C1Innenfeuchte_h/' .   # Get the next 1 byte
[... 68 lines cut ...]
    'RelDruck_h' => $array['RelDruck_h'],
    'RelDruck_m' => $array['RelDruck_m'],
    'Windspeed' => $array['Windspeed'] / 10 * 3.6,
    'Windspeed_h' => $array['Windspeed_h'],
    'Windspeed_m' => $array['Windspeed_m'],
    'Windboe' => $array['Windboe'] / 10 * 3.6,
    'Windboe_h' => $array['Windboe_h'],
    'Windboe_m' => $array['Windboe_m'],
    'Regen' => $array['Regen'] / 10,
    'Regen_h' => $array['Regen_h'],
    'Regen_m' => $array['Regen_m'],
    'Licht' => $array['Licht'] / 10,
    'Licht_h' => $array['Licht_h'],
    'Licht_m' => $array['Licht_m'],
    'UvRaw' => $array['UvRaw'],
    'UvRaw_h' => $array['UvRaw_h'],
    'UvRaw_m' => $array['UvRaw_m'],
    'UvIdxRaw' => $array['UvIdxRaw'],
    'UvIdxRaw_h' => $array['UvIdxRaw_h'],
    'UvIdxRaw_m' => $array['UvIdxRaw_m']
  );
  //Werte-Array in JSON wandeln...
  $myJsonResultat = json_encode($myResultat);
  echo $myJsonResultat;
?>`

phifogg, 2022-01-09:
Ah... h und m sind stunden und minuten... ich hatte irgendwie 'high' oder 'max' gedacht und mich gewundert.
Ja, das sollte gehen. Ich schaue mir das die Tage mal an.

paulinchen14, 2022-01-10:
Würde mir so völlig genügen, Danke.

phifogg, 2022-01-12:
Ich habe die Version 0.8.0 frei gegeben. Muss nur noch warten bis die auftaucht :D 
Oder direkt aus dem Repo hier installieren. Dann ich mal gespannt ob es funktioniert.

paulinchen14, 2022-01-12:
Hallo Daniel,
vielen Dank für die schnelle Implementierung.
Prinzipiell funktioniert das auch prima.

Bei einigen Werten werden allerdings noch falsche Werte angezeigt.
Außerdem wäre es super, wenn die Stunden immer 2 stellig wären, dann könnte ich den String dann leichter zerlegen, also statt "1:15" wäre "01:15" besser.
 
Hier mal 2 Screenshots:

[image]

[image]

paulinchen14, 2022-01-12:
Nur zur Info:
bei den beiden Zeiten, wo fehlerhaft "15:NaN" ermittelt wird, wäre eigentlich "00:15" richtig.
Hilft Dir vielleicht bei der Fehlersuche...

phifogg, 2022-01-12:
Danke, das hilft auf alle Fälle. Ich denke auch alle Stunden unter 10 könnten noch Probleme machen 🤦 . 
Ich schaue mir das an und mache ein update.

phifogg, 2022-01-14:
Sollte nun mit 0.8.1 funktionieren.

paulinchen14, 2022-01-14:
Ja, das sieht deutlich besser aus.
Es gibt noch ein Problem:
Bei einem Wert wird mir als Zeitpunkt "1:531" angezeigt. Der korrekte Zeitpunkt wäre "00:00".
Wahrscheinlich wieder so ein "Spezialfall" mit der 0...

phifogg, 2022-01-19:
hm... sollte mit dem code aber auch gehen. sicher dass das nicht ein alter wert ohne update ist?
sonst gern nochmal das debug log schicken und ich schaue rein.

phifogg, 2022-01-20:
Schick mir doch bitte noch das debug log dazu. Wenn da 'nur' eine 0 drin steht sieht es so aus als ob er gar kein update bekommen hat auf dem Datenpunkt. Muss ich schauen ob IOB den Wert wegwirft weil 0 oder ob ich gar keinen schicke.

paulinchen14, 2022-01-20:
Mache ich gerne.
Das ganze tritt aber natürlich nur selten auf. Ich schau mal die nächsten Tage...

paulinchen14, 2022-01-20:
Hallo,
Ich habe alle Datenpunkte manuell gelöscht und den Adapter neu gestartet.
Es geht um den Zeitpunkt der "indoortemp". Der wäre eigentlich um "00:00", der Wert wird aber nicht in den Datenpunkt geschrieben?

hier mal die Debug-Ausgabe:

2022-01-20 20:25:19.472 - debug: sainlogic.0 (23952) Scheduler connected to weather station run 255,255,11,0,6,7,7,31
--
2022-01-20 20:25:19.495 - debug: sainlogic.0 (23952) FW Scheduler Received data string: ffff0b004d074100ca0000420028003a430014012e450028003a46370000475d023b48273a14044927e414044b004b09034c006b07214e0000001e021b5500048fee0c2b5600020c0d570e0c0d7440
2022-01-20 20:25:19.496 - debug: sainlogic.0 (23952) Data Command received: 11 subcommand 7
2022-01-20 20:25:19.499 - debug: sainlogic.0 (23952) Data object: {"indoortemp":202,"indoortemp_time":0,"outdoortemp":40,"outdoortemp_time":58,"dewpointtemp":20,"dewpointtemp_time":302,"windchilltemp":40,"windchilltemp_time":58,"indoorhumidity":55,"indoorhumidity_time":0,"outdoorhumidity":93,"outdoorhumidity_time":571,"pressureabs":10042,"pressureabs_time":5124,"pressurerel":10212,"pressurerel_time":5124,"windspeed":75,"windspeed_time":2307,"windgustspeed":107,"windgustspeed_time":1825,"dailyrain":30,"dailyrain_time":539,"solarradiation":298990,"solarradiation_time":3115,"UVraw":2,"UVraw_time":3085,"uvi":14,"uvi_time":3085}
2022-01-20 20:25:19.499 - info: sainlogic.0 (23952) Scheduler updating IOBroker states
2022-01-20 20:25:19.507 - debug: sainlogic.0 (23952) Setting value from data for weather.maxvalues.daily.UVraw to 2
2022-01-20 20:25:19.508 - debug: sainlogic.0 (23952) Setting value from data for weather.maxvalues.daily.UVraw_time to 12:13
2022-01-20 20:25:19.508 - debug: sainlogic.0 (23952) Setting value from data for weather.maxvalues.daily.dailyrain to 3
2022-01-20 20:25:19.509 - debug: sainlogic.0 (23952) Target for dailyrain unit is set: mm, using conversion: null
2022-01-20 20:25:19.509 - debug: sainlogic.0 (23952) Setting value from data for weather.maxvalues.daily.dailyrain_time to 02:27
2022-01-20 20:25:19.510 - debug: sainlogic.0 (23952) Setting value from data for weather.maxvalues.daily.dewpointtemp to 2
2022-01-20 20:25:19.510 - debug: sainlogic.0 (23952) Target for dewpointtemp unit is set: °C, using conversion: null
2022-01-20 20:25:19.510 - debug: sainlogic.0 (23952) Setting value from data for weather.maxvalues.daily.dewpointtemp_time to 01:46
2022-01-20 20:25:19.511 - debug: sainlogic.0 (23952) Setting value from data for weather.maxvalues.daily.indoorhumidity to 55
2022-01-20 20:25:19.511 - debug: sainlogic.0 (23952) Setting value from data for weather.maxvalues.daily.indoortemp to 20.2
2022-01-20 20:25:19.511 - debug: sainlogic.0 (23952) Target for indoortemp unit is set: °C, using conversion: null
2022-01-20 20:25:19.512 - debug: sainlogic.0 (23952) Setting value from data for weather.maxvalues.daily.outdoorhumidity to 93
2022-01-20 20:25:19.512 - debug: sainlogic.0 (23952) Setting value from data for weather.maxvalues.daily.outdoorhumidity_time to 02:59
2022-01-20 20:25:19.513 - debug: sainlogic.0 (23952) Setting value from data for weather.maxvalues.daily.outdoortemp to 4
2022-01-20 20:25:19.513 - debug: sainlogic.0 (23952) Target for outdoortemp unit is set: °C, using conversion: null
2022-01-20 20:25:19.513 - debug: sainlogic.0 (23952) Setting value from data for weather.maxvalues.daily.outdoortemp_time to 00:58

Maxtor62, 2022-05-04:
Hallo zusammen,
hier braucht mal wieder ein Anfänger Hilfe. Habe den aktuellen Adapter installiert. Bekomme bei Ecowitt alle Daten der Station angezeigt. Leider habe ich, wenn ich den Adapter v0.8.2 starte sofort den Error:

2022-05-04 07:38:31.722 - error: sainlogic.0 (148666) Connection error on 192.168.178.31:45000: Error: connect EISCONN 192.168.178.31:45000 - Local (192.168.178.38:40722)
--

Wäre toll wenn jemand helfen könnte.
Danke und Grüße

phifogg, 2022-05-04:
@Maxtor62: Mach doch bitte einen eigenen Issue dafür auf. Und am besten dazuschreiben ob Du Listener oder/und Scheduler benutzt.

phifogg, 2025-01-21:
@paulinchen14  Ist das immer noch ein issue?

phifogg, 2026-08-28:
Closing as old.

closed 2026-08-28 by phifogg