phifogg/ioBroker.sainlogic #107: Fehlende Werte bei Sainlogic WS 3500

Sneaker29, 2022-01-12:
Hallo,
ich lese mit dem Adapter meine Sainlogic WS 3500 aus. Das klappt soweit auch alles super, leider werden 4 Werte nicht ausgelesen:

UVraw
dewpointtemp
windchilltemp
solarradiation

Bei den ersten 3 dieser Werte steht nur (null) gefolgt von der Einheit. Bei solarradiation steht immer 0 lx. Die Einheit ist hier aber eigentlich w/m².

Die fehlenden Werte werden von der WS 3500 geliefert und auch im weatherunderground-Adapter angezeigt. Ebenso bei Ecowitt und in der App.

Ich würde aber gerne den Sainlogic-Adapter nutzen, weil beim weatherunderground-Adapter andere Werte fehlen.

Ist das bekannt, dass diese Werte nicht ausgelesen werden oder liegt das Problem eher auf meiner Seite?

Verwendeter Adapter ist 0.73
javascript: 3.3.22

Viele Grüße
Sneaker

phifogg, 2022-01-12:
Hallo Sneaker, kannst Du den Adapter mal in debug-modus versetzen und einen lauf machen? Dann kann ich sehen welche Infos kommen. Vielleicht heissen die Felder nur anders wie ich erwarte.

Sneaker29, 2022-01-13:
Hallo phiogg,

ich hoffe, Du kannst hiermit was anfangen:

```
2022-01-13 23:02:04.334  - debug:  sainlogic.0 (5930) Listener body is  PASSKEY=CD4199C743F163BA5924C84E66579702&stationtype=EasyWeatherV1.5.9&dateutc=2022-01-13+22:02:02&tempinf=71.6&humidityin=52&baromrelin=30.213&baromabsin=30.526&tempf=34.5&humidity=93&winddir=326&windspeedmph=1.8&windgustmph=3.4&maxdailygust=10.3&rainratein=0.000&eventrainin=0.000&hourlyrainin=0.000&dailyrainin=0.000&weeklyrainin=0.390&monthlyrainin=1.720&totalrainin=5.370&solarradiation=0.00&uv=0&wh65batt=0&freq=868M&model=WS2900_V2.01.10
--
2022-01-13 23:02:04.341  - debug: sainlogic.0 (5930) No mapping found for dataelement: PASSKEY
2022-01-13 23:02:04.342  - debug: sainlogic.0 (5930) Extracting value for info.softwaretype(stationtype)
2022-01-13 23:02:04.343  - debug: sainlogic.0 (5930) No mapping found for dataelement: dateutc
2022-01-13 23:02:04.344  - debug: sainlogic.0 (5930) Extracting value for weather.current.indoortemp(tempinf)
2022-01-13 23:02:04.348  - debug: sainlogic.0 (5930) Extracting value for weather.current.indoorhumidity(humidityin)
2022-01-13 23:02:04.351  - debug: sainlogic.0 (5930) Extracting value for weather.current.pressurerel(baromrelin)
2022-01-13 23:02:04.353  - debug: sainlogic.0 (5930) Extracting value for weather.current.pressureabs(baromabsin)
2022-01-13 23:02:04.359  - debug: sainlogic.0 (5930) Extracting value for weather.current.outdoortemp(tempf)
2022-01-13 23:02:04.364  - debug: sainlogic.0 (5930) Extracting value for weather.current.outdoorhumidity(humidity)
2022-01-13 23:02:04.365  - debug: sainlogic.0 (5930) Extracting value for weather.current.winddir(winddir)
2022-01-13 23:02:04.366  - debug: sainlogic.0 (5930) Extracting value for weather.current.windspeed(windspeedmph)
2022-01-13 23:02:04.367  - debug: sainlogic.0 (5930) Extracting value for weather.current.windgustspeed(windgustmph)
2022-01-13 23:02:04.371  - debug: sainlogic.0 (5930) Extracting value for weather.current.maxdailygust(maxdailygust)
2022-01-13 23:02:04.373  - debug: sainlogic.0 (5930) Extracting value for weather.current.rain(rainratein)
2022-01-13 23:02:04.374  - debug: sainlogic.0 (5930) No mapping found for dataelement: eventrainin
2022-01-13 23:02:04.374  - debug: sainlogic.0 (5930) No mapping found for dataelement: hourlyrainin
2022-01-13 23:02:04.375  - debug: sainlogic.0 (5930) Extracting value for weather.current.dailyrain(dailyrainin)
2022-01-13 23:02:04.376  - debug: sainlogic.0 (5930) Extracting value for weather.current.weeklyrain(weeklyrainin)
2022-01-13 23:02:04.377  - debug: sainlogic.0 (5930) Extracting value for weather.current.monthlyrain(monthlyrainin)
2022-01-13 23:02:04.378  - debug: sainlogic.0 (5930) No mapping found for dataelement: totalrainin
2022-01-13 23:02:04.379  - debug: sainlogic.0 (5930) Extracting value for weather.current.solarradiation(solarradiation)
2022-01-13 23:02:04.381  - debug: sainlogic.0 (5930) Extracting value for weather.current.uvi(uv)
2022-01-13 23:02:04.382  - debug: sainlogic.0 (5930) No mapping found for dataelement: wh65batt
2022-01-13 23:02:04.384  - debug: sainlogic.0 (5930) Extracting value for info.freq(freq)
2022-01-13 23:02:04.385  - debug: sainlogic.0 (5930) Extracting value for info.model(model)
2022-01-13 23:02:04.387  - debug: sainlogic.0 (5930) Setting value from data for info.softwaretype to EasyWeatherV1.5.9
2022-01-13 23:02:04.393  - debug: sainlogic.0 (5930) Setting value from data for weather.current.indoortemp to 22
2022-01-13 23:02:04.394  - debug: sainlogic.0 (5930) Target for indoortemp unit is set: °C, using conversion: null
2022-01-13 23:02:04.395  - debug: sainlogic.0 (5930) Setting value from data for weather.current.indoorhumidity to 52
2022-01-13 23:02:04.396  - debug: sainlogic.0 (5930) Setting value from data for weather.current.pressurerel to 1023
2022-01-13 23:02:04.397  - debug: sainlogic.0 (5930) Target for pressurerel unit is set: hPa, using conversion: null
2022-01-13 23:02:04.399  - debug: sainlogic.0 (5930) Setting value from data for weather.current.pressureabs to 1033.6
2022-01-13 23:02:04.400  - debug: sainlogic.0 (5930) Target for pressureabs unit is set: hPa, using conversion: null
2022-01-13 23:02:04.401  - debug: sainlogic.0 (5930) Setting value from data for weather.current.outdoortemp to 1.4
2022-01-13 23:02:04.402  - debug: sainlogic.0 (5930) Target for outdoortemp unit is set: °C, using conversion: null
2022-01-13 23:02:04.403  - debug: sainlogic.0 (5930) Setting value from data for weather.current.outdoorhumidity to 93
2022-01-13 23:02:04.403  - debug: sainlogic.0 (5930) Setting value from data for weather.current.winddir to 326
2022-01-13 23:02:04.404  - debug: sainlogic.0 (5930) Setting value from data for weather.current.windspeed to 2.9
2022-01-13 23:02:04.405  - debug: sainlogic.0 (5930) Target for windspeed unit is set: km/h, using conversion: null
2022-01-13 23:02:04.406  - debug: sainlogic.0 (5930) Setting value from data for weather.current.windgustspeed to 5.5
2022-01-13 23:02:04.407  - debug: sainlogic.0 (5930) Target for windgustspeed unit is set: km/h, using conversion: null
2022-01-13 23:02:04.409  - debug: sainlogic.0 (5930) Setting value from data for weather.current.maxdailygust to 16.6
2022-01-13 23:02:04.410  - debug: sainlogic.0 (5930) Target for maxdailygust unit is set: km/h, using conversion: null
2022-01-13 23:02:04.411  - debug: sainlogic.0 (5930) Setting value from data for weather.current.rain to 0
[... cut ...] 0.000","hourlyrainin":"0.000","dailyrainin":"0.000","weeklyrainin":"0.390","monthlyrainin":"1.720","totalrainin":"5.370","solarradiation":"0.00","uv":"0","wh65batt":"0","freq":"868M","model":"WS2900_V2.01.10"}
2022-01-13 23:02:04.500  - info: sainlogic.0 (5930) Unit changed for sainlogic.0.weather.current.solarradiation from lx to W/m2, updating data point
2022-01-13 23:02:04.515  - debug: sainlogic.0 (5930) Target for solarradiation unit is set: W/m2, using conversion: roundTo(x / 126.7, 1)
```

phifogg, 2022-01-14:
Hi Sneaker,
danke für das Log.
Deine Wetterstation sendet nur diese Werte, sieht man in der Zeile 'Listener body is':

PASSKEY=CD4199C743F163BA5924C84E66579702
stationtype=EasyWeatherV1.5.9
dateutc=2022-01-13+22:02:02
tempinf=71.6
humidityin=52
baromrelin=30.213
baromabsin=30.526
tempf=34.5
humidity=93
winddir=326
windspeedmph=1.8
windgustmph=3.4
maxdailygust=10.3
rainratein=0.000
eventrainin=0.000
hourlyrainin=0.000
dailyrainin=0.000
weeklyrainin=0.390
monthlyrainin=1.720
totalrainin=5.370
solarradiation=0.00
uv=0
wh65batt=0
freq=868M
model=WS2900_V2.01.10

Kann es sein dass Du die Wetterstation noch via Ecowitt Protokoll laufen lässt? Ich glaube das schickt weniger Werte als Wunderground.

Sneaker29, 2022-01-14:
Hi phifogg.
Ja, bei Customized Server war noch das Ecowitt Protokoll ausgewählt. Ich habe das jetzt mal umgestellt auf Wunderground, erhalte aber keine Werte.

Wenn ich "Station ID" und "Station Key" leer lasse, bleiben alle Werte bei (null)

```
2022-01-14 11:35:53.890 - debug: sainlogic.0 (18440) Redis Objects: Use Redis connection: 127.0.0.1:9001
2022-01-14 11:35:53.950 - debug: sainlogic.0 (18440) Objects client ready ... initialize now
2022-01-14 11:35:53.953 - debug: sainlogic.0 (18440) Objects create System PubSub Client
2022-01-14 11:35:53.955 - debug: sainlogic.0 (18440) Objects create User PubSub Client
2022-01-14 11:35:53.956 - debug: sainlogic.0 (18440) Objects client initialize lua scripts
2022-01-14 11:35:53.994 - debug: sainlogic.0 (18440) Objects connected to redis: 127.0.0.1:9001
2022-01-14 11:35:54.003 - debug: sainlogic.0 (18440) objectDB connected
2022-01-14 11:35:54.006 - debug: sainlogic.0 (18440) Redis States: Use Redis connection: 127.0.0.1:9000
2022-01-14 11:35:54.069 - debug: sainlogic.0 (18440) States create System PubSub Client
2022-01-14 11:35:54.070 - debug: sainlogic.0 (18440) States create User PubSub Client
2022-01-14 11:35:54.191 - debug: sainlogic.0 (18440) States connected to redis: 127.0.0.1:9000
2022-01-14 11:35:54.193 - debug: sainlogic.0 (18440) statesDB connected
2022-01-14 11:35:56.126 - info: sainlogic.0 (18440) starting. Version 0.7.3 in /opt/iobroker/node_modules/iobroker.sainlogic, node: v12.18.3, js-controller: 3.3.22
2022-01-14 11:35:56.160 - debug: sainlogic.0 (18440) Listener IP: 0.0.0.0
2022-01-14 11:35:56.161 - debug: sainlogic.0 (18440) Listener port: 45000
2022-01-14 11:35:56.161 - debug: sainlogic.0 (18440) Listener path: /weatherstation/updateweatherstation.php
2022-01-14 11:35:56.162 - debug: sainlogic.0 (18440) Listener protocol: wunderground
2022-01-14 11:35:56.163 - debug: sainlogic.0 (18440) Forward URL:
2022-01-14 11:35:56.164 - info: sainlogic.0 (18440) Starting Listener
2022-01-14 11:35:56.241 - info: sainlogic.0 (18440) Unit changed for sainlogic.0.weather.current.solarradiation from lx to W/m2, updating data point
2022-01-14 11:35:56.244 - info: sainlogic.0 (18440) Unit changed for sainlogic.0.weather.maxvalues.absolut.solarradiation from lx to W/m2, updating data point
2022-01-14 11:35:56.246 - info: sainlogic.0 (18440) Unit changed for sainlogic.0.weather.maxvalues.daily.solarradiation from lx to W/m2, updating data point
2022-01-14 11:35:56.276 - debug: sainlogic.0 (18440) Target for solarradiation unit is set: W/m2, using conversion: roundTo(x / 126.7, 1)
2022-01-14 11:35:56.286 - debug: sainlogic.0 (18440) Target for solarradiation unit is set: W/m2, using conversion: roundTo(x / 126.7, 1)
2022-01-14 11:35:56.288 - debug: sainlogic.0 (18440) Target for solarradiation unit is set: W/m2, using conversion: roundTo(x / 126.7, 1)
```

Wenn ich allerdings "Station ID" und "Station Key" ausfülle, bekomme ich alle 60 Sekunden eine Meldung "Listener received illegal request" und dahinter stehen dann die richtigen Werte, werden aber nicht in die Datenpunkte eingetragen. ID und Passwort habe ich im Protokoll anonymisiert:

```
2022-01-14 11:39:13.186 - debug: sainlogic.0 (18474) Redis Objects: Use Redis connection: 127.0.0.1:9001
2022-01-14 11:39:13.254 - debug: sainlogic.0 (18474) Objects client ready ... initialize now
2022-01-14 11:39:13.274 - debug: sainlogic.0 (18474) Objects create System PubSub Client
2022-01-14 11:39:13.276 - debug: sainlogic.0 (18474) Objects create User PubSub Client
2022-01-14 11:39:13.278 - debug: sainlogic.0 (18474) Objects client initialize lua scripts
2022-01-14 11:39:13.317 - debug: sainlogic.0 (18474) Objects connected to redis: 127.0.0.1:9001
2022-01-14 11:39:13.327 - debug: sainlogic.0 (18474) objectDB connected
2022-01-14 11:39:13.330 - debug: sainlogic.0 (18474) Redis States: Use Redis connection: 127.0.0.1:9000
2022-01-14 11:39:13.350 - debug: sainlogic.0 (18474) States create System PubSub Client
2022-01-14 11:39:13.351 - debug: sainlogic.0 (18474) States create User PubSub Client
2022-01-14 11:39:13.430 - debug: sainlogic.0 (18474) States connected to redis: 127.0.0.1:9000
2022-01-14 11:39:13.431 - debug: sainlogic.0 (18474) statesDB connected
2022-01-14 11:39:18.583 - info: sainlogic.0 (18474) starting. Version 0.7.3 in /opt/iobroker/node_modules/iobroker.sainlogic, node: v12.18.3, js-controller: 3.3.22
2022-01-14 11:39:18.623 - debug: sainlogic.0 (18474) Listener IP: 0.0.0.0
2022-01-14 11:39:18.623 - debug: sainlogic.0 (18474) Listener port: 45000
2022-01-14 11:39:18.624 - debug: sainlogic.0 (18474) Listener path: /weatherstation/updateweatherstation.php
2022-01-14 11:39:18.624 - debug: sainlogic.0 (18474) Listener protocol: wunderground
2022-01-14 11:39:18.625 - debug: sainlogic.0 (18474) Forward URL:
2022-01-14 11:39:18.625 - info: sainlogic.0 (18474) Starting Listener
2022-01-14 11:39:18.688 - info: sainlogic.0 (18474) Unit changed for sainlogic.0.weather.current.solarradiation from lx to W/m2, updating data point
2022-01-14 11:39:18.691 - info: sainlogic.0 (18474) Unit changed for sainlogic.0.weather.maxvalues.absolut.solarradiation from lx to W/m2, updating data point
2022-01-14 11:39:18.693 - info: sainlogic.0 (18474) Unit changed for sainlogic.0.weather.maxvalues.daily.solarradiation from lx to W/m2, updating data point
2022-01-14 11:39:18.717 - debug: sainlogic.0 (18474) Target for solarradiation unit is set: W/m2, using conversion: roundTo(x / 126.7, 1)
[... cut ...]
2022-01-14 11:40:12.466 - warn: sainlogic.0 (18474) Listener received illegal request: /weatherstation/updateweatherstation.phpID=***&PASSWORD=***&indoortempf=71.6&tempf=35.4&dewptf=33.6&windchillf=35.4&indoorhumidity=52&humidity=94&windspeedmph=1.3&windgustmph=2.2&winddir=301&absbaromin=30.473&baromin=30.160&rainin=0.000&dailyrainin=0.000&weeklyrainin=0.390&monthlyrainin=1.720&solarradiation=28.99&UV=0&dateutc=2022-01-14%2010:40:10&softwaretype=EasyWeatherV1.6.1&action=updateraw&realtime=1&rtfreq=5
```

phifogg, 2022-01-14:
Hi Sneaker,
kann sein das ich station ID anschaue, eigentlich ist der Wert egal.
Es scheint noch ein Fehler in der URL zu sein, hier fehlt ein '?'.

/weatherstation/updateweatherstation.php**?**ID=...

Schau mal ob das in der Station eingetragen ist.

Sneaker29, 2022-01-14:
Hi phifogg,

in der URL ist weder in der App noch im Adapter ein ? eingetragen. Mit dem Ecowitt-Protokoll ging es auch ohne das ?, nur halt mit den 4 fehlenden Werten, die Ecowitt anscheinend nicht überträgt.

[image]

[image]

Ich habe dann testweise mal in der App und im Adapter ein ? an die URL angehängt, das ändert aber nichts. Ohne StationID kommen keine Werte und mit StationID kommt "Listener received illegal request".

```
2022-01-14 13:54:35.313  - debug: sainlogic.0 (20630) Redis Objects: Use Redis connection: 127.0.0.1:9001
--
2022-01-14 13:54:35.397  - debug: sainlogic.0 (20630) Objects client ready ... initialize now
2022-01-14 13:54:35.449  - debug: sainlogic.0 (20630) Objects create System PubSub Client
2022-01-14 13:54:35.451  - debug: sainlogic.0 (20630) Objects create User PubSub Client
2022-01-14 13:54:35.452  - debug: sainlogic.0 (20630) Objects client initialize lua scripts
2022-01-14 13:54:35.487  - debug: sainlogic.0 (20630) Objects connected to redis: 127.0.0.1:9001
2022-01-14 13:54:35.498  - debug: sainlogic.0 (20630) objectDB connected
2022-01-14 13:54:35.501  - debug: sainlogic.0 (20630) Redis States: Use Redis connection: 127.0.0.1:9000
2022-01-14 13:54:36.469  - debug: sainlogic.0 (20630) States create System PubSub Client
2022-01-14 13:54:36.472  - debug: sainlogic.0 (20630) States create User PubSub Client
2022-01-14 13:54:36.750  - debug: sainlogic.0 (20630) States connected to redis: 127.0.0.1:9000
2022-01-14 13:54:36.752  - debug: sainlogic.0 (20630) statesDB connected
2022-01-14 13:54:37.924  - info:  sainlogic.0 (20630) starting. Version 0.7.3 in  /opt/iobroker/node_modules/iobroker.sainlogic, node: v12.18.3,  js-controller: 3.3.22
2022-01-14 13:54:37.972  - debug: sainlogic.0 (20630) Listener IP: 0.0.0.0
2022-01-14 13:54:37.973  - debug: sainlogic.0 (20630) Listener port: 45000
2022-01-14 13:54:37.973  - debug: sainlogic.0 (20630) Listener path: /weatherstation/updateweatherstation.php?
2022-01-14 13:54:37.974  - debug: sainlogic.0 (20630) Listener protocol: wunderground
2022-01-14 13:54:37.975  - debug: sainlogic.0 (20630) Forward URL:
2022-01-14 13:54:37.977  - info: sainlogic.0 (20630) Starting Listener
2022-01-14 13:54:38.087  - info: sainlogic.0 (20630) Unit changed for sainlogic.0.weather.current.solarradiation from lx to W/m2, updating data point
2022-01-14 13:54:38.093  - info: sainlogic.0 (20630) Unit changed for sainlogic.0.weather.maxvalues.absolut.solarradiation from lx to W/m2, updating data point
2022-01-14 13:54:38.095  - info: sainlogic.0 (20630) Unit changed for sainlogic.0.weather.maxvalues.daily.solarradiation from lx to W/m2, updating data point
2022-01-14 13:54:38.128  - debug: sainlogic.0 (20630) Target for solarradiation unit is set: W/m2, using conversion: roundTo(x / 126.7, 1)
2022-01-14 13:54:38.159  - debug: sainlogic.0 (20630) Target for solarradiation unit is set: W/m2, using conversion: roundTo(x / 126.7, 1)
2022-01-14 13:54:38.161  - debug: sainlogic.0 (20630) Target for solarradiation unit is set: W/m2, using conversion: roundTo(x / 126.7, 1)
2022-01-14 13:55:09.734  - warn:  sainlogic.0 (20630) Listener received illegal request:  /weatherstation/updateweatherstation.php?ID=stationid&PASSWORD=password&indoortempf=71.6&tempf=37.4&dewptf=35.6&windchillf=37.4&indoorhumidity=53&humidity=94&windspeedmph=1.6&windgustmph=2.2&winddir=290&absbaromin=30.449&baromin=30.136&rainin=0.000&dailyrainin=0.000&weeklyrainin=0.390&monthlyrainin=1.720&solarradiation=24.22&UV=0&dateutc=2022-01-14%2012:55:06&softwaretype=EasyWeatherV1.6.1&action=updateraw&realtime=1&rtfreq=5
```

phifogg, 2022-01-14:
Im Broker sieht das gut aus, mach in der Station mal bitte ein '?' hinten dran. Dann sollte es gehen.

Sneaker29, 2022-01-15:
Ja tatsächlich, es lag an dem fehlenden ? in der WS View App.
Max/Min-Werte scheint die Sainlogic WS3500 nicht zu übertragen, die Felder bleiben leer. Auch sogar maxwindgust.

Es scheint aber noch einen Fehler zu geben bei Solarradiation. Wenn ich die Einheit auf lx lasse, wird der Luxwert übertragen. Stelle ich aber auf das bei Weatherunderground eigentlich übliche w/m² um, zeigt er nur noch 0 an.

closed 2022-01-15 by Sneaker29

Pointer-de, 2022-06-16:
Moin.
Ich möchte den Thread hier kurz aufleben lassen. Habe seit kurzem eine Sainlogic WS3500Pro welche soweit auch problemlos läuft. allerdings wird mir der Solarradiation-Wert in W/m² nicht angezeigt, obwohl er laut LOG von der Wetterstation übertragen wird.
Muss da noch irgendwo was eingestellt werden? Wäre für Hilfe sehr dankbar.
In der Anlage und im Adapter ist w/m² eingetragen, verwendet wird V0.8.2

Anbei die Log einer Datenübertragung und der passende Screenshot. 
[image]
[image]
[SaonlogicAdapter-Log single.txt](https://github.com/phifogg/ioBroker.sainlogic/files/8918101/SaonlogicAdapter-Log.single.txt)

phifogg, 2022-06-18:
Lt. Log wird der Wert auch gesetzt:
```2022-06-16 12:23:22.381 - debug: sainlogic.0 (4575) Setting value from data for weather.current.solarradiation to 92755
```
Tauchen im Log ausserhalb sainlogic noch meldungen auf? Kennt er ggf. die Einheit gar nicht?

Pointer-de, 2022-06-18:
```
2022-06-16 12:23:22.381 - debug: sainlogic.0 (4575) Setting value from data for weather.current.solarradiation to 92755
2022-06-16 12:23:22.382 - debug: sainlogic.0 (4575) Target for solarradiation unit is set: W/m2, using conversion: roundTo(x / 126.7, 1)
```
Hier würde die Rechnung ja auch passen mit 92755/126,7=732,08 - das entspricht ja exakt dem übertragenen Wert und auch dem, der auf dem Sainlogic-LCD angezeigt wird

```
...
2022-06-16 12:23:22.442 - info: sainlogic.0 (4575) Unit changed for sainlogic.0.weather.current.solarradiation from lx to W/m2, updating data point
2022-06-16 12:23:22.455 - debug: sainlogic.0 (4575) Target for solarradiation unit is set: W/m2, using conversion: roundTo(x / 126.7, 1)
```
Aber was macht er hier nochmal?

Ansonsten ist die Log frei von Fehlern. Umrechnung passt auch, nur eben die "Ausgabe" des Werts wird nicht vorgenommen.