phifogg/ioBroker.sainlogic #109: Iobroker objects will not be updated after update to 0.8.0

onki2, 2022-01-13:
I have installed the sainlogic adapter a few days ago in combination with my weather station running the FW WH2650A_V1.6.8. All was woriking and i smoothly got all the values. 

Today i updated the sainlogic adapter to 0.8.0, and after that the objects will no longer be updated, even if i got valid data from the weather station:

`2022-01-13 14:50:29.526 - debug: sainlogic.0 (5150) Listener received WU update: {"ID":"123","PASSWORD":"12345","tempf":"69.3","humidity":"32","dewptf":"38.1","windchillf":"69.3","winddir":"83","windspeedmph":"0.00","windgustmph":"0.00","rainin":"0.000","dailyrainin":"0.031","weeklyrainin":"0.031","monthlyrainin":"0.031","yearlyrainin":"0.031","solarradiation":"0.00","UV":"0","indoortempf":"70.0","indoorhumidity":"33","baromin":"30.499","lowbatt":"0","dateutc":"now","softwaretype":"WH2650A_V1.6.8","action":"updateraw","realtime":"1","rtfreq":"5"}
2022-01-13 14:50:29.529 - debug: sainlogic.0 (5150) Extracting value for weather.maxvalues.daily.UVraw_time(ID)
2022-01-13 14:50:29.531 - debug: sainlogic.0 (5150) Extracting value for weather.maxvalues.daily.UVraw_time(PASSWORD)
2022-01-13 14:50:29.532 - debug: sainlogic.0 (5150) Extracting value for weather.maxvalues.daily.UVraw_time(tempf)
2022-01-13 14:50:29.533 - debug: sainlogic.0 (5150) Extracting value for weather.maxvalues.daily.UVraw_time(humidity)
2022-01-13 14:50:29.535 - debug: sainlogic.0 (5150) Extracting value for weather.maxvalues.daily.UVraw_time(dewptf)
2022-01-13 14:50:29.536 - debug: sainlogic.0 (5150) Extracting value for weather.maxvalues.daily.UVraw_time(windchillf)
2022-01-13 14:50:29.537 - debug: sainlogic.0 (5150) Extracting value for weather.maxvalues.daily.UVraw_time(winddir)
2022-01-13 14:50:29.538 - debug: sainlogic.0 (5150) Extracting value for weather.maxvalues.daily.UVraw_time(windspeedmph)
2022-01-13 14:50:29.539 - debug: sainlogic.0 (5150) Extracting value for weather.maxvalues.daily.UVraw_time(windgustmph)
2022-01-13 14:50:29.541 - debug: sainlogic.0 (5150) Extracting value for weather.maxvalues.daily.UVraw_time(rainin)
2022-01-13 14:50:29.542 - debug: sainlogic.0 (5150) Extracting value for weather.maxvalues.daily.UVraw_time(dailyrainin)
2022-01-13 14:50:29.543 - debug: sainlogic.0 (5150) Extracting value for weather.maxvalues.daily.UVraw_time(weeklyrainin)
2022-01-13 14:50:29.544 - debug: sainlogic.0 (5150) Extracting value for weather.maxvalues.daily.UVraw_time(monthlyrainin)
2022-01-13 14:50:29.551 - debug: sainlogic.0 (5150) Extracting value for weather.maxvalues.daily.UVraw_time(yearlyrainin)
2022-01-13 14:50:29.553 - debug: sainlogic.0 (5150) Extracting value for weather.maxvalues.daily.UVraw_time(solarradiation)
2022-01-13 14:50:29.557 - debug: sainlogic.0 (5150) Extracting value for weather.maxvalues.daily.UVraw_time(UV)
2022-01-13 14:50:29.559 - debug: sainlogic.0 (5150) Extracting value for weather.maxvalues.daily.UVraw_time(indoortempf)
2022-01-13 14:50:29.560 - debug: sainlogic.0 (5150) Extracting value for weather.maxvalues.daily.UVraw_time(indoorhumidity)
2022-01-13 14:50:29.562 - debug: sainlogic.0 (5150) Extracting value for weather.maxvalues.daily.UVraw_time(baromin)
2022-01-13 14:50:29.563 - debug: sainlogic.0 (5150) Extracting value for weather.maxvalues.daily.UVraw_time(lowbatt)
2022-01-13 14:50:29.564 - debug: sainlogic.0 (5150) Extracting value for weather.maxvalues.daily.UVraw_time(dateutc)
2022-01-13 14:50:29.566 - debug: sainlogic.0 (5150) Extracting value for info.softwaretype(softwaretype)
2022-01-13 14:50:29.567 - debug: sainlogic.0 (5150) Extracting value for weather.maxvalues.daily.UVraw_time(action)
2022-01-13 14:50:29.569 - debug: sainlogic.0 (5150) Extracting value for weather.maxvalues.daily.UVraw_time(realtime)
2022-01-13 14:50:29.571 - debug: sainlogic.0 (5150) Extracting value for weather.maxvalues.daily.UVraw_time(rtfreq)
2022-01-13 14:50:29.572 - debug: sainlogic.0 (5150) Extracting value for info.last_listener_update(last_listener_update)
2022-01-13 14:50:29.574 - debug: sainlogic.0 (5150) Setting value from data for weather.maxvalues.daily.UVraw_time to 5
2022-01-13 14:50:29.575 - debug: sainlogic.0 (5150) Setting value from data for info.softwaretype to WH2650A_V1.6.8
2022-01-13 14:50:29.576 - debug: sainlogic.0 (5150) Setting value from data for info.last_listener_update to {"ID":"123","PASSWORD":"12345","tempf":"69.3","humidity":"32","dewptf":"38.1","windchillf":"69.3","winddir":"83","windspeedmph":"0.00","windgustmph":"0.00","rainin":"0.000","dailyrainin":"0.031","weeklyrainin":"0.031","monthlyrainin":"0.031","yearlyrainin":"0.031","solarradiation":"0.00","UV":"0","indoortempf":"70.0","indoorhumidity":"33","baromin":"30.499","lowbatt":"0","dateutc":"now","softwaretype":"WH2650A_V1.6.8","action":"updateraw","realtime":"1","rtfreq":"5"}`

Only objects are up to date are the the "Info" section:
[image]

lhoem, 2022-01-13:
Hello,

the same problem.  No values since update 0.8.0. I delete the object tree, but no change... 

Regards 
lhoem

onki2, 2022-01-13:
I did a further test, and downgraded the adapter to version 0.7.3. After that it is working again.

phifogg, 2022-01-13:
Thanks for letting me know as well as the debug log. I'll check into this tomorrow and try to get an update ready.

phifogg, 2022-01-14:
0.8.1 has just been released and this should be fixed again.

Thanks for reporting.

onki2, 2022-01-15:
I can confim that it is working now with 0.8.1. 
Thanks for the quick BugFix.

closed 2022-01-17 by phifogg