phifogg/ioBroker.sainlogic #112: Iobroker objects will not be updated

3PO, 2022-01-27:
Hi,
it seems, that saintlogis does not update the objects:

[image]

saintlogic is version 0.8.1
Weatherstation is a WH2650A

Log:
```

2022-01-27 12:06:38.337  - info: host.iobroker "system.adapter.sainlogic.0" enabled
2022-01-27 12:06:38.498  - info: host.iobroker instance system.adapter.sainlogic.0 started with pid 41801
2022-01-27 12:06:40.218  - info:  sainlogic.0 (41801) starting. Version 0.8.1 in  /opt/iobroker/node_modules/iobroker.sainlogic, node: v17.4.0,  js-controller: 3.3.22
2022-01-27 12:06:40.269  - info: sainlogic.0 (41801) Starting Scheduler
2022-01-27 12:06:40.271  - info: sainlogic.0 (41801) Scheduler call ws_getfirmware activated
2022-01-27 12:06:40.273  - info: sainlogic.0 (41801) Scheduler call ws_getcurrent activated
2022-01-27 12:06:40.273  - info: sainlogic.0 (41801) Scheduler call ws_getmax activated
2022-01-27 12:06:40.275  - info: sainlogic.0 (41801) Scheduler call ws_getmin activated
2022-01-27 12:06:40.275  - info: sainlogic.0 (41801) Scheduler call ws_getmaxdaily activated
2022-01-27 12:06:40.276  - info: sainlogic.0 (41801) Scheduler call ws_getmindaily activated
2022-01-27 12:06:40.282  - info: sainlogic.0 (41801) Starting Listener
2022-01-27 12:06:55.281  - info: sainlogic.0 (41801) Scheduler pulling for new data
2022-01-27 12:06:55.338  - error: sainlogic.0 (41801) Scheduler received data for unkown command 84 subcommand 49
2022-01-27 12:07:10.281  - info: sainlogic.0 (41801) Scheduler pulling for new data
2022-01-27 12:07:10.310  - error: sainlogic.0 (41801) Scheduler received data for unkown command 84 subcommand 49
2022-01-27 12:07:25.282  - info: sainlogic.0 (41801) Scheduler pulling for new data
2022-01-27 12:07:25.304  - error: sainlogic.0 (41801) Scheduler received data for unkown command 84 subcommand 49
2022-01-27 12:07:40.283  - info: sainlogic.0 (41801) Scheduler pulling for new data
2022-01-27 12:07:40.304  - error: sainlogic.0 (41801) Scheduler received data for unkown command 84 subcommand 49
```

Headhunter70, 2022-01-29:
Hi there same issue here, switched back to 0.7.3.

phifogg, 2022-03-28:
oops, didn't see your issue before. Sorry, must have escaped my inbox.
Can you put the adapter into debug mode?
With the WH2650A it may be that only the listener mode is supported. The weather station probably expects some different commands to answer properly.

Headhunter70, 2022-03-29:
Hi there,

The problem disappeared after upgrade to v 0.8.1. I have however no values for max and min. Is that related to Listener mode ?

phifogg, 2022-03-29:
That is correct. Listener mode does not return min/max values unfortunately.

I could think about doing the min/max logic inside the adapter to replicate kind of what the station delivers.

Headhunter70, 2022-03-29:
Not a big issue for me, I am logging the data i am interested in anyhow. Great Job !

closed 2022-05-30 by phifogg