phifogg/ioBroker.sainlogic #116: UVraw value greater than max

sansonifabio, 2022-03-25:
**value greater than max**  

State value to set for "sainlogic.0.weather.current.UVraw" has value "**2094**" greater than max "**2000**"

 - Adapter version: 0.8.1
 - JS-Controller version: 4.0.21
 - Node version: v14.19.1
 - Operating system: linux

Jetzt wo das Wetter so schön ist laufen die Werte für UV raw über.

phifogg, 2022-03-25:
oh.. hoppla. na das lässt sich doch schnell korrigieren ;D Danke!

sansonifabio, 2022-03-25:
Gerne ;-)

phifogg, 2022-03-28:
Auf dem 'dev' Branch ist das behoben, ich scheitere nur leider mal wieder am publishing der neuen Version. Sobald ich hier eine Info bekomme was falsch ist gibt es ein Update.

sansonifabio, 2022-03-29:
> Auf dem 'dev' Branch ist das behoben, ich scheitere nur leider mal wieder am publishing der neuen Version. Sobald ich hier eine Info bekomme was falsch ist gibt es ein Update.

Vielen Dank auch ich kann warten kein Problem.;-)

phifogg, 2022-03-30:
Heute ging der publish. Sollte in Kürze in ioBroker als 0.8.2 auftauchen.

closed 2022-03-30 by phifogg

sansonifabio, 2022-03-30:
> Heute ging der publish. Sollte in Kürze in ioBroker als 0.8.2 auftauchen.

Dann noch mal vielen Dank,  ich sag dir dann Bescheid wenn der Adapter aktualisiert ist.

sansonifabio, 2022-03-31:
So wurde im ioBroker angezeigt und Update verlief ohne Probleme, jetzt aber ist das Wetter wieder schlechter geworden.;-)
Ich denke aber das alles läuft. Ich Danke dir

phifogg, 2022-03-31:
Für das Wetter kann ich leider nichts ;D

sansonifabio, 2022-04-02:
Es tut mir leid aber der Fehler ist wieder da,  mach ich das was falsch?

sainlogic.0 2022-04-02 13:56:02.260 warn	
State value to set for "sainlogic.0.weather.current.UVraw" has value "2036" greater than max "2000"