phifogg/ioBroker.sainlogic #118: tate value to set for "sainlogic.0.weather.current.UVraw" has value "2036" greater than max "2000"

sansonifabio, 2022-04-03:
Es tut mir leid aber der Fehler ist wieder da.

sainlogic.0 2022-04-02 13:56:02.260 warn
State value to set for "sainlogic.0.weather.current.UVraw" has value "2036" greater than max "2000"
Kann es sein das du evtl. einen übersehen hast es gibt ja mehrere UV Raw.

es gibt ja einen unter sainlogic.0.weather.current und Max. UV (raw) (absolut)

phifogg, 2022-04-03:
Nein, liegt nicht an Dir. Ich konnte es nachvollziehen. Weil es den Datenpunkt schon gibt wird der Max-Wert nicht korrigiert. Doof, muss ich mal schauen ob/wie ich das einbaue.

Ein einfacher work-around ist den Datenpunkt zu löschen. Beim nächsten Lauf legt er ihn dann auch korrekt an.

sansonifabio, 2022-04-03:
... so habe die Datenpunkte  gelöscht und wurde von Adapter neu angelegt aber der maxValue absolut.UVraw bringt noch das:
State value to set for "sainlogic.0.weather.maxvalues.absolut.UVraw" has value "65535" greater than max "4000"

Dir fällt bestimmt was ein;-)

phifogg, 2022-04-03:
65535??? Was liefert der denn für Werte bei Dir? Ich hab nun auf 4000 erhöht, klar dass das nicht reinpasst :)
65k klingt aber fast nach einer Fehlermeldung.

sansonifabio, 2022-04-03:
- [ 
[image]
]

phifogg, 2022-04-03:
well. then... ich mach nochmals ein update fertig mit 100'000 oder so. Zwischenzeitlich kannst das mit dem State selbst von hand so eintragen. Sollte gehen.

sansonifabio, 2022-04-03:
Vielen dank dir ;-)

phifogg, 2022-05-30:
Fixed in 0.9.0 - make sure to delete the state before restarting the adapter.

closed 2022-05-30 by phifogg

sansonifabio, 2022-05-31:
Everything went super. I thank you for your efforts.

Alles super gelaufen. ich Danke dir für deine Mühen.;-)
Grüße
Fabio