phifogg/ioBroker.sainlogic #119: Additional EW fields for HP1000SE-PRO and additional sensors

JustJoe21, 2022-05-23:
Hi,

I use this adapter with an HP1000SE-PRO (Froggit) and it works quite well. However, some fields from the EW protocol JSON file are not recognized by the adapter, specifically temperature values (°F) from external temperatur-only sensors:

    "tf_ch1": "66.9",
    "tf_batt1": "1.56",

It would be great, if this prefix "tf_ch#" could be added to the parser so that these fields can also be read.

I have attached the entire JSON file since there are some more fields that might be interesting to add.

Thanks so far!

Best regards,
     
Joe

`
{
    "PASSKEY": "XXXXXXXXXXXXXXXXXXXXXXXXXXXXXX",
    "stationtype": "EasyWeatherV1.6.4",
    "dateutc": "2022-05-23 19:09:11",
    "tempinf": "76.5",
    "humidityin": "56",
    "baromrelin": "29.416",
    "baromabsin": "29.185",
    "tempf": "63.7",
    "humidity": "92",
    "winddir": "101",
    "winddir_avg10m": "85",
    "windspeedmph": "0.9",
    "windspdmph_avg10m": "0.2",
    "windgustmph": "2.2",
    "maxdailygust": "14.8",
    "rainratein": "0.000",
    "eventrainin": "0.280",
    "hourlyrainin": "0.000",
    "dailyrainin": "0.280",
    "weeklyrainin": "0.280",
    "monthlyrainin": "1.728",
    "yearlyrainin": "4.839",
    "solarradiation": "3.43",
    "uv": "0",
    "temp1f": "76.8",
    "humidity1": "53",
    "temp2f": "70.3",
    "humidity2": "54",
    "soilmoisture1": "43",
    "soilmoisture2": "47",
    "soilmoisture3": "56",
    "tf_ch1": "66.9",
    "wh65batt": "0",
    "wh25batt": "0",
    "batt1": "0",
    "batt2": "0",
    "soilbatt1": "1.4",
    "soilbatt2": "1.4",
    "soilbatt3": "1.5",
    "tf_batt1": "1.56",
    "freq": "868M",
    "model": "HP1000SE-PRO_Pro_V1.6.9"
}`

phifogg, 2022-05-30:
Available in 0.9.0 - let me know if it works as I cannot really test it.

phifogg, 2022-06-18:
Sure, feel free to fork the code and check it out.
Closing for now.

closed 2022-06-18 by phifogg