phifogg/ioBroker.sainlogic #120: Dewpoint temperature shows (null)

themuck, 2022-05-24:
I use a WH2650A station, the most thinks work fine but the sainlogic.0.weather.current.dewpointtemp shows (null).

themuck, 2022-05-30:
maybe this help? When i am right there is no Dewpoint Data in the massage? 

`2022-05-30 15:58:35.220 - debug: sainlogic.0 (475999) Listener body is PASSKEY=XYYZ=WH2650A_V1.7.3&runtime=167433&dateutc=2022-05-30+13:58:34&tempinf=67.64&humidityin=57&baromrelin=31.993&baromabsin=29.430&tempf=57.74&humidity=63&winddir=230&windspeedmph=2.01&windgustmph=3.36&maxdailygust=8.05&solarradiation=196.84&uv=1&rainratein=0.000&eventrainin=0.000&hourlyrainin=0.000&dailyrainin=0.000&weeklyrainin=0.051&monthlyrainin=0.409&yearlyrainin=0.409&wh65batt=0&wh25batt=0&freq=868M&model=WH2650A
2022-05-30 15:58:35.221 - debug: sainlogic.0 (475999) No mapping found for dataelement: PASSKEY
2022-05-30 15:58:35.221 - debug: sainlogic.0 (475999) Extracting value for info.softwaretype(stationtype)
2022-05-30 15:58:35.221 - debug: sainlogic.0 (475999) No mapping found for dataelement: runtime
2022-05-30 15:58:35.221 - debug: sainlogic.0 (475999) No mapping found for dataelement: dateutc
2022-05-30 15:58:35.221 - debug: sainlogic.0 (475999) Extracting value for weather.current.indoortemp(tempinf)
2022-05-30 15:58:35.223 - debug: sainlogic.0 (475999) Extracting value for weather.current.indoorhumidity(humidityin)
2022-05-30 15:58:35.223 - debug: sainlogic.0 (475999) Extracting value for weather.current.pressurerel(baromrelin)
2022-05-30 15:58:35.223 - debug: sainlogic.0 (475999) Extracting value for weather.current.pressureabs(baromabsin)
2022-05-30 15:58:35.223 - debug: sainlogic.0 (475999) Extracting value for weather.current.outdoortemp(tempf)
2022-05-30 15:58:35.223 - debug: sainlogic.0 (475999) Extracting value for weather.current.outdoorhumidity(humidity)
2022-05-30 15:58:35.224 - debug: sainlogic.0 (475999) Extracting value for weather.current.winddir(winddir)
2022-05-30 15:58:35.224 - debug: sainlogic.0 (475999) Extracting value for weather.current.windspeed(windspeedmph)
2022-05-30 15:58:35.224 - debug: sainlogic.0 (475999) Extracting value for weather.current.windgustspeed(windgustmph)
2022-05-30 15:58:35.224 - debug: sainlogic.0 (475999) Extracting value for weather.current.maxdailygust(maxdailygust)
2022-05-30 15:58:35.224 - debug: sainlogic.0 (475999) Extracting value for weather.current.solarradiation(solarradiation)
2022-05-30 15:58:35.224 - debug: sainlogic.0 (475999) Extracting value for weather.current.uvi(uv)
2022-05-30 15:58:35.224 - debug: sainlogic.0 (475999) Extracting value for weather.current.rain(rainratein)
2022-05-30 15:58:35.224 - debug: sainlogic.0 (475999) No mapping found for dataelement: eventrainin
2022-05-30 15:58:35.224 - debug: sainlogic.0 (475999) No mapping found for dataelement: hourlyrainin
2022-05-30 15:58:35.225 - debug: sainlogic.0 (475999) Extracting value for weather.current.dailyrain(dailyrainin)
2022-05-30 15:58:35.225 - debug: sainlogic.0 (475999) Extracting value for weather.current.weeklyrain(weeklyrainin)
2022-05-30 15:58:35.225 - debug: sainlogic.0 (475999) Extracting value for weather.current.monthlyrain(monthlyrainin)
2022-05-30 15:58:35.225 - debug: sainlogic.0 (475999) Extracting value for weather.current.yearlyrain(yearlyrainin)
2022-05-30 15:58:35.225 - debug: sainlogic.0 (475999) No mapping found for dataelement: wh65batt
2022-05-30 15:58:35.225 - debug: sainlogic.0 (475999) No mapping found for dataelement: wh25batt
2022-05-30 15:58:35.225 - debug: sainlogic.0 (475999) Extracting value for info.freq(freq)
2022-05-30 15:58:35.225 - debug: sainlogic.0 (475999) Extracting value for info.model(model)
2022-05-30 15:58:35.226 - debug: sainlogic.0 (475999) Setting value from data for info.softwaretype to WH2650A_V1.7.3
2022-05-30 15:58:35.226 - debug: sainlogic.0 (475999) Setting value from data for weather.current.indoortemp to 19.8
2022-05-30 15:58:35.226 - debug: sainlogic.0 (475999) Target for indoortemp unit is set: °C, using conversion: null
2022-05-30 15:58:35.226 - debug: sainlogic.0 (475999) Setting value from data for weather.current.indoorhumidity to 57
2022-05-30 15:58:35.226 - debug: sainlogic.0 (475999) Setting value from data for weather.current.pressurerel to 1083.3
2022-05-30 15:58:35.226 - debug: sainlogic.0 (475999) Target for pressurerel unit is set: hPa, using conversion: null
2022-05-30 15:58:35.226 - debug: sainlogic.0 (475999) Setting value from data for weather.current.pressureabs to 996.5
2022-05-30 15:58:35.226 - debug: sainlogic.0 (475999) Target for pressureabs unit is set: hPa, using conversion: null
2022-05-30 15:58:35.226 - debug: sainlogic.0 (475999) Setting value from data for weather.current.outdoortemp to 14.3
2022-05-30 15:58:35.226 - debug: sainlogic.0 (475999) Target for outdoortemp unit is set: °C, using conversion: null
2022-05-30 15:58:35.226 - debug: sainlogic.0 (475999) Setting value from data for weather.current.outdoorhumidity to 63
2022-05-30 15:58:35.227 - debug: sainlogic.0 (475999) Setting value from data for weather.current.winddir to 230
2022-05-30 15:58:35.227 - debug: sainlogic.0 (475999) Setting value from data for weather.current.windspeed to 3.2
2022-05-30 15:58:35.227 - debug: sainlogic.0 (475999) Target for windspeed unit is set: km/h, using conversion: null
2022-05-30 15:58:35.227 - debug: sainlogic.0 (475999) Setting value from data for weather.current.windgustspeed to 5.4
2022-05-30 15:58:35.227 - debug: sainlogic.0 (475999) Target for windgustspeed unit is set: km/h, using conversion: null
2022-05-30 15:58:35.227 - debug: sainlogic.0 (475999) Setting value from data for weather.current.maxdailygust to 13
[... cut ...]
2022-05-30 15:58:35.228 - debug: sainlogic.0 (475999) Target for monthlyrain unit is set: mm, using conversion: null
2022-05-30 15:58:35.228 - debug: sainlogic.0 (475999) Setting value from data for weather.current.yearlyrain to 10.4
2022-05-30 15:58:35.228 - debug: sainlogic.0 (475999) Target for yearlyrain unit is set: mm, using conversion: null
2022-05-30 15:58:35.228 - debug: sainlogic.0 (475999) Setting value from data for info.freq to 868M
2022-05-30 15:58:35.228 - debug: sainlogic.0 (475999) Setting value from data for info.model to WH2650A
`

phifogg, 2022-06-18:
You are correct, no input data on dewpoint, hence no value.
Sometimes the stations behave differently depening on the protocol used.

Eistee82, 2022-08-28:
I have the same problem that it reports null to values from time to time. Is there a way to check this in the adapter and not to write null?

phifogg, 2023-01-06:
What would be your wanted behaviour? The adapter doesn't get any update on the value and hence doesn't update the states. So whatever the old value was would stay.

phifogg, 2023-05-07:
closing as no update since january. feel free to reopen.

closed 2023-05-07 by phifogg

Eistee82, 2023-05-07:
I am no longer using this adapter because of this