phifogg/ioBroker.sainlogic #121: Wunderground vs EcoWitt

pancetta, 2022-05-28:
**Describe the bug**  
I just got my Froggit DP1500 + moisture sensors and followed the [guide](https://github.com/phifogg/ioBroker.sainlogic#usage) to get the data into my iobroker. Nothing happened in listener mode, nothing in scheduler mode. After changing the protocol to Ecowitt (instead of Wunderground as described in the guide), the data started to show up. 

**To Reproduce**  
Follow the [guide](https://github.com/phifogg/ioBroker.sainlogic#usage) for a Froggit DP1500, iobroker v5.3.8.

**Expected behavior**  
Either data should show up when using Wunderground or the guide needs to mention Ecowitt instead. I'm not fully sure whether this may not be a bug in the DP1500, though.

**Screenshots & Logfiles**  
If applicable, add screenshots and logfiles to help explain your problem.

**Versions:**  
 - Adapter version: 0.8.2
 - JS-Controller version: 4.0.21
 - Node version: 12.22.5
 - Operating system: iobroker v5.3.8

**Additional context**  
Since stuff works now for me, no immediate action is required. However, if this is a bug in the guide or the adapter, this should be fixed or mentioned. In any case, thanks for the great work with this adapter!

phifogg, 2022-05-30:
Thanks for the detailed issue report.
Can you set your settings to WU protocol and run the adapater in debug mode? This would allow me to see if the data is actually transmitted or not. 
I have seen some differences in the data sent by the different stations.

pancetta, 2022-05-30:
The last line also shows issue #122..

pancetta, 2022-05-30:
Yes, that's what I did, presumably. Everything else was the same and I double-checked the other parameters, also when using WU. I guess having a comment in the readme is the easiest way to help here.

closed 2022-05-30 by phifogg

mcm1957, 2022-06-25:
Did you set Station-id and Ststion-Key to ANY non blank value?
My Eurochron did not send any data as long as stations-id and station.key have been empty.
see #130