phifogg/ioBroker.sainlogic #124: Soilmoisture Datenpunkt und neue DP's im GW2000A

pp-git, 2022-06-01:
Hi Phifogg,

irgendwas stimmt mit dem soilmoisture DP nicht,

error im IoBroker:

`State value to set for "sainlogic.0.weather.current.1.soilmoisture" has value "31" greater than max "24"`

sainlogic.0.weather.current.1.soilmoisture wird korrekt mit **31%** dargestellt

JSON raw:

```
{
  "PASSKEY": "XXXXXXXXXX",
  "stationtype": "GW2000A_V2.1.4",
  "runtime": "2424",
  "dateutc": "2022-06-01 11:17:48",
  "tempinf": "80.1",
  "humidityin": "27",
  "baromrelin": "29.964",
  "baromabsin": "29.108",
  "tempf": "73.9",
  "humidity": "36",
  "winddir": "144",
  "windspeedmph": "2.91",
  "windgustmph": "6.04",
  "maxdailygust": "7.38",
  "solarradiation": "272.38",
  "uv": "2",
  "rrain_piezo": "0.000",
  "erain_piezo": "0.000",
  "hrain_piezo": "0.000",
  "drain_piezo": "0.000",
  "wrain_piezo": "0.000",
  "mrain_piezo": "0.000",
  "yrain_piezo": "0.000",
  "ws90cap_volt": "1.6",
  "ws90_ver": "119",
  "soilmoisture1": "31",
  "soilbatt1": "1.6",
  v
  "freq": "868M",
  "model": "GW2000A"
}
```
Kannst Du bitte auch nach den Regen DP's schauen, hier sind die neuen Piezo Bezeichnungen wohl anders als mit dem normalen Sensor.

Außerdem:

"ws90cap_volt": "1.6", ist der eingebaute Photovoltaik-Kondensator
"wh90batt": "3.18", ist die Backupbatterie falls der Kondensator leer ist 

Danke und Grüße
Paul

phifogg, 2022-06-26:
Hm.. die piezo heissen anders, das kann ich schon mappen. In welcher Einheit kommt es da denn? Sind das immernoch inch?

Soilmoisture sollte behoben sein, kann sein dass Du den Datenpunkt löschen musst damit er neu angelegt wird. Maxvalue sollte nun 100 sein.

phifogg, 2023-01-06:
Added to version 0.10.0. Please report back with confirmation or any issues you see.

closed 2023-01-06 by phifogg