phifogg/ioBroker.sainlogic #128: no objects for lightning sensor DP60 (Ecowitt WH57)

Vertigo1206, 2022-06-25:
Hello,

I am using the Froggit DP1500 (WiFi Adapter) with the Sainlogic adapter in iobroker.
Works fine so far.
Now I added a lightning sensor DP60 (Ecowitt WH57) 
In the WSView app I can see the sensor including his values
[image]

But I did not get any new objects in iobroker for this sensor.
Does this adapter support this sensor?

Allready installed & configured the adapter new from scratch.

Versions:
Sainlogic: v0.9.0
iobroker: 4.0.23
Node: 14.19.3
Docker on synology NAS

phifogg, 2022-06-26:
Can you run the adapter in debug mode and attach the log file?
This would allow me to see how the variables are named for these data points.

Vertigo1206, 2022-06-27:
Thanks for the fast response!
Please find attached the log file

I found the elements:

> 2022-06-27 09:32:28.935 - debug: sainlogic.0 (469) No mapping found for dataelement: lightning_num
> 2022-06-27 09:32:28.935 - debug: sainlogic.0 (469) No mapping found for dataelement: lightning
> 2022-06-27 09:32:28.936 - debug: sainlogic.0 (469) No mapping found for dataelement: lightning_time
> 2022-06-27 09:32:28.936 - debug: sainlogic.0 (469) No mapping found for dataelement: wh65batt
> 2022-06-27 09:32:28.936 - debug: sainlogic.0 (469) No mapping found for dataelement: batt1
> 2022-06-27 09:32:28.936 - debug: sainlogic.0 (469) No mapping found for dataelement: wh57batt
> 

I also found some additional values without mapping

> 2022-06-27 09:32:28.934 - debug: sainlogic.0 (469) No mapping found for dataelement: eventrainin
> 2022-06-27 09:32:28.935 - debug: sainlogic.0 (469) No mapping found for dataelement: hourlyrainin
> 2022-06-27 09:32:28.935 - debug: sainlogic.0 (469) No mapping found for dataelement: totalainin

Maybe you can add this objects too?

[debug-sainlogic.log](https://github.com/phifogg/ioBroker.sainlogic/files/8989636/debug-sainlogic.log)

Vertigo1206, 2022-06-29:
Hello,

I tried to adjust the constants.js with the lightning variables lightning, lightning_num & lightning_time
(just the current values, not jet min & max)
But I always geht the lightning_time value on the lightning object

```
{
        id: 'lightning_num',
        channels: [{
            channel: 'weather.current',
            name: 'lightning daily count'
        },{
            channel: 'weather.maxvalues.absolut',
            name: 'lightning daily count (absolut)'
        },{
            channel: 'weather.maxvalues.daily',
            name: 'lightning daily count (daily)'
        }],
        type: 'number',
        unit: '',
        role: 'value',
        wunderground: '^lightning_num',
        ecowitt: '^lightning_num',
        scheduler: 'lightning_num',
        listener_conversion: null,
        scheduler_conversion: null
    },
    {
        id: 'lightning',
        channels: [{
            channel: 'weather.current',
            name: 'lightning distance'
        },{
            channel: 'weather.maxvalues.absolut',
            name: 'Max. lightning distance (absolut)'
        },{
            channel: 'weather.maxvalues.daily',
            name: 'Max. lightning distance (daily)'
        }],
        type: 'number',
        unit: 'km',
        role: 'value',
        min: 0,
        wunderground: null,
        ecowitt: '^lightning',
        scheduler: 'lightning',
        listener_conversion: null,
        scheduler_conversion: null
    },
    {
        id: 'lightning_time',
        channels: [{
            channel: 'weather.current',
            name: 'lightning (time)'
        }],
        type: 'number',
        role: 'value',
        wunderground: null,
        ecowitt: '^lightning_time',
        scheduler: 'lightning_time',
        listener_conversion: null,
        scheduler_conversion: null
    },
```
```
        name: 'Get Current Data',
        configurable: true,
        config_variable: 'ws_getcurrent',
        channel: 'weather.current',
        command: [0xFF, 0xFF, 0x0B, 0x00, 0x06, 0x04, 0x04, 0x19],
        command_int: 11,
        subcommand_int: 4,
        parser: new BinaryParser()
            .endianess('big').seek(7)
            .int16('indoortemp').seek(1)
            .int16('outdoortemp').seek(1)
            .int16('dewpointtemp').seek(1)
            .int16('windchilltemp').seek(1)
            .int16('heatindex').seek(1)
            .uint8('indoorhumidity').seek(1)
            .uint8('outdoorhumidity').seek(1)
            .uint16('pressureabs').seek(1)
            .uint16('pressurerel').seek(1)
            .uint16('winddir').seek(1)
            .uint16('windspeed').seek(1)
            .uint16('windgustspeed').seek(1)
            .uint32('rain').seek(1)
            .uint32('dailyrain').seek(1)
            .uint32('weeklyrain').seek(1)
            .uint32('monthlyrain').seek(1)
            .uint32('yearlyrain').seek(1)
            .uint32('raintotal').seek(1)
            .uint32('solarradiation').seek(1)
            .uint16('UVraw').seek(1)
            .uint16('lightning_num').seek(1)
            .uint16('lightning').seek(1)
            .uint16('lightning_time').seek(1)
            .uint8('uvi')
```
[constants.txt](https://github.com/phifogg/ioBroker.sainlogic/files/9008440/constants.txt)

Cause my programming skills are not so good, what does the "seek(1)" function?

phifogg, 2022-06-29:
I assume you use the ecowitt protocoll, correct?

If so, your matching rule currently is ' ecowitt: '^lightning'' ... the ^ is a regular expression for starts-with. Remove that and it should work.

Vertigo1206, 2022-07-29:
I started from the scratch with modifying.
Still get only 2 values.
Maybe you can help mw with the command/parser section

```
 name: 'Get Current Data',
        configurable: true,
        config_variable: 'ws_getcurrent',
        channel: 'weather.current',
        command: [0xFF, 0xFF, 0x0B, 0x00, 0x06, 0x04, 0x04, 0x19],
        command_int: 11,
        subcommand_int: 4,
        parser: new BinaryParser()
            .endianess('big').seek(7)
            .int16('indoortemp').seek(1)
            .int16('outdoortemp').seek(1)
            .int16('dewpointtemp').seek(1)
            .int16('windchilltemp').seek(1)
            .int16('heatindex').seek(1)
            .uint8('indoorhumidity').seek(1)
            .uint8('outdoorhumidity').seek(1)
            .uint16('pressureabs').seek(1)
            .uint16('pressurerel').seek(1)
            .uint16('winddir').seek(1)
            .uint16('windspeed').seek(1)
            .uint16('windgustspeed').seek(1)
            .uint32('rain').seek(1)
            .uint32('dailyrain').seek(1)
            .uint32('weeklyrain').seek(1)
            .uint32('monthlyrain').seek(1)
            .uint32('yearlyrain').seek(1)
            .uint32('raintotal').seek(1)
            .uint32('solarradiation').seek(1)
            .uint16('UVraw').seek(1)
            .uint16('lightning').seek(1)
            .uint16('lightning_num').seek(1)
            .int16('lightning_time').seek(1)
            .uint8('wh57batt').seek(1)
            .uint8('uvi')
```

```
{
        id: 'lightning_time',
        channels: [{
            channel: 'weather.current',
            name: 'last lightning(time)'
        }],
        type: 'number',
        role: 'value',
        wunderground: null,
        ecowitt: 'lightning_time',
        scheduler: null,
        listener_conversion: null,
        scheduler_conversion: null
    },
    {
        id: 'lightning',
        channels: [{
            channel: 'weather.current',
            name: 'last lightning (distance)'
        }],
        type: 'number',
        unit: 'km',
        role: 'value',
        wunderground: null,
        ecowitt: 'lightning',
        scheduler: null,
        listener_conversion: null,
        scheduler_conversion: null
    },
    {
        id: 'lightning_num',
        channels: [{
            channel: 'weather.current',
            name: 'lightning count (day)'
        }],
        type: 'number',
        unit: '',
        role: 'value',
        wunderground: null,
        ecowitt: 'lightning_num',
        scheduler: null,
        listener_conversion: null,
        scheduler_conversion: null
    },
```

phifogg, 2022-09-11:
I don't think you need anything in the parser section, this would only be for scheduler based data loads. The listener works different. Can you post a debug after you have changed the constants.php? On a first glance your file looks good.

Vertigo1206, 2022-09-15:
Now I startet from the beginning.
Now It get the correct values for Ligthning and Ligthning_Time, but no data for ligthning_num

The listener get the values 
`Listener received EW update: {"PASSKEY":"loxberry","stationtype":"GW1000A_V1.7.5","runtime":"1238539","dateutc":"2022-09-15 16:21:44","tempinf":"76.10","humidityin":"61","baromrelin":"29.763","baromabsin":"27.873","tempf":"60.44","humidity":"99","winddir":"197","windspeedmph":"2.91","windgustmph":"5.82","maxdailygust":"17.22","solarradiation":"12.11","uv":"0","rainratein":"0.000","eventrainin":"2.339","hourlyrainin":"0.039","dailyrainin":"0.969","weeklyrainin":"2.949","monthlyrainin":"3.598","yearlyrainin":"33.374","totalrainin":"33.374","temp1f":"67.28","humidity1":"72","**lightning_num":"10**","lightning":"20","lightning_time":"1663247104","wh65batt":"0","batt1":"0","wh57batt":"4","freq":"868M","model":"GW1000A_Pro"}`

Please find attached the log and the constants.js
[sainlogic.zip](https://github.com/phifogg/ioBroker.sainlogic/files/9577144/sainlogic.zip)

phifogg, 2023-01-06:
Added your states to the latest dev branch for now. They look ok, but your log states:
```
2022-09-15 17:56:19.702  - [34mdebug[39m: sainlogic.0 (647) Extracting value for weather.current.lightning(lightning_num)
```
This means somehohw the count is transferred to the wrong state value. Let's try that again when I release the version.

phifogg, 2023-01-06:
Added to version 0.10.0. - Please report back with this update to see if all states are transferred correctly.

phifogg, 2023-02-06:
no update, assume working. reopen if not.

closed 2023-02-06 by phifogg