phifogg/ioBroker.sainlogic #129: Object sainlogic.0.info.last_listener_update is invalid

mcm1957, 2022-06-25:
**Describe the bug**  

During (inital) startup the follwoing warning ist reported:

sainlogic.0 | 2022-06-25 19:58:37.799 | info | State value to set for "sainlogic.0.info.last_listener_update" has to be type "string" but received type "number"
sainlogic.0 | 2022-06-25 19:58:37.408 | warn | This object will not be created in future versions. Please report this to the developer.
sainlogic.0 | 2022-06-25 19:58:37.408 | warn | Object sainlogic.0.info.last_listener_update is invalid: Default value has to be type "string" but received type "number"

**To Reproduce**  
Steps to reproduce the behavior:
Perform an initail installtion of an instance

**Expected behavior**  
No such warning should occure

**Versions:**  
 - Adapter version: sainlogic 0.9.0
 - JS-Controller version: 4.0.21
 - Node version: 14.19.0
 - Operating system: raspian

phifogg, 2022-06-26:
Can you show the value it is trying to set into listener_update? Debug log should show it.
Based on the code it should be a string value....

mcm1957, 2022-06-26:
The error does NOT occure when new values arrive.

The error occures when starting a new instance and the datapoints are initially created. The message does not occure a second time or during normal operation. Looks like some sort of initialization error or type mismatch during creation of instance.

I've taken a look into you script and io-package.json but I do not see a problem there. 
Maybe the empty string for the object (at io-package.json) is considered numeric - but that would be an surprise.
Anyway:
>Object sainlogic.0.info.last_listener_update is invalid: Default value has to be type "string" but received type "number"<
indicates that the problem occured during creation of the object.
 
I'll try to reproduce at a dev installation to get the exact time of error. But this might last until mid or even end of this week. Sorry.

mcm1957, 2022-07-12:
Sorry for the late reply. I did an installation at a testenvironment (windows ioBroker) today.
The warning was reproduceable with the following steps:

a) Precondition:
----------------
New system, sainlogic was never installed at this system before.

b) I set default loggin to DEBUG at system settings
---------------------------------------------------

c) I installed sainlogic@0.9.0 via admin
----------------------------------------

The following was logged at installation console (admin console window):

$ iobroker add sainlogic auto --host Speedy

NPM version: 6.14.16

Installing iobroker.sainlogic@0.9.0... (System call)
+ iobroker.sainlogic@0.9.0 added 61 packages from 30 contributors and audited 511 packages in 19.5s
45 packages are looking for funding run `npm fund` for details
found 12 vulnerabilities (4 low, 4 moderate, 3 high, 1 critical) run `npm audit fix` to fix them, or `npm audit` for details
host.Speedy install adapter sainlogic
upload [4] sainlogic.admin C:/ioBroker/node_modules/iobroker.sainlogic/admin/admin.d.ts admin.d.ts video/mp2t
upload [3] sainlogic.admin C:/ioBroker/node_modules/iobroker.sainlogic/admin/index_m.html index_m.html text/html
upload [2] sainlogic.admin C:/ioBroker/node_modules/iobroker.sainlogic/admin/sainlogic.png sainlogic.png image/png
upload [1] sainlogic.admin C:/ioBroker/node_modules/iobroker.sainlogic/admin/style.css style.css text/css
upload [0] sainlogic.admin C:/ioBroker/node_modules/iobroker.sainlogic/admin/words.js words.js application/javascript
host.Speedy object system.adapter.sainlogic created/updated
host.Speedy create instance sainlogic
host.Speedy object system.adapter.sainlogic.0.alive created
host.Speedy object system.adapter.sainlogic.0.connected created
host.Speedy object system.adapter.sainlogic.0.compactMode created
host.Speedy object system.adapter.sainlogic.0.cpu created
host.Speedy object system.adapter.sainlogic.0.cputime created
host.Speedy object system.adapter.sainlogic.0.memHeapUsed created
host.Speedy object system.adapter.sainlogic.0.memHeapTotal created
host.Speedy object system.adapter.sainlogic.0.memRss created
host.Speedy object system.adapter.sainlogic.0.uptime created
host.Speedy object system.adapter.sainlogic.0.inputCount created
host.Speedy object system.adapter.sainlogic.0.outputCount created
host.Speedy object system.adapter.sainlogic.0.eventLoopLag created
host.Speedy object system.adapter.sainlogic.0.sigKill created
host.Speedy object system.adapter.sainlogic.0.logLevel created
host.Speedy object sainlogic.0.info created
host.Speedy object sainlogic.0.info.last_update created
host.Speedy object sainlogic.0.info.softwaretype created
host.Speedy object sainlogic.0.weather created
host.Speedy object sainlogic.0.weather.current created
host.Speedy object sainlogic.0.weather.current.windheading created
host.Speedy object sainlogic.0.weather.maxvalues created
host.Speedy object sainlogic.0.weather.maxvalues.absolut created
host.Speedy object sainlogic.0.weather.minvalues.absolut created
host.Speedy object sainlogic.0.weather.maxvalues.daily created
host.Speedy object sainlogic.0.weather.minvalues.daily created
host.Speedy Set default value of sainlogic.0.info.last_update:
host.Speedy Set default value of sainlogic.0.info.softwaretype:
host.Speedy object system.adapter.sainlogic.0 created
Process exited with code 0

------------

The following was logged at protocol logging:

sainlogic.0 2022-07-12 17:17:36.967	info	starting. Version 0.9.0 in C:/ioBroker/node_modules/iobroker.sainlogic, node: v14.19.1, js-controller: 4.0.23
sainlogic.0 2022-07-12 17:17:36.920	debug	States connected to redis: 127.0.0.1:9000
sainlogic.0 2022-07-12 17:17:36.917	debug	States create User PubSub Client
sainlogic.0 2022-07-12 17:17:36.917	debug	States create System PubSub Client
sainlogic.0 2022-07-12 17:17:36.915	debug	Redis States: Use Redis connection: 127.0.0.1:9000
[... 47 lines cut ...]
host.Speedy 2022-07-12 17:17:12.529	info	iobroker Installing iobroker.sainlogic@0.9.0... (System call)
host.Speedy 2022-07-12 17:17:12.528	info	iobroker NPM version: 6.14.16
host.Speedy 2022-07-12 17:17:11.218	info	iobroker add sainlogic auto --host Speedy

---------

d) Next I entered the instance setup menu 
-----------------------------------------
and 
-) marked "enabled" box at listener config
-) selected 0.0.0.0 -all IPs
-) closed the setup with save and exit

After this stehp the error was logged again:
sainlogic.0 2022-07-12 17:27:28.173	info	State value to set for "sainlogic.0.info.last_listener_update" has to be type "string" but received type "number"
sainlogic.0 2022-07-12 17:27:28.141	warn	This object will not be created in future versions. Please report this to the developer.
sainlogic.0 2022-07-12 17:27:28.141	warn	Object sainlogic.0.info.last_listener_update is invalid: Default value has to be type "string" but received type "number"
sainlogic.0 2022-07-12 17:27:28.131	info	Creating new data point: weather.info.tf_batt
sainlogic.0 2022-07-12 17:27:28.131	info	Creating new data point: weather.current.tf_ch
:
: see above
:

I hope this information helps diagnosing the cause of the warning.
If I can help further, please contact me (mcm57@gmx.at). German is ok too if prefered.

phifogg, 2022-09-11:
Hmm. Unfortunately the log doesn't tell what the value was it tried to set :/ 
I'll check into the codebase...

phifogg, 2023-01-06:
Can't find anything in code base which would send a number to this state value :/
Changed one line of code to match EW with WU  code.

phifogg, 2023-02-06:
Can you reproduce this issue with the latest version or is it solved now?

phifogg, 2023-05-07:
Added to release 0.10.1

closed 2023-05-07 by phifogg