phifogg/ioBroker.sainlogic #130: [Dokumentation] small correctsions to custom settings parameters

mcm1957, 2022-06-25:
I've connected a Eurochron EFWS2900 (Conrad) to this adapter.
I suggest the following extensions (or correctsion) to the documentation:

a) When using Upload Customized with Protocol Wunderground the fields Station Id and Station Key must not be empty. I did not receive any data at ioBroker as long as those fields have been empty. Any input is OK as the values are not used by the adapter. Adapter docu states that thea are optional

b) The path inside ws-view app must end with a questionmark (/weatherstation/updateweatherstation.php?) while the path addad as paramater to the adpater doesn mot need the questaion mark. Adapter docu states that they should be identical. if the questionmark is missing at ws-view a warning is logged:

Listener received illegal request: /weatherstation/updateweatherstation.phpID=id&PASSWORD=key&indoortempf=79.2&temp....

In addition feel free to add the following device as supported / tested device:

Conrad Eurochron EFWS2900 (https://www.conrad.com/p/eurochron-efws-2900-wireless-digital-weather-station-forecasts-for-12-to-24-hours-1508025)

mcm1957, 2022-06-25:
#126 relates to paragraph b) of this issue

phifogg, 2022-06-26:
Readme updated.

closed 2022-06-26 by phifogg