phifogg/ioBroker.sainlogic #131: Min / Max Values not populated

mcm1957, 2022-06-26:
**Describe the bug**  
ALL Min / Max values are null while actual values are reportted correctly. (Some values like battery level are missing, but this seems to be ok.

**To Reproduce**  
I've set up a Conrad  Eurochron EFWS2900 in listener mode (using local LAN)
SW recorded at ioBorker is EasyWeatherV1.6.4

**Expected behavior**  
I would expect that the Min / Max values are populated

**Versions:**  
 - Adapter version: 0.9.0
 - JS-Controller version: 4.0.23
 - Node version: 14.19.3
 - Operating system: raspian

**Additional context**  
Note: Setting collect Min/Max at scheduler tab does not help. Scheduler it self is disabled as EFW2900 does not support scheduler anyway.

phifogg, 2022-06-26:
If the Weatherstation does not transmit min/max values, than nothing is stored in ioBroker. The adapter will not automatically update them.

mcm1957, 2022-06-26:
Ok, I'll check - but I assume this is the reason for the empty values.

Please consider that future versions do not create all datapoints at startup even if they will not be populated later. This is a little bit confusing. I assume the overhead to create the data points when the first min/max values are received. I do not write an extra (feature) issue for this unless you explicitly would prefer it.

mcm1957, 2022-06-26:
Setting value from data for info.last_listener_update to {"ID":"id","PASSWORD":"key","indoortempf":"80.6","tempf":"86.5","dewptf":"64.6","windchillf":"86.5","indoorhumidity":"60","humidity":"48","windspeedmph":"0.0","windgustmph":"0.0","winddir":"315","absbaromin":"29.383","baromin":"30.467","rainin":"0.000","dailyrainin":"0.000","weeklyrainin":"0.000","monthlyrainin":"5.740","solarradiation":"60.19","UV":"0","dateutc":"2022-06-26 16:32:48","softwaretype":"EasyWeatherV1.6.4","action":"updateraw","realtime":"1","rtfreq":"5"}

I do  not see any min/max values sent. 
Issue can be closed from my side. I will have to add some additional handling / wrapper.

closed 2022-06-26 by mcm1957