phifogg/ioBroker.sainlogic #133: no objects with DP250 in iobroker

lemi296, 2022-09-11:
Hello,

I am using the Froggit DP1500 (WiFi Adapter) with the Sainlogic adapter in iobroker.
It works fine so far.
Now I added a  DP250 "CO2-Messer"
In the WSView app I can see the sensor including his values but not in the sainlogic. Can I fix it myself or what can I do? Here is a photo of the data in the WSview-app.

[image]

Thanks in advance.
lemi

phifogg, 2022-09-11:
Wenn Du den Adapter auf 'debug' umstellst siehst Du im Log welche Werte er bekommt aber nicht verarbeitet. Dafür muss ein Mapping angelegt werden. Wenn Du es selbst probieren willst am besten ein Fork von dem Git-Repo machen und in die constants.php schauen. 
Sonst kannst mir auch das Log hier posten und ich schaue mal rein.

lemi296, 2022-09-12:
Hallo Phifogg, 
finde die constants.php nicht, nur die.js und bin da vermutlich beim Anpassen auch etwas überfordert. Wäre super wenn du das ergänzen könntest. Dann kann ich mir die Anpassung mal anschauen und verstehe  vielleicht wie ich weitere Komponenten einbinden könnte.
Hier das DebugLog. Vielleicht kannst du mir auch sagen warum der Scheduler immer in einen error läuft.

Danke schon mal.

Gruß lemi

<html><body>

sainlogic.0 | 6266 | 2022-09-12 17:16:38.177 | debug | Listener  received EW update:  {"PASSKEY":"xxxxCE8F7E35DB33B050124EC928xxxx","stationtype":"GW1100A_V2.1.8","runtime":"89100","dateutc":"2022-09-12   15:16:37","tempinf":"77.00","humidityin":"48","baromrelin":"29.441","baromabsin":"29.441","rainratein":"0.000","eventrainin":"0.000","hourlyrainin":"0.000","dailyrainin":"0.000","weeklyrainin":"0.000","monthlyrainin":"0.000","yearlyrainin":"0.000","totalrainin":"0.000","soilmoisture1":"43","tf_co2":"74.48","humi_co2":"58","pm25_co2":"3.3","pm25_24h_co2":"5.0","pm10_co2":"4.0","pm10_24h_co2":"5.8","co2":"459","co2_24h":"759","wh40batt":"1.5","soilbatt1":"1.6","co2_batt":"6","freq":"868M","model":"GW1100A"}
-- | -- | -- | -- | --
sainlogic.0 | 6266 | 2022-09-12 17:16:38.177 | debug | Setting value from data for info.model to GW1100A
sainlogic.0 | 6266 | 2022-09-12 17:16:38.176 | debug | Setting value from data for info.freq to 868M
sainlogic.0 | 6266 | 2022-09-12 17:16:38.176 | debug | Setting value from data for weather.info.1.soilbatt to 1.6
sainlogic.0 | 6266 | 2022-09-12 17:16:38.175 | debug | Setting value from data for weather.current.1.soilmoisture to 43
sainlogic.0 | 6266 | 2022-09-12 17:16:38.174 | debug | Target for yearlyrain unit is set: mm, using conversion: null
sainlogic.0 | 6266 | 2022-09-12 17:16:38.174 | debug | Setting value from data for weather.current.yearlyrain to 0
sainlogic.0 | 6266 | 2022-09-12 17:16:38.173 | debug | Target for monthlyrain unit is set: mm, using conversion: null
sainlogic.0 | 6266 | 2022-09-12 17:16:38.173 | debug | Setting value from data for weather.current.monthlyrain to 0
sainlogic.0 | 6266 | 2022-09-12 17:16:38.172 | debug | Target for weeklyrain unit is set: mm, using conversion: null
sainlogic.0 | 6266 | 2022-09-12 17:16:38.172 | debug | Setting value from data for weather.current.weeklyrain to 0
sainlogic.0 | 6266 | 2022-09-12 17:16:38.171 | debug | Target for dailyrain unit is set: mm, using conversion: null
sainlogic.0 | 6266 | 2022-09-12 17:16:38.170 | debug | Setting value from data for weather.current.dailyrain to 0
sainlogic.0 | 6266 | 2022-09-12 17:16:38.170 | debug | Target for rain unit is set: mm, using conversion: null
sainlogic.0 | 6266 | 2022-09-12 17:16:38.163 | debug | Setting value from data for weather.current.rain to 0
sainlogic.0 | 6266 | 2022-09-12 17:16:38.163 | debug | Target for pressureabs unit is set: hPa, using conversion: null
sainlogic.0 | 6266 | 2022-09-12 17:16:38.162 | debug | Setting value from data for weather.current.pressureabs to 996.9
sainlogic.0 | 6266 | 2022-09-12 17:16:38.162 | debug | Target for pressurerel unit is set: hPa, using conversion: null
sainlogic.0 | 6266 | 2022-09-12 17:16:38.161 | debug | Setting value from data for weather.current.pressurerel to 996.9
sainlogic.0 | 6266 | 2022-09-12 17:16:38.161 | debug | Setting value from data for weather.current.indoorhumidity to 48
sainlogic.0 | 6266 | 2022-09-12 17:16:38.160 | debug | Target for indoortemp unit is set: °C, using conversion: null
sainlogic.0 | 6266 | 2022-09-12 17:16:38.160 | debug | Setting value from data for weather.current.indoortemp to 25
sainlogic.0 | 6266 | 2022-09-12 17:16:38.159 | debug | Setting value from data for info.softwaretype to GW1100A_V2.1.8
sainlogic.0 | 6266 | 2022-09-12 17:16:38.158 | debug | Extracting value for info.model(model)
sainlogic.0 | 6266 | 2022-09-12 17:16:38.157 | debug | Extracting value for info.freq(freq)
sainlogic.0 | 6266 | 2022-09-12 17:16:38.156 | debug | No mapping found for dataelement: co2_batt
sainlogic.0 | 6266 | 2022-09-12 17:16:38.156 | debug | Extracting value for weather.info.1.soilbatt(soilbatt1)
sainlogic.0 | 6266 | 2022-09-12 17:16:38.154 | debug | No mapping found for dataelement: wh40batt
sainlogic.0 | 6266 | 2022-09-12 17:16:38.153 | debug | No mapping found for dataelement: co2_24h
sainlogic.0 | 6266 | 2022-09-12 17:16:38.152 | debug | No mapping found for dataelement: co2
sainlogic.0 | 6266 | 2022-09-12 17:16:38.151 | debug | No mapping found for dataelement: pm10_24h_co2
sainlogic.0 | 6266 | 2022-09-12 17:16:38.150 | debug | No mapping found for dataelement: pm10_co2
sainlogic.0 | 6266 | 2022-09-12 17:16:38.146 | debug | No mapping found for dataelement: pm25_24h_co2
sainlogic.0 | 6266 | 2022-09-12 17:16:38.145 | debug | No mapping found for dataelement: pm25_co2
sainlogic.0 | 6266 | 2022-09-12 17:16:38.144 | debug | No mapping found for dataelement: humi_co2
sainlogic.0 | 6266 | 2022-09-12 17:16:38.143 | debug | No mapping found for dataelement: tf_co2
sainlogic.0 | 6266 | 2022-09-12 17:16:38.142 | debug | Extracting value for weather.current.1.soilmoisture(soilmoisture1)
sainlogic.0 | 6266 | 2022-09-12 17:16:38.141 | debug | No mapping found for dataelement: totalrainin
sainlogic.0 | 6266 | 2022-09-12 17:16:38.138 | debug | Extracting value for weather.current.yearlyrain(yearlyrainin)
sainlogic.0 | 6266 | 2022-09-12 17:16:38.136 | debug | Extracting value for weather.current.monthlyrain(monthlyrainin)
[... cut ...] 3&tf_co2=74.48&humi_co2=58&pm25_co2=3.3&pm25_24h_co2=5.0&pm10_co2=4.0&pm10_24h_co2=5.8&co2=459&co2_24h=759&wh40batt=1.5&soilbatt1=1.6&co2_batt=6&freq=868M&model=GW1100A
sainlogic.0 | 6266 | 2022-09-12 17:16:18.319 | debug | FW Scheduler Connection closed
sainlogic.0 | 6266 | 2022-09-12 17:16:18.317 | error | Connection error on 192.168.178.93:9187: Error: connect ECONNREFUSED 192.168.178.93:9187
sainlogic.0 | 6266 | 2022-09-12 17:16:18.234 | info | Scheduler pulling for new data

</body>
</html>

phifogg, 2023-01-06:
Added to Version 0.10.0. Please check in to confirm or report any issues with it.

closed 2023-01-06 by phifogg