phifogg/ioBroker.sainlogic #136: [FEATURE] Reduce redundant warnings

mcm1957, 2022-09-27:
Please consider reducing the number of redundant warnings.

For an unknown reason my system sends from time to time invalid dewpointtemps. Ok, thats nothing the adapter can fix. And its ok to issue a warning. BUT please evaluate whether it would be possible to output the warning ONCE and ignore further invalid values until at least one valid value was received (or the adapter was restarted). Maybe a reset at midnicht would be useful too.

The gool should be to reduce warnings without real ne information

Example
```
sainlogic.0 | 2022-09-27 01:54:55.748 | warn | State value to set for "sainlogic.0.weather.current.dewpointtemp" has value "-5572.8" less than min "-40"
-- | -- | -- | --
sainlogic.0 | 2022-09-27 01:54:23.769 | warn | State value to set for "sainlogic.0.weather.current.dewpointtemp" has value "-5572.8" less than min "-40"
sainlogic.0 | 2022-09-27 01:53:32.056 | warn | State value to set for "sainlogic.0.weather.current.dewpointtemp" has value "-5572.8" less than min "-40"
```

Above warning fills the log ...

phifogg, 2022-09-27:
There is not a lot I can do on this warning. It is IOB complaining the value is lower then it should be. Only path would be to lower the minimum value for this object, but -5000 is reeeeeally low.

I could implement a check if the value is less then ... say -100 just to ignore it. ¯\_(ツ)_/¯

phifogg, 2022-09-27:
But then... I would not have an easy way to know if this has been reported before or not. So I cannot do a report once and then ignore.

phifogg, 2022-09-27:
you can change the the minimum temperature in ioB yourself but would probably need to do everytime you reinstall the adapter. Go to the object, select dewpointtemp and click the pencil icon. You will see a string like this.

{
  "type": "state",
  "common": {
    "name": "Dewpoint temperature",
    "type": "number",
    "unit": "°C",
    "role": "value.temperature",
    "min": -40,

Change the min to something in line with your station.

mcm1957, 2022-09-27:
> But then... I would not have an easy way to know if this has been reported before or not. So I cannot do a report once and then ignore.

Thanks for the fast reply. I'll try the workaround.

I think the message / logging handler could have a functionality like

-) if invalid value at name / state "xxx" 
          then 
                if flag in cache (index = name) is NOT set  then output warning and set flag in cache (index = name)
-) else (if  the value is valid)
          clear flag in cache

I'm sure you could not decide what is a valid value and what is an error.
But you should be able to track the status of every measurementpoint with in internal cache and issue the warning only once.

phifogg, 2023-02-06:
There is no such thing as a cache here. The adapter is being executed once per listener input. I would need to persist this type of information in additional states in IOB. Not sure this a good use of resources. If your station is sending bad that I would recommend to modify the state definitions as mentioned above.

mcm1957, 2023-03-12:
I can agree that the station was defective. so feel free to close this issue.

A cache to reduce such alarms would NOT need addition states. It could be implemented at javascript object too, thus not needeing too much resources. If NO alartm is output it would even need (almost) no ram at all (empty object).

But of course its up to you to implemenzt such a feature. Its OK for me to close issue.

phifogg, 2023-03-12:
Thanks for the update, glad to hear it was a defectice device.

Feel free to fork and add the code as you see fit :D Happy to look at your merge request.

closed 2023-03-12 by phifogg