phifogg/ioBroker.sainlogic #137: Datenpunkte werden nicht auf 0 zurückgesetzt

mangel8, 2022-10-18:
Hallo zusammen,
vielleicht könnte sich das bitte jemand im Adapter anschauen. Die Datenpunkte sainlogic.0.weather.current.dailyrain und sainlogic.0.weather.current.rain werden nach Tagesablauf bzw. wenn es aufhört zu regnen nicht zurückgesetzt.
Wetterstation Froggit WH4000SE
Danke und Grüße
Michael

Hello together,
maybe someone can take a look at the adapter. The Objects sainlogic.0.weather.current.dailyrain and sainlogic.0.weather.current.rain will not be set back to zero at a new day or if the rain stops.
Weatherstation Froggit WH4000SE
Thank you greetings
Michael

Makobe, 2023-01-05:
Dies kann ich bestätigen.
Instanz läuft als Scheduler mit einer wS980 mit EasyWeather V1.6.4
Datenpunkt:
sainlogic.0.weather.current.rain wird nach Beendigung des Regens nicht zurückgesetzt sondern bleibt auf dem letzten Stand. Bei mir z.B. 1.8 mm

phifogg, 2023-01-06:
Should be solved in latest published version. please reopen if not.

closed 2023-01-06 by phifogg