phifogg/ioBroker.sainlogic #139: More than 8 divices possible?

epiphysus, 2022-11-21:
Guten Tag. Ich habe die HP1000SE PRO Funkwetterstation. Mit dem Protokoll Listener/ Ecowitt bekommeich alle Daten tadellos im ioBroker angezeigt. Ich habe 8 Sensoren (DL5000, DP100, DP30) die als weather/current/ 1 bis 8 anzeigt werden. 7 fehlt, aber das ist vielleicht der Sensor, der die Daten unter indoor.. meldet. Jetzt habe ich einen weiteren Sensor gekauft. In dem Display der Funkwetterstation werden die Werte dazu angezeigt, aber im ioBroker wird der Sensor nicht aufgeführt. Mache ich etwas falsch? Danke im Voraus für die Hilfe. Grüße
---------------------------------
Hi. I have the HP1000SE PRO wireless weather station. With the Listener/ Ecowitt protocol, all data is displayed perfectly in ioBroker. I have 8 sensors (DL5000, DP100, DP30) which are displayed as weather/current/ 1 to 8. 7 is missing, but this is perhaps the sensor that reports the data under indoor.. . Now I bought another sensor. The values are shown in the display of the wireless weather station, but the sensor is not listed in the ioBroker. Am I doing something wrong? Thanks in advance for the help. Greetings

phifogg, 2022-11-21:
Hallo,
kannst du mir die Debug Ausgaben mal schicken?
Danke!

epiphysus, 2022-11-22:
Danke für die schnelle Antwort. Ich hoffe ich sende das richtige Protokoll. Instanz einmal neu gestartet. Dazu das Protokoll. Grüße.
[image]

phifogg, 2022-11-22:
Fast :D
Du kannst den Adapeter selbst in den debug mode setzen. Das geht wenn Du die Developer Tools aktivierst auf der Adapter seite. Anschliessenend einen Listener Aufruf abwarten. Dann sollte im Log u.a. auch der komplette String von der Wetterstation drin stehen.

epiphysus, 2022-11-22:
Ich übermittle mal als Text, der besseren Lesbarkeit halber. Ich hoffe das ist so in Ordnung. Passkey habe ich geixt.

Listener received EW update: {"PASSKEY":"xxx","stationtype":"EasyWeatherV1.6.4","dateutc":"2022-11-22 08:07:48","tempinf":"40.5","humidityin":"79","baromrelin":"28.976","baromabsin":"28.976","tempf":"41.5","humidity":"97","winddir":"251","winddir_avg10m":"175","windspeedmph":"0.2","windspdmph_avg10m":"1.3","windgustmph":"3.4","maxdailygust":"6.9","rainratein":"0.000","eventrainin":"0.091","hourlyrainin":"0.012","dailyrainin":"0.051","weeklyrainin":"0.339","monthlyrainin":"2.591","yearlyrainin":"25.220","solarradiation":"24.22","uv":"0","temp1f":"40.1","humidity1":"99","temp4f":"69.1","humidity4":"44","temp8f":"45.5","soilmoisture2":"38","soilmoisture3":"55","soilmoisture4":"25","soilmoisture5":"69","soilmoisture6":"60","wh65batt":"0","wh25batt":"0","batt1":"0","batt2":"0","batt3":"0","batt4":"0","batt8":"0","soilbatt2":"1.5","soilbatt3":"1.5","soilbatt4":"1.6","soilbatt5":"1.6","soilbatt6":"1.5","freq":"868M","model":"HP2550A_Pro_V1.7.5"}

phifogg, 2022-11-22:
Ok, danke. soweit ich sehe kommen:
- Station selbst
- 1. Zusätzlicher Temperatur- und Feuchtigkeitssensor
- 2. Zusätzlicher Bodenfeuchtigkeitssensor
- 3. Zusätzlicher Bodenfeuchtigkeitssensor
- 4. Zusätzlicher Bodenfeuchtigkeitssensor
- 5. Zusätzlicher Bodenfeuchtigkeitssensor
- 6. Zusätzlicher Bodenfeuchtigkeitssensor
- 7.  .. nichts ?
- 8. Zusätzlicher Temperatursensor

Also 7 scheint nichts zu schicken. ¯\_(ツ)_/¯

Vermutlich kriegst Du auch nicht alle Batterie-Werte, denke die soilbatt hole ich glaub noch nicht raus.

Und was genau fehlt Dir nun im ioB? Der 7-er?

epiphysus, 2022-11-22:
Zuerst zum 7ner Sensor: der Zeigt sich als Indoorhumidity und indoortemp (siehe Bild)

Ich vermisse meinen neuen, neunten Sensor, Das Display der Wetterstation zeigt die Werte, er ist also dort "angelernt", aber in ioBroker taucht ebendieser neunte Sensor nicht auf...

[image]

phifogg, 2022-11-22:
Als was wird der denn im display angezeigt?

epiphysus, 2022-11-22:
Ich sehe gerade, das im Ordner weather/current/4 der neue Sensor, den ich auf Kanal 4 eingestellt hatte mit seinen Werten Auftaucht. Ich hatte gedacht, das die Ordner je Sensor hochzählen und hatte auf einen Ordner 9 gewartet. Sorry.
Also alles ist korrekt. Der Fehler lag bei mir. Mea culpa.

epiphysus, 2022-11-22:
Danke für die Hilfe...

phifogg, 2022-11-22:
👍

closed 2022-11-22 by phifogg