phifogg/ioBroker.sainlogic #142: Scheduler received data for unkown command 84 subcommand 49

grichter24, 2023-01-10:
Hallo,

heute wollte ich den Adapter auf Version 0.10.0 updaten von glaube 0.8.0?

Dabei ist mein ganzes iobroker abgestürzt. Danach waren unter opt/iobroker/node-modules ganz viele Ordner weg. Habe ich alle aus einer Sicherung wiederhergestellt. iobroker läuft wieder.

Im Protokoll kommen jetzt aber immer beim Adapter die Meldung
"Scheduler received data for unkown command 84 subcommand 49"

Habe sainlogic schon komplett deinstalliert und neu installiert. 
Leider bleibt der Fehler.

Die Werte der Station werden auch aktualisiert. Zusätzlich sollen die Daten noch über eine Weiterleitungs-URL an meine Homepage geschickt werden. Dort kommen sie aber nicht an.

Hilfe!

Gruß
Gordon

phifogg, 2023-01-10:
Huch.. .ein kompletter ioB absturz? Hast Du da vielleicht noch irgendwelche Logs gezogen davon? Syslog oder so? Das klingt ziemlich heftig - tut mir leid und wüsste leider auch nicht woran das liegen sollte.

Was für eine Station hast Du? Bei einigen Stationen geht der Scheduler nicht. Weiterleitung klingt eigentlich nach Listener Mode. Versuche doch den Adapter in Debug modus zu setzen, dann kann ich Dir mit dem Log mehr helfen.

grichter24, 2023-01-11:
Guten Morgen,
 
ja, war ein Komplettabsturz. Aber ist nichts weiter passiert, konnte alles reparieren. Und gute Erinnerung daran, mal wieder ein Backup zu machen :)
 
Im Verzeichnis opt/iobroker/node_modules waren es knapp 200 Verzeichnisse weniger als in meiner Sicherung. Die kompletten "node-..."-Verzeichnisse waren z.B. weg.
 
Also mit der Version 0.8.0 funktioniert die Datenübertragung tatsächlich. Ich lass die erstmal installiert. Komischerweise kommen die Meldungen trotzdem.
 
Ich schau mal nach den Logs --> hängen an der Mail.
 
Ich habe eine Sainlogic WS3500.
 
Danke für die Hilfe :)
 
LG
Gordon 
[Logs.zip](https://github.com/phifogg/ioBroker.sainlogic/files/10389858/Logs.zip)

phifogg, 2023-01-11:
Puh.. in den Logs was zu finden ist nicht leicht wenn man nicht weiss nach was man suchen soll :D
Wann hast Du denn das Upgrade des Adapters angeworfen? Das IOBroker log ist nur von danach, also da sehe ich eh nichts mehr.

Der erste Fehler den ich im syslog sehe wäre der hier:
```
Jan 10 21:39:03 openhabian bash[1384]: Cannot get installed host information: The "path" argument must be of type string. Received undefined
Jan 10 21:39:03 openhabian bash[1384]: TypeError [ERR_INVALID_ARG_TYPE]: The "path" argument must be of type string. Received undefined
Jan 10 21:39:03 openhabian bash[1384]:     at new NodeError (internal/errors.js:322:7)
Jan 10 21:39:03 openhabian bash[1384]:     at validateString (internal/validators.js:124:11)
Jan 10 21:39:03 openhabian bash[1384]:     at Object.join (path.js:1148:7)
Jan 10 21:39:03 openhabian bash[1384]:     at Object.getInstalledInfo (/opt/iobroker/node_modules/@iobroker/js-controller-common/lib/common/tools.js:872:41)
Jan 10 21:39:03 openhabian bash[1384]:     at writeUpdateInfo (/opt/iobroker/node_modules/iobroker.admin/main.js:335:27)
Jan 10 21:39:03 openhabian bash[1384]:     at Timeout._onTimeout (/opt/iobroker/node_modules/iobroker.admin/main.js:65:21)
Jan 10 21:39:03 openhabian bash[1384]:     at listOnTimeout (internal/timers.js:557:17)
Jan 10 21:39:03 openhabian bash[1384]:     at processTimers (internal/timers.js:500:7)
```
Das kommt irgendwo aus den (Un-)tiefen des IOB. Da kann ich nichts zu sagen. Kann aber natürlich ein Folgefehler sein von dem abgestürtzten upgrade.

Zu der Fehlermeldung im Log jetzt, Deine WS3500 geht nur über den Listener. Du must den Scheduler/Zeitplan ausschalten. Dann ist die Meldung auch weg.

Der Listener dagegen scheint nicht aktiv zu sein. Zumindest sehe ich davon in den Logs nichts.

grichter24, 2023-01-11:
ok, danke schonmal :)

Ich weiß nicht mehr wann ich das Update gestartet hatte. Hat aber auch grundsätzlich dann seeeeehr lange gedauert. Die Neuinstallation war ja innerhalb von Sekunden durch.

Vielleicht teste ich das Update nochmal. Aber da kamen ja irgendwie keine Daten auf meiner Homepage an...

Also ich hänge mal 3 Screenshots von den Einstellungen ran. Damit funktioniert alles in der 0.8.0 (nur die Fehlermeldungen kommen trotzdem, weiß aber nicht mehr, ob das vorher auch so war.

[image]
[image]
[image]

grichter24, 2023-01-11:
sieht man auf den Screenshot nicht, aber bei Listener und Zeitplan ist der Haken bei "aktiv" gesetzt

grichter24, 2023-01-11:
und nochmal...

habe jetzt Update zu 0.10.1 gemacht. Lief problemlos durch... komisch gestern das war ;)

aber trotzdem werden die Daten mit der 0.10.0 nicht an die Weiterleitungs-URL geschickt. Die Daten im iob bekommen das Update, aber weiter an meine Homepage geht es nicht

phifogg, 2023-01-11:
Das hier ist aus dem 0.8.0-er log, da sieht man die forward URL schön:
```
2023-01-11 12:18:06.372  - [34mdebug[39m: sainlogic.0 (32647) Listener IP: 192.168.77.47
2023-01-11 12:18:06.374  - [34mdebug[39m: sainlogic.0 (32647) Listener port: 45000
2023-01-11 12:18:06.375  - [34mdebug[39m: sainlogic.0 (32647) Listener path: /weatherstation/updateweatherstation.php
2023-01-11 12:18:06.376  - [34mdebug[39m: sainlogic.0 (32647) Listener protocol: wunderground
2023-01-11 12:18:06.377  - [34mdebug[39m: sainlogic.0 (32647) Forward URL: http://www.fam-richter.de/meteo/wetterupload.php?
```

Mit dem 0.10.0-er update sehe ich diese Meldung:
```
2023-01-11 12:17:09.208  - [32minfo[39m: sainlogic.0 (32586) undefined
```

Das dürfte das problem sein. Jetzt muss ich suchen was 'undefined' ist. Eigentlich habe ich da nichts wissentlich geändert. Ich schaue später mal rein. Danke für das log!

phifogg, 2023-01-11:
Sehe gerade, das sieht auch spannend aus mit 0.10.0
```
Extracting value for weather.maxvalues.daily.UVraw_time(ID)
2023-01-11 12:17:08.392  - [34mdebug[39m: sainlogic.0 (32586) Extracting value for weather.maxvalues.daily.UVraw_time(PASSWORD)
2023-01-11 12:17:08.394  - [34mdebug[39m: sainlogic.0 (32586) Extracting value for weather.maxvalues.daily.UVraw_time(indoortempf)
2023-01-11 12:17:08.395  - [34mdebug[39m: sainlogic.0 (32586) Extracting value for weather.maxvalues.daily.UVraw_time(tempf)
2023-01-11 12:17:08.396  - [34mdebug[39m: sainlogic.0 (32586) Extracting value for weather.maxvalues.daily.UVraw_time(dewptf)
2023-01-11 12:17:08.397  - [34mdebug[39m: sainlogic.0 (32586) Extracting value for weather.maxvalues.daily.UVraw_time(windchillf)
2023-01-11 12:17:08.398  - [34mdebug[39m: sainlogic.0 (32586) Extracting value for weather.maxvalues.daily.UVraw_time(indoorhumidity)
2023-01-11 12:17:08.399  - [34mdebug[39m: sainlogic.0 (32586) Extracting value for weather.maxvalues.daily.UVraw_time(humidity)
2023-01-11 12:17:08.400  - [34mdebug[39m: sainlogic.0 (32586) Extracting value for weather.maxvalues.daily.UVraw_time(windspeedmph)
2023-01-11 12:17:08.401  - [34mdebug[39m: sainlogic.0 (32586) Extracting value for weather.maxvalues.daily.UVraw_time(windgustmph)
2023-01-11 12:17:08.403  - [34mdebug[39m: sainlogic.0 (32586) Extracting value for weather.maxvalues.daily.UVraw_time(winddir)
2023-01-11 12:17:08.404  - [34mdebug[39m: sainlogic.0 (32586) Extracting value for weather.maxvalues.daily.UVraw_time(absbaromin)
2023-01-11 12:17:08.405  - [34mdebug[39m: sainlogic.0 (32586) Extracting value for weather.maxvalues.daily.UVraw_time(baromin)
2023-01-11 12:17:08.406  - [34mdebug[39m: sainlogic.0 (32586) Extracting value for weather.maxvalues.daily.UVraw_time(rainin)
2023-01-11 12:17:08.407  - [34mdebug[39m: sainlogic.0 (32586) Extracting value for weather.maxvalues.daily.UVraw_time(dailyrainin)
2023-01-11 12:17:08.408  - [34mdebug[39m: sainlogic.0 (32586) Extracting value for weather.maxvalues.daily.UVraw_time(weeklyrainin)
2023-01-11 12:17:08.409  - [34mdebug[39m: sainlogic.0 (32586) Extracting value for weather.maxvalues.daily.UVraw_time(monthlyrainin)
2023-01-11 12:17:08.410  - [34mdebug[39m: sainlogic.0 (32586) Extracting value for weather.maxvalues.daily.UVraw_time(solarradiation)
2023-01-11 12:17:08.411  - [34mdebug[39m: sainlogic.0 (32586) Extracting value for weather.maxvalues.daily.UVraw_time(UV)
2023-01-11 12:17:08.412  - [34mdebug[39m: sainlogic.0 (32586) Extracting value for weather.maxvalues.daily.UVraw_time(dateutc)
```

Der scheint alle Attribute auf UVraw_time setzen zu wollen. das ist nicht gut.

grichter24, 2023-01-11:
was mir noch aufgefallen ist gerade...

Es werden nur die Listenerwerte aktualisiert. Unter Objekte --> sainlogic.0.weather.current steht alles auf NULL. 

Der Upload auf die Weiterleitungurl hat aber die Daten. Nur im current steht nichts drin. Dadurch sind die Daten dann im MQTT auch leer. Kannst du da nochmal schauen?

phifogg, 2023-01-12:
Da ist wirklich was schief bei Dir. Hängt sicher noch an dem Absturz. Habe eben mein 0.10.0er angeschaut, da klappt das mapping auf die Werte.

Um ihn komplett loszuwerden in ioB deinstallen (soweit halt möglich), dann alle Objekte in sainlogic löschen und zur sicherheit auf der Disk in node-modules das Verzeichnis iobroker.sainlogic löschen. Dann sollte eine neue installation auch wieder gehen.

phifogg, 2023-02-06:
Funktioniert es nun bei Dir? Würde sonst gerne das issue schliessen.

grichter24, 2023-02-06:
Funktioniert nach Neuinstallation alles wieder super. Da war scheinbar in der iob-Installation der Wurm drin...

Vielen Dank

closed 2023-02-06 by grichter24

grichter24, 2023-02-16:
Hallo Daniel,
 
neuerdings habe ich in unregelmäßigen Abständen merkwürdige Werte bei mir stehen.
 
Die Außentempertur liegt dann bei -5572,8 und die Luftfeuchtigkeit bei -9999
 
Nach ein paar Minuten ist wieder alles schick. In dieser Zeit werden auch keine Daten an Weatherunderground geschickt. Auch auf meine Homepage (Meteotemplate) kommen diese Daten nicht an. Per MQTT werden sie allerdings veröffentlicht.
 
Ist das Gerät defekt? Oder sagt dir das irgendetwas?
 
Vielen Dank.
 
Gruß
Gordon

Message ID: ***@***.***>