phifogg/ioBroker.sainlogic #143: Forwarding does not work

bw1faeh0, 2023-01-16:
Hi,

I'm trying to forward my ecowitt data (ecowitt protocol) to my (custom) server. On this server I'm logging my data, too.

The setup looks like this:
- Listen on all IPs, Port 4500
- path: `/index.php`
- protocol: `Ecowitt`
- forwarding url: `http://wetterupdate.flaemig42.de/index.php`

I can read all weather data from the objects within IOBroker, so basically the adapter and instance is working.

But I can read from my access logs of my server, that the plugin does not forward any data, since there is no POST request on the index.php from the IP of my IObroker (home).

How can I do further debugging?

phifogg, 2023-01-16:
Hi,
you can set the adapter to debug mode. Wait for one incoming listener call and send me the (filtered) log entries. This should at least show if/where it tries to forward to.

bw1faeh0, 2023-01-16:
Since I'm new to IOBroker, I'm not sure, if this log is the one you are looking for...

[debug.log](https://github.com/phifogg/ioBroker.sainlogic/files/10424560/debug.log)

phifogg, 2023-01-16:
Ok, I see the forward URL is set. Let me check the codebase on when this is actually used. Can't find another output with that.

What I do see though is your weather station seems to send the data every 1 second. This is a high frequency as ioB needs almost a full second to process. In case your ioB is really busy (high load) this could be a reason. Might be an idea to lower the frequency for updates a little bit in the weather station.

bw1faeh0, 2023-01-16:
The weatherstation is set to send data every 60 seconds. An this ist what I can see in IOBroker, too. 
The values get updated every 60 seconds. 

[image]

bw1faeh0, 2023-01-16:
Which port is used for forwarding? 45000? 80?

phifogg, 2023-01-16:
Forwarding uses standard http port (80).

bw1faeh0, 2023-01-18:
I just created a workaround:

I set up my custom server within the weatherstation again (as it was before using sainlogic). Then I created a ssh tunnel from my custom server to the device my IObroker is running on (forwarding http traffic to port 45000 on the device). 
My http server on my custom server is now (additionally) forwarding the POST data from the weatherstation through the ssh tunnel to the sainlogic instance.

phifogg, 2023-01-18:
Hm.. ok. this seems a viable choice if you can run a custom server.
I'd still be interessted what works or doesn't work in your scenario.

Based on the code, the forwarding will happen anytime I get a request just after processing all the data internally. A log message will be generated **IF** the receiving servers is giving any kind of response to the request. **OR** if there is any exception thrown.

As there is no log message in your case, it seems the receiving server just accepted it and didn't respond? Is that an option?

I can add more debug statements to try, or you could - sounds like you are quite good on linux hosts :D - use tcpdump to monitor outgoing traffic from your iob host.

phifogg, 2023-01-18:
AH.. if you need a specific port in the forward, you need to specify it as part of the URL. Like https://my.host:45000/upload. This should work for custom ports.

bw1faeh0, 2023-01-18:
> Based on the code, the forwarding will happen anytime I get a request just after processing all the data internally. A log message will be generated **IF** the receiving servers is giving any kind of response to the request. **OR** if there is any exception thrown.
> 
I took a look into you code, too, and noticed the same: data should be forwarded every time. Maybe `got` has a problem

> As there is no log message in your case, it seems the receiving server just accepted it and didn't respond? Is that an option?

The nginx server logs every access attempt, even it would result in an error (not found, not allowed, etc.). As there is no connection attempt logged, my guess is, that there is no traffic going out from IObroker/sainlogic.

> I can add more debug statements to try, or you could - sounds like you are quite good on linux hosts :D - use tcpdump to monitor outgoing traffic from your iob host.

I will check in the next days what traffic is generated using tcpdump.
As I found a workaround, the urgency has decreased

bw1faeh0, 2023-01-18:
> AH.. if you need a specific port in the forward, you need to specify it as part of the URL. Like https://my.host:45000/upload. This should work for custom ports.

Nope, my server is using port 80 for incoming weather data.

PastCoder, 2023-03-26:
I am also trying to forward information, but don't succeed.
When a forward URL is defined, I get periodic error message:
RequestError: Expected value which is `plain object`, received value of type `number`.
Unfortunately I don't see what the result was to investigate in the problem. Even better would to see (in debug log or silly settings) the URL which was used to upload. Not only the base formward URL when starting but with each access. Then I could copy&paste it to see the server (forward target) answer.

tomschlde, 2023-05-06:
Hi devs, no idea how to fix it?
Or is the adapter depreciated?

phifogg, 2023-05-06:
No, not dead... just slow as so many other things to do :)

Looking into this, issue could be in the code, retry policy is wrongly used -> need update
(Comment to myself to fix it: {limit: 0})

phifogg, 2023-05-07:
Added to release 0.10.1

tomschlde, 2023-05-09:
Unfortunately same bahavior with 0.10.3 as described in #154 .
No forwarding to be noted.
Maybe some logging hast tro be added here.

Here some logs from my test:

`2023-05-09 15:50:18.103 - debug: sainlogic.0 (16734) Listener body is PASSKEY=XX&stationtype=GW2000A_V2.2.3&runtime=1883565&dateutc=2023-05-09+13:50:18&tempinf=74.12&humidityin=53&baromrelin=29.613&baromabsin=29.613&tempf=61.52&humidity=90&winddir=264&windspeedmph=1.57&windgustmph=3.80&maxdailygust=12.53&solarradiation=106.24&uv=0&rrain_piezo=0.378&erain_piezo=0.098&hrain_piezo=0.098&drain_piezo=0.098&wrain_piezo=0.185&mrain_piezo=0.484&yrain_piezo=3.012&ws90cap_volt=5.2&ws90_ver=126&soilmoisture1=61&soilmoisture2=52&soilmoisture3=49&soilmoisture4=60&soilmoisture5=61&soilmoisture6=50&soilbatt1=1.6&soilbatt2=1.6&soilbatt3=1.6&soilbatt4=1.6&soilbatt5=1.7&soilbatt6=1.6&wh90batt=2.62&freq=868M&model=GW2000A&interval=20
2023-05-09 15:50:18.104 - debug: sainlogic.0 (16734) Listener received EW update: {"PASSKEY":"XX","stationtype":"GW2000A_V2.2.3","runtime":"1883565","dateutc":"2023-05-09 13:50:18","tempinf":"74.12","humidityin":"53","baromrelin":"29.613","baromabsin":"29.613","tempf":"61.52","humidity":"90","winddir":"264","windspeedmph":"1.57","windgustmph":"3.80","maxdailygust":"12.53","solarradiation":"106.24","uv":"0","rrain_piezo":"0.378","erain_piezo":"0.098","hrain_piezo":"0.098","drain_piezo":"0.098","wrain_piezo":"0.185","mrain_piezo":"0.484","yrain_piezo":"3.012","ws90cap_volt":"5.2","ws90_ver":"126","soilmoisture1":"61","soilmoisture2":"52","soilmoisture3":"49","soilmoisture4":"60","soilmoisture5":"61","soilmoisture6":"50","soilbatt1":"1.6","soilbatt2":"1.6","soilbatt3":"1.6","soilbatt4":"1.6","soilbatt5":"1.7","soilbatt6":"1.6","wh90batt":"2.62","freq":"868M","model":"GW2000A","interval":"20"}
2023-05-09 15:50:18.106 - debug: sainlogic.0 (16734) No mapping found for dataelement: PASSKEY
2023-05-09 15:50:18.107 - debug: sainlogic.0 (16734) Extracting value for info.softwaretype(stationtype)
2023-05-09 15:50:18.108 - debug: sainlogic.0 (16734) No mapping found for dataelement: runtime
2023-05-09 15:50:18.109 - debug: sainlogic.0 (16734) No mapping found for dataelement: dateutc
2023-05-09 15:50:18.110 - debug: sainlogic.0 (16734) Extracting value for weather.current.indoortemp(tempinf)
2023-05-09 15:50:18.112 - debug: sainlogic.0 (16734) Extracting value for weather.current.indoorhumidity(humidityin)
2023-05-09 15:50:18.114 - debug: sainlogic.0 (16734) Extracting value for weather.current.pressurerel(baromrelin)
2023-05-09 15:50:18.116 - debug: sainlogic.0 (16734) Extracting value for weather.current.pressureabs(baromabsin)
2023-05-09 15:50:18.118 - debug: sainlogic.0 (16734) Extracting value for weather.current.outdoortemp(tempf)
2023-05-09 15:50:18.121 - debug: sainlogic.0 (16734) Extracting value for weather.current.outdoorhumidity(humidity)
2023-05-09 15:50:18.122 - debug: sainlogic.0 (16734) Extracting value for weather.current.winddir(winddir)
2023-05-09 15:50:18.123 - debug: sainlogic.0 (16734) Extracting value for weather.current.windspeed(windspeedmph)
2023-05-09 15:50:18.126 - debug: sainlogic.0 (16734) Extracting value for weather.current.windgustspeed(windgustmph)
2023-05-09 15:50:18.127 - debug: sainlogic.0 (16734) Extracting value for weather.current.maxdailygust(maxdailygust)
2023-05-09 15:50:18.132 - debug: sainlogic.0 (16734) Extracting value for weather.current.solarradiation(solarradiation)
2023-05-09 15:50:18.136 - debug: sainlogic.0 (16734) Extracting value for weather.current.uvi(uv)
2023-05-09 15:50:18.137 - debug: sainlogic.0 (16734) Extracting value for weather.current.rain_piezo(rrain_piezo)
2023-05-09 15:50:18.138 - debug: sainlogic.0 (16734) Extracting value for weather.current.eventrain_piezo(erain_piezo)
2023-05-09 15:50:18.140 - debug: sainlogic.0 (16734) Extracting value for weather.current.hourlyrain_piezo(hrain_piezo)
2023-05-09 15:50:18.141 - debug: sainlogic.0 (16734) Extracting value for weather.current.dailyrain_piezo(drain_piezo)
2023-05-09 15:50:18.142 - debug: sainlogic.0 (16734) Extracting value for weather.current.weeklyrain_piezo(wrain_piezo)
2023-05-09 15:50:18.143 - debug: sainlogic.0 (16734) Extracting value for weather.current.monthlyrain_piezo(mrain_piezo)
2023-05-09 15:50:18.144 - debug: sainlogic.0 (16734) Extracting value for weather.current.yearlyrain_piezo(yrain_piezo)
2023-05-09 15:50:18.145 - debug: sainlogic.0 (16734) No mapping found for dataelement: ws90cap_volt
2023-05-09 15:50:18.145 - debug: sainlogic.0 (16734) No mapping found for dataelement: ws90_ver
2023-05-09 15:50:18.147 - debug: sainlogic.0 (16734) Extracting value for weather.current.1.soilmoisture(soilmoisture1)
2023-05-09 15:50:18.148 - debug: sainlogic.0 (16734) Extracting value for weather.current.2.soilmoisture(soilmoisture2)
2023-05-09 15:50:18.149 - debug: sainlogic.0 (16734) Extracting value for weather.current.3.soilmoisture(soilmoisture3)
2023-05-09 15:50:18.149 - debug: sainlogic.0 (16734) Extracting value for weather.current.4.soilmoisture(soilmoisture4)
2023-05-09 15:50:18.150 - debug: sainlogic.0 (16734) Extracting value for weather.current.5.soilmoisture(soilmoisture5)
2023-05-09 15:50:18.151 - debug: sainlogic.0 (16734) Extracting value for weather.current.6.soilmoisture(soilmoisture6)
2023-05-09 15:50:18.151 - debug: sainlogic.0 (16734) Extracting value for weather.info.1.soilbatt(soilbatt1)
[... cut ...] piezo":"0.484","yrain_piezo":"3.012","ws90cap_volt":"5.2","ws90_ver":"126","soilmoisture1":"61","soilmoisture2":"52","soilmoisture3":"49","soilmoisture4":"60","soilmoisture5":"61","soilmoisture6":"50","soilbatt1":"1.6","soilbatt2":"1.6","soilbatt3":"1.6","soilbatt4":"1.6","soilbatt5":"1.7","soilbatt6":"1.6","wh90batt":"2.62","freq":"868M","model":"GW2000A","interval":"20"}
2023-05-09 15:50:18.358 - info: sainlogic.0 (16734) undefined
2023-05-09 15:50:18.358 - info: sainlogic.0 (16734) undefined`

phifogg, 2023-05-10:
there is actually a log message once the forwarding has happened.
Can you put that startup part of your adapter log? This should print the config it has.

phifogg, 2023-05-28:
0.10.4 has updated logging, it should not the URL as well as the response received from the request.

phifogg, 2023-07-23:
Can you confirm this still be an issue? If yes, check the logs.
Once you go to 0.10.5, make sure you put the adapter in debug mode, logs only show in debug.