phifogg/ioBroker.sainlogic #144: Solar radtion is 0 if sainlogic is set to W/m2

bw1faeh0, 2023-01-18:
**Describe the bug**  
If I configure sainlogic to use `W/m2` instead of `lx` for the solar radiation, the object value is always set to 0.
When set to `W/m2` the log tells `Target for solarradiation unit is set: W/m2, using conversion: roundTo(x / 126.7, 1)` **every time** data arrives from the weather station.

**To Reproduce**  
Steps to reproduce the behavior:
1. Go to instance settings of sainloigc
2. Click on 'device settings'
3. Set solar radiation to 'W/m2'

**Expected behavior**  
correct calculation

**Versions:**  
 - Adapter version: v0.13.1
 - JS-Controller version: 4.0.23
 - Node version: 16.19.0
 - Operating system: debian

oelison, 2023-04-27:
I had the same issue. Push the edit button and set the unit to W/m2. It is cyclicly calling the checkUnit function and divide it to zero and stops then. I think the function need to set the unit of the state correct and then this will disapear also on such change.

phifogg, 2023-05-07:
Added to release 0.10.1

closed 2023-05-07 by phifogg