phifogg/ioBroker.sainlogic #145: Yearly Rain Warning / WS3500

mrfenyx, 2023-01-18:
For some reason (might be the default state), the value for Yearly Rain (yearlyrainin) that the WS is sending is sometimes negative, which makes no sense. Could a check be implemented so that if certain values are < 0 then they are converted to 0 and no warning is shown?

**To Reproduce**  
Steps to reproduce the behavior:
1. On the Controller / Tablet long-press (2-3) seconds the "Rain" key so that the statistical values are reset
2. In ioBroker, check the value that the WS3500 is sending for "yearlyrainin" -> in my case, "-9999"
3. Check the ioBroker Log -> probably the message `State value to set for "sainlogic.0.weather.current.yearlyrain" has value "-253974.6" less than min "0"` is displayed.

**Versions:**  
 - Adapter version: 0.10.0
 - JS-Controller version: 4.0.24
 - Node version: 16.19.0
 - Operating system: Raspberry Pi OS

mrfenyx, 2023-02-03:
UPDATE: seems that the actual value used by Weather Underground and Ecowitt for "Yearly Rain" is "totalrainin" from the raw data sent by the station. "yearlyrainin" remains "-9999" in my case, even as the weather station reports other rain values.

phifogg, 2023-02-06:
As a workaround, you can check into the adapter state definition and modify the min-value for yearlyrainin:

```
{
  "type": "state",
  "common": {
    "name": "Max. Rainrate (Yearly) (absolut)",
    "type": "number",
    "unit": "mm",
    "role": "value",
    "min": -260000,
    "max": 3000,
    "def": 0,
    "read": true,
    "write": false,
    "mobile": {
      "admin": {
        "visible": true
      }
    }
  },
}
```
This would at leats get rid of the warning. This change will be persistent until you remove the adapter and reinstall. The adapter won't change the setting once it is running.

mrfenyx, 2023-02-08:
@phifogg thanks, that works as a workaround. The actual issue is still there though and currently the (correct) value "totalrainin" is not actually used. At least for the WS3500 it doesn't look like yearlyrainin is used.

phifogg, 2023-05-07:
Added to release 0.10.1

closed 2023-05-07 by phifogg