phifogg/ioBroker.sainlogic #147: 1 Wetterstation Sainlogic WS 3500 in zwei ioBroker-Instanzen verwenden

Pandia22, 2023-02-02:
<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:w="urn:schemas-microsoft-com:office:word"
xmlns:m="http://schemas.microsoft.com/office/2004/12/omml"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Word.Document>
<meta name=Generator content="Microsoft Word 12">
<meta name=Originator content="Microsoft Word 12">
<link rel=File-List
href="file:///C:\Users\Heiko\AppData\Local\Temp\msohtmlclip1\01\clip_filelist.xml">
<link rel=themeData
href="file:///C:\Users\Heiko\AppData\Local\Temp\msohtmlclip1\01\clip_themedata.thmx">
<link rel=colorSchemeMapping
href="file:///C:\Users\Heiko\AppData\Local\Temp\msohtmlclip1\01\clip_colorschememapping.xml">

<style>

</style>

</head>

<body lang=DE style='tab-interval:35.4pt'>

<p class=MsoNormal style='margin-bottom:0cm;margin-bottom:.0001pt;line-height:
normal'>**Describe the bug**<span style='mso-spacerun:yes'>  </span></p>

<p class=MsoNormal style='margin-bottom:0cm;margin-bottom:.0001pt;line-height:
normal'>Bei mir laufen zwei identische ioBroker-Server auf unterschiedlicher
Hardware. Auf dem ersten Server ist des problemlos gelungen die Wetterstation
Sainlogic WS3500 mit dem gleichnamigen Adapter in das System zu integrieren.
Auf den zweiten startet der Adapter und ist grün. Es werden jedoch keine
Wetterdaten geschrieben.</p>

<p class=MsoNormal style='margin-bottom:0cm;margin-bottom:.0001pt;line-height:
normal'><o:p>&nbsp;</o:p></p>

<p class=MsoNormal style='margin-bottom:0cm;margin-bottom:.0001pt;line-height:
normal'>**To Reproduce**<span style='mso-spacerun:yes'>  </span></p>

<p class=MsoNormal style='margin-bottom:0cm;margin-bottom:.0001pt;line-height:
normal'>Steps to reproduce the behavior:</p>

<p class=MsoNormal style='margin-bottom:0cm;margin-bottom:.0001pt;line-height:
normal'>Start – Stop des Adpaters</p>

<p class=MsoNormal style='margin-bottom:0cm;margin-bottom:.0001pt;line-height:
normal'><o:p>&nbsp;</o:p></p>

<p class=MsoNormal style='margin-bottom:0cm;margin-bottom:.0001pt;line-height:
normal'>**Expected behavior**<span style='mso-spacerun:yes'>  </span></p>

<p class=MsoNormal style='margin-bottom:0cm;margin-bottom:.0001pt;line-height:
normal'>Aktualisierung der Datenpunkte auf beiden ioBroker-Servern.</p>

<p class=MsoNormal style='margin-bottom:0cm;margin-bottom:.0001pt;line-height:
normal'><o:p>&nbsp;</o:p></p>

<p class=MsoNormal style='margin-bottom:0cm;margin-bottom:.0001pt;line-height:
normal'>**Screenshots &amp; Logfiles**<span style='mso-spacerun:yes'>  </span></p>

<p class=MsoNormal style='margin-bottom:0cm;margin-bottom:.0001pt;line-height:
normal'>If applicable, add screenshots and logfiles to help explain your
problem.</p>

sainlogic.0 | 2023-02-02 08:12:31.231 | info | Starting Listener
-- | -- | -- | --
sainlogic.0 | 2023-02-02 08:12:31.230 | debug | Forward URL:
sainlogic.0 | 2023-02-02 08:12:31.229 | debug | Listener protocol: ecowitt
sainlogic.0 | 2023-02-02 08:12:31.228 | debug | Listener path: /weatherstation/data
sainlogic.0 | 2023-02-02 08:12:31.228 | debug | Listener port: 45000
sainlogic.0 | 2023-02-02 08:12:31.227 | debug | Listener IP: 0.0.0.0
sainlogic.0 | 2023-02-02 08:12:31.173 | info | starting. Version 0.10.0 in   /opt/iobroker/node_modules/iobroker.sainlogic, node: v19.3.0, js-controller:   4.0.24
sainlogic.0 | 2023-02-02 08:12:30.892 | debug | States connected to redis: 127.0.0.1:9000
sainlogic.0 | 2023-02-02 08:12:30.848 | debug | States create User PubSub Client
sainlogic.0 | 2023-02-02 08:12:30.846 | debug | States create System PubSub Client
sainlogic.0 | 2023-02-02 08:12:30.820 | debug | Redis States: Use Redis connection: 127.0.0.1:9000
sainlogic.0 | 2023-02-02 08:12:30.778 | debug | Objects connected to redis: 127.0.0.1:9001
sainlogic.0 | 2023-02-02 08:12:30.770 | debug | Objects client initialize lua scripts
sainlogic.0 | 2023-02-02 08:12:30.684 | debug | Objects create User PubSub Client
sainlogic.0 | 2023-02-02 08:12:30.682 | debug | Objects create System PubSub Client
sainlogic.0 | 2023-02-02 08:12:30.680 | debug | Objects client ready ... initialize now
sainlogic.0 | 2023-02-02 08:12:30.614 | debug | Redis Objects: Use Redis connection: 127.0.0.1:9001
host.Poseidon | 2023-02-02 08:12:28.659 | info | instance system.adapter.sainlogic.0 started with pid   28139

<p class=MsoNormal style='margin-bottom:0cm;margin-bottom:.0001pt;line-height:
normal'><o:p>&nbsp;</o:p></p>

<p class=MsoNormal style='margin-bottom:0cm;margin-bottom:.0001pt;line-height:
normal'>**Versions:**<span style='mso-spacerun:yes'>  </span></p>

<p class=MsoNormal style='margin-bottom:0cm;margin-bottom:.0001pt;line-height:
normal'><span style='mso-spacerun:yes'> </span>- Adapter version: v0.10.0</p>

<p class=MsoNormal style='margin-bottom:0cm;margin-bottom:.0001pt;line-height:
normal'><span style='mso-spacerun:yes'> </span>- JS-Controller version: 4.0.24</p>

<p class=MsoNormal style='margin-bottom:0cm;margin-bottom:.0001pt;line-height:
normal'><span style='mso-spacerun:yes'> </span>- Node version: v19.3.0</p>

<p class=MsoNormal style='margin-bottom:0cm;margin-bottom:.0001pt;line-height:
normal'><span style='mso-spacerun:yes'> </span>- Operating system: Linux 5.10.103-v7l+
#1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l</p>

<p class=MsoNormal style='margin-bottom:0cm;margin-bottom:.0001pt;line-height:
normal'><o:p>&nbsp;</o:p></p>

<p class=MsoNormal style='margin-bottom:0cm;margin-bottom:.0001pt;line-height:
normal'>**Additional context**<span style='mso-spacerun:yes'>  </span></p>

<p class=MsoNormal style='margin-bottom:0cm;margin-bottom:.0001pt;line-height:
normal'>Add any other context about the problem here.</p>

</body>

</html>

Pandia22, 2023-02-02:
URL (http://192.168.178.82:45000/weatherstation/data?x=y) im Browser ausgeführt.
Adapter gibt eine Warnung aus. Weiter passiert nichts.

<html>
<body>

sainlogic.0 | 2023-02-02 20:12:19.503 | warn | Listener received illegal request: (GET) /weatherstation/data?x=y
-- | -- | -- | --

</body>
</html>

Mir ist grundsätzlich unklar, wie der Adapter die Verbindung zur Sainlogic-Station aufbaut. Im Lisener-Mode hat er nur die IP-Adresse vom iobroker-Server: 0.0.0.0, den Port: 45000, den Pfad: /weatherstation/data und das Protokoll: Ecowitt. Wird die Station am Protokoll erkannt? Ich verwende auf dem Handy die App von Ecowitt. Gibt es irgendwo Restriktionen, dass die Verbindung nur zu einem Adapter aufgebaut werden kann. Wie gesagt, auf dem anderen Server läuft der Adapter ohne Probleme. Wenn ich diesen stoppe, passiert auf der anderen jedoch auch nichts. Hast du noch eine Idee?

phifogg, 2023-02-06:
Der Adapter baut zur Station nur dann eine Verbindung auf wenn der 'Scheduler' verwendet wird. Das geht aber nicht bei allen Stationen.
Bei den meisten - und das scheint auch bei Dir der Fall zu sein - läuft der Adapter im Listener Modus. D.h. die Station sendet ihre Infos an den Adapter. Das kannst Du mit der mobile App der Station einstellen. Die Station (also alle die ich bisher gesehen habe) kann aber nur an eine URL schicken. Daher kriegst Du werte nur in einem ioBroker.

Der Adapter hat noch die Funktion die empfangenen Werte an eine andere Adresse weiterzuleiten (Forwarding). Du könntest also den Empfänger-Adapter so einstellen, dass er die Daten direkt so an den zweiten Host weiterleitet.

Pandia22, 2023-02-07:
Hey Daniel, OK. Meine Station liefert nur im Listener Modus Daten an ioBroker. Forwarding wäre eine gute Idee. Wie muss die URL aussehen? Ich habe es mit der Adresse vom Admin des IOB-Salve (https://ipslave:port) probiert. Leider werden keine Daten geschickt. Die Umgehungslösung hat noch einen entscheidenden Fehler. Der IOB-Master empfängt die Daten. Bei Ausfall des Masters wird auf den Slave umgeschaltet. Da der Master offline ist, kann dieser auch das Forwarding durchführen. Hast du noch eine Idee? Kann der Datenstrom von der Station noch über andere Mittel abgezogen werden? MQTT? Home Assistant?

phifogg, 2023-04-07:
Grundsätzlich trägst du da die URL zu dem sainlogic Adapter des zweiten Hosts ein. Quasi:

http://<iobrokerhost2>:<port des sainlogic adapter, default 45000>/<url wie im adapter konfiguriert>

Es gibt wohl einen Bug derzeit dass Forwarding nicht richtig geht: #143 . Das ist noch auf meiner ToDo Liste...

phifogg, 2023-05-07:
Added to release 0.10.1

closed 2023-05-07 by phifogg