phifogg/ioBroker.sainlogic #149: last_listener_update

pp-git, 2023-02-03:
Hallo Daniel,

irgendwie hat der Adapter seit dem 10.1.2023 kein update mehr gesendet/in den DP eingetragen.
Ich brauche allerdings die Rohdaten da ich hier die  
"wh90batt": "2.88",
"ws90cap_volt": "2.5",
auslese, die speziell als Batterie und Kondensator-Akku in meiner Station verbaut sind.
{
  "PASSKEY": "",
  "stationtype": "GW2000A_V2.2.0",
  "runtime": "4641298",
   **"dateutc": "2023-01-10 08:24:34",**
  "tempinf": "90.32",
  "humidityin": "27",
  "baromrelin": "30.165",
  "baromabsin": "29.309",
  "tempf": "39.92",
  "humidity": "88",
  "winddir": "91",
  "windspeedmph": "2.24",
  "windgustmph": "9.39",
  "maxdailygust": "12.75",
  "solarradiation": "17.60",
  "uv": "0",
  "rrain_piezo": "0.000",
  "erain_piezo": "1.197",
  "hrain_piezo": "0.000",
  "drain_piezo": "0.000",
  "wrain_piezo": "1.197",
  "mrain_piezo": "1.307",
  "yrain_piezo": "1.307",
  "ws90cap_volt": "2.5",
  "ws90_ver": "119",
  "soilmoisture1": "43",
  "soilmoisture2": "59",
  "soilmoisture3": "58",
  "soilbatt1": "1.5",
  "soilbatt2": "1.4",
  "soilbatt3": "1.4",
  "wh90batt": "2.88",
  "freq": "868M",
  "model": "GW2000A"
}

phifogg, 2023-02-06:
Kannst Du ein Debug-Log anfügen? Sendet die Station noch?

pp-git, 2023-02-07:
Ja alle anderen DP's sind aktuell und funktionieren, bis auf den DP sainlogic.0.info.last_listener_update. Der steckt wie oben gesagt am 10.1.2023 fest und aktualisiert sich nicht mehr.

[image]

`2023-02-07 10:18:56.907  - [34mdebug[39m: sainlogic.0 (20417) Listener body is PASSKEY=XXXXXXXXXX&stationtype=GW2000A_V2.2.2&runtime=341117&dateutc=2023-02-07+09:18:56&tempinf=87.80&humidityin=26&baromrelin=30.629&baromabsin=29.772&tempf=31.28&humidity=70&winddir=89&windspeedmph=6.49&windgustmph=9.84&maxdailygust=17.00&solarradiation=163.85&uv=1&rrain_piezo=0.000&erain_piezo=0.000&hrain_piezo=0.000&drain_piezo=0.000&wrain_piezo=0.437&mrain_piezo=0.992&yrain_piezo=4.106&ws90cap_volt=2.8&ws90_ver=119&soilmoisture1=39&soilmoisture2=64&soilmoisture3=56&soilbatt1=1.4&soilbatt2=1.4&soilbatt3=1.4&wh90batt=2.82&freq=868M&model=GW2000A&interval=1
2023-02-07 10:18:56.907  - [34mdebug[39m: sainlogic.0 (20417) No mapping found for dataelement: PASSKEY
2023-02-07 10:18:56.908  - [34mdebug[39m: sainlogic.0 (20417) Extracting value for info.softwaretype(stationtype)
2023-02-07 10:18:56.908  - [34mdebug[39m: sainlogic.0 (20417) No mapping found for dataelement: runtime
2023-02-07 10:18:56.908  - [34mdebug[39m: sainlogic.0 (20417) No mapping found for dataelement: dateutc
2023-02-07 10:18:56.908  - [34mdebug[39m: sainlogic.0 (20417) Extracting value for weather.current.indoortemp(tempinf)
2023-02-07 10:18:56.909  - [34mdebug[39m: sainlogic.0 (20417) Extracting value for weather.current.indoorhumidity(humidityin)
2023-02-07 10:18:56.909  - [34mdebug[39m: sainlogic.0 (20417) Extracting value for weather.current.pressurerel(baromrelin)
2023-02-07 10:18:56.909  - [34mdebug[39m: sainlogic.0 (20417) Extracting value for weather.current.pressureabs(baromabsin)
2023-02-07 10:18:56.910  - [34mdebug[39m: sainlogic.0 (20417) Extracting value for weather.current.outdoortemp(tempf)
2023-02-07 10:18:56.910  - [34mdebug[39m: sainlogic.0 (20417) Extracting value for weather.current.outdoorhumidity(humidity)
2023-02-07 10:18:56.910  - [34mdebug[39m: sainlogic.0 (20417) Extracting value for weather.current.winddir(winddir)
2023-02-07 10:18:56.910  - [34mdebug[39m: sainlogic.0 (20417) Extracting value for weather.current.windspeed(windspeedmph)
2023-02-07 10:18:56.911  - [34mdebug[39m: sainlogic.0 (20417) Extracting value for weather.current.windgustspeed(windgustmph)
2023-02-07 10:18:56.911  - [34mdebug[39m: sainlogic.0 (20417) Extracting value for weather.current.maxdailygust(maxdailygust)
2023-02-07 10:18:56.911  - [34mdebug[39m: sainlogic.0 (20417) Extracting value for weather.current.solarradiation(solarradiation)
2023-02-07 10:18:56.912  - [34mdebug[39m: sainlogic.0 (20417) Extracting value for weather.current.uvi(uv)
2023-02-07 10:18:56.912  - [34mdebug[39m: sainlogic.0 (20417) Extracting value for weather.current.rain_piezo(rrain_piezo)
2023-02-07 10:18:56.912  - [34mdebug[39m: sainlogic.0 (20417) Extracting value for weather.current.eventrain_piezo(erain_piezo)
2023-02-07 10:18:56.912  - [34mdebug[39m: sainlogic.0 (20417) Extracting value for weather.current.hourlyrain_piezo(hrain_piezo)
2023-02-07 10:18:56.913  - [34mdebug[39m: sainlogic.0 (20417) Extracting value for weather.current.dailyrain_piezo(drain_piezo)
2023-02-07 10:18:56.913  - [34mdebug[39m: sainlogic.0 (20417) Extracting value for weather.current.weeklyrain_piezo(wrain_piezo)
2023-02-07 10:18:56.913  - [34mdebug[39m: sainlogic.0 (20417) Extracting value for weather.current.monthlyrain_piezo(mrain_piezo)
2023-02-07 10:18:56.914  - [34mdebug[39m: sainlogic.0 (20417) Extracting value for weather.current.yearlyrain_piezo(yrain_piezo)
2023-02-07 10:18:56.914  - [34mdebug[39m: sainlogic.0 (20417) No mapping found for dataelement: ws90cap_volt
2023-02-07 10:18:56.914  - [34mdebug[39m: sainlogic.0 (20417) No mapping found for dataelement: ws90_ver
2023-02-07 10:18:56.914  - [34mdebug[39m: sainlogic.0 (20417) Extracting value for weather.current.1.soilmoisture(soilmoisture1)
2023-02-07 10:18:56.915  - [34mdebug[39m: sainlogic.0 (20417) Extracting value for weather.current.2.soilmoisture(soilmoisture2)
2023-02-07 10:18:56.915  - [34mdebug[39m: sainlogic.0 (20417) Extracting value for weather.current.3.soilmoisture(soilmoisture3)
2023-02-07 10:18:56.915  - [34mdebug[39m: sainlogic.0 (20417) Extracting value for weather.info.1.soilbatt(soilbatt1)
2023-02-07 10:18:56.915  - [34mdebug[39m: sainlogic.0 (20417) Extracting value for weather.info.2.soilbatt(soilbatt2)
2023-02-07 10:18:56.915  - [34mdebug[39m: sainlogic.0 (20417) Extracting value for weather.info.3.soilbatt(soilbatt3)
2023-02-07 10:18:56.915  - [34mdebug[39m: sainlogic.0 (20417) No mapping found for dataelement: wh90batt
2023-02-07 10:18:56.916  - [34mdebug[39m: sainlogic.0 (20417) Extracting value for info.freq(freq)
2023-02-07 10:18:56.916  - [34mdebug[39m: sainlogic.0 (20417) Extracting value for info.model(model)
2023-02-07 10:18:56.916  - [34mdebug[39m: sainlogic.0 (20417) No mapping found for dataelement: interval
2023-02-07 10:18:56.916  - [34mdebug[39m: sainlogic.0 (20417) Setting value from data for info.softwaretype to GW2000A_V2.2.2
2023-02-07 10:18:56.916  - [34mdebug[39m: sainlogic.0 (20417) Setting value from data for weather.current.indoortemp to 31
2023-02-07 10:18:56.916  - [34mdebug[39m: sainlogic.0 (20417) Target for indoortemp unit is set: °C, using conversion: null
[... cut ...] ty":"70","winddir":"89","windspeedmph":"6.49","windgustmph":"9.84","maxdailygust":"17.00","solarradiation":"163.85","uv":"1","rrain_piezo":"0.000","erain_piezo":"0.000","hrain_piezo":"0.000","drain_piezo":"0.000","wrain_piezo":"0.437","mrain_piezo":"0.992","yrain_piezo":"4.106","ws90cap_volt":"2.8","ws90_ver":"119","soilmoisture1":"39","soilmoisture2":"64","soilmoisture3":"56","soilbatt1":"1.4","soilbatt2":"1.4","soilbatt3":"1.4","wh90batt":"2.82","freq":"868M","model":"GW2000A","interval":"1"}`

phifogg, 2023-02-07:
Ok, danke. jetzt sehe ich dass dein Listener im Ecowitt-Protokoll läuft. Ich schau mir das an.

phifogg, 2023-05-07:
Ist diese Problem noch aktuell? In meinem Adapter wird der Wert sauber gesetzt. Ggf. hab ich das behoben ohne zu wissen wie/wann :D

Welche Version hast Du im Einsatz?

pp-git, 2023-05-07:
Ja das problem besteht weiterhin, 
ich habe die v0.10.0 in der es nicht funktioniert. 
Downgrade ich auf v0.9.0 geht der DP **sainlogic.0.info.last_listener_update** problemlos
Upgrade ich wieder auf v.0.10.0 bleibt er sofort stehen und ....  siehe 1. Post

Matzebhv, 2023-05-07:
Ich habe das grade bei mir geschaut. Das letzte mal wurde der DP hier am 28.03. aktualisiert. An diesem Datum habe ich meine defekte Station ausgetauscht und dabei das Protokoll von WU nach Ecowitt gewechselt.

Gruß
Matthias

phifogg, 2023-05-08:
Danke... das war des Rätsels Lösung! Fehler ist gefunden, update kommt vermutlich im Laufe des Tages!

closed 2023-05-08 by phifogg

pp-git, 2023-05-08:
Danke Dir, funktioniert wieder, last_listener_update bringt wieder aktuelle Daten

elek76, 2023-09-03:
Bei mir wird last_listener_update wieder nicht aktualisiert (habe den aktuellen Code von GitHub installiert).
Grund scheint das hier zu sein:
https://github.com/phifogg/ioBroker.sainlogic/commit/b75e53c5ba264b6d8807a1cefaaf2d6580c67651#r126304701