phifogg/ioBroker.sainlogic #150: dnt WeatherScreen Pro

RasputinBerry, 2023-03-03:
Hallo phifogg,

o.g. Wetterstation ist in der Adapter Beschreibung nicht aufgeführt ... Ich habe trotzdem versucht sie einzubinden. 
Sie wird erkannt und die Datenpunkte werden auch erstellt aber sie bleiben leer.
Wird sie wirklich nicht unterstützt oder muss man bei den Einstellungen etwas "besonderes" beachten ?
(Ich habe unter "ZEITPLANEINSTELLUNGEN" die IP, das Intervall=30 eingetragen, Häkchen "Aktiv" gesetzt ... sonst nix)

Im Protokoll kommt das:

[image]

Danke für die Erleuchtung, Grüße.

RasputinBerry, 2023-03-05:
EDIT:

Ich habe mich jetzt "eingelesen" und den Listener mode (mit div. Einstellungen try&error) konfiguriert ... jetzt sehe ich aktuelle Daten ...

closed 2023-03-05 by RasputinBerry

phifogg, 2023-03-05:
Super, danke für das Update. Das wäre auch meine Empfehlung gewesen. 
Um im Scheduler mode mit der Station zu reden muss ich deren API kennen. Da gibt es von den wenigsten Herstellern eine öffentliche Doku,