phifogg/ioBroker.sainlogic #155: Feature Request: Unused DataPoints

EvilEls, 2023-04-06:
Hi,
currently the adapter creates a default set of datapoints, even though the weatherstation does not support them.
For my WS3500 roughly half of the created DPs are not used, showing NULL.

Please only create DPs if there is actually data to import - or auto remove DPs still showing NULL after x data imports.

Thanks

phifogg, 2023-04-07:
Good point, this will take some rework on the logic to know which data ports are required or not.

phifogg, 2023-07-23:
Version 0.11.1 will only create needed data points.
If you want to test, install via guthub from deb branch:
https://github.com/phifogg/ioBroker.sainlogic/tree/dev

Delete all states from sainlogic, start adapter. It will only create the info states at startup, all others as data is recieved.

phifogg, 2023-08-06:
Available as now on master branch v 0.11.1

closed 2023-08-09 by phifogg