phifogg/ioBroker.sainlogic #156: String expected, number delivered

Nackensteak, 2023-04-06:
Hi, 
ich bin neu im IOBroker und dem ganzen Kram, denke aber ich habe einen Bug gefunden.
Ich habe die WS3500 Sainlogic Station seit ein paar Tagen im Einsatz mit der neusten Firmware 2.0.38
Außerdem den aktuellen Adapter 0.10.0
Die Station an sich funktioniert, das Display zeigt alles an.

Ich bekomme aber einfach keine Daten in den IOBroker
Ich habe Ecowitt sowie Wunderground versucht.
Ich habe den Pfad mit und ohne "?" versucht. 
Ich habe die IP 0.0.0.0 oder direkt die vom Broker versucht.
Ich habe den Adapter bereits einmal gelöscht und wieder installiert. 

Es sieht aus als kommen Daten-Updates jede Minute, aber alle Werte sind "null".
[image]

Ich erhalte folgenden Log:
sainlogic.0 | 12719 | 2023-04-06 15:58:17.845 | warn | This object will not be created in future versions. Please report this to the developer.
sainlogic.0 | 12719 | 2023-04-06 15:58:17.844 | warn | Object sainlogic.0.info.last_listener_update is invalid: Default value has to be type "string" but receive

[image]
[image]

Für Hilfe und Tips wäre ich dankbar.

phifogg, 2023-04-07:
Bei Server-IP in der EasyWeather-App solltest Du den Port 8081 weglassen. Hier muss nur die IP des IOBroker hosts rein.

Nackensteak, 2023-04-07:
Danke, probiere ich aus! Weil da IP/Host stand dachte ich der Port ist gemeint. Gruß, Sascha--Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.

Nackensteak, 2023-04-07:
Mit Ecowitt hab ich es jetzt zum laufe bekommen, danke. --Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.

phifogg, 2023-05-07:
Closed - working.

closed 2023-05-07 by phifogg