phifogg/ioBroker.sainlogic #157: ECOWITT  GW1101AE

Wulfgar77, 2023-04-18:
Hi guys,

I have mounted a new ECOWITT GW1101AE weather station on my observatory and integrated it in the adapter Sainlogic 0.10.0. Additionally I have integrated a soil moisture sensor WH51A on the weather station.

IOroker: 4.0.24
Node.js 16.19.1
NPM 8.19.3

So far everything works fine. Unfortunately I see the same error messages in the IObroker log.

<html>
<body>

sainlogic.0 | 2023-04-18 18:57:34.858 | debug | FW Scheduler Connection closed
sainlogic.0 | 2023-04-18 18:57:34.858 | error | Scheduler received data for unkown command 91 subcommand 97
sainlogic.0 | 2023-04-18 18:57:34.857 | debug | Data Command received: 91 subcommand 97
sainlogic.0 | 2023-04-18 18:57:34.857 | debug | FW Scheduler Received data string: ffff5b040261
sainlogic.0 | 2023-04-18 18:57:34.848 | debug | Scheduler connected to weather station run 255,255,11,0,6,8,8,33
sainlogic.0 | 2023-04-18 18:57:34.848 | error | Scheduler received data for unkown command 91 subcommand 97
sainlogic.0 | 2023-04-18 18:57:34.847 | debug | Data Command received: 91 subcommand 97
sainlogic.0 | 2023-04-18 18:57:34.847 | debug | FW Scheduler Received data string: ffff5b040261
sainlogic.0 | 2023-04-18 18:57:34.838 | debug | Scheduler connected to weather station run 255,255,11,0,6,7,7,31
sainlogic.0 | 2023-04-18 18:57:34.838 | error | Scheduler received data for unkown command 91 subcommand 97
sainlogic.0 | 2023-04-18 18:57:34.837 | debug | Data Command received: 91 subcommand 97
sainlogic.0 | 2023-04-18 18:57:34.837 | debug | FW Scheduler Received data string: ffff5b040261
sainlogic.0 | 2023-04-18 18:57:34.828 | debug | Scheduler connected to weather station run 255,255,11,0,6,6,6,29
sainlogic.0 | 2023-04-18 18:57:34.828 | error | Scheduler received data for unkown command 91 subcommand 97
sainlogic.0 | 2023-04-18 18:57:34.828 | debug | Data Command received: 91 subcommand 97
sainlogic.0 | 2023-04-18 18:57:34.827 | debug | FW Scheduler Received data string: ffff5b040261
sainlogic.0 | 2023-04-18 18:57:34.819 | debug | Scheduler connected to weather station run 255,255,11,0,6,5,5,27
sainlogic.0 | 2023-04-18 18:57:34.819 | error | Scheduler received data for unkown command 91 subcommand 97
sainlogic.0 | 2023-04-18 18:57:34.819 | debug | Data Command received: 91 subcommand 97
sainlogic.0 | 2023-04-18 18:57:34.819 | debug | FW Scheduler Received data string: ffff5b040261
sainlogic.0 | 2023-04-18 18:57:34.808 | debug | Scheduler connected to weather station run 255,255,11,0,6,4,4,25
sainlogic.0 | 2023-04-18 18:57:34.808 | error | Scheduler received data for unkown command 80 subcommand 71
sainlogic.0 | 2023-04-18 18:57:34.808 | debug | Data Command received: 80 subcommand 71
sainlogic.0 | 2023-04-18 18:57:34.807 | debug | FW Scheduler Received data string: ffff50120e475731313030415f56322e322e33b9
sainlogic.0 | 2023-04-18 18:57:34.796 | debug | Scheduler connected to weather station run 255,255,80,3,83
sainlogic.0 | 2023-04-18 18:57:34.790 | info | Scheduler pulling for new data
sainlogic.0 | 2023-04-18 18:57:19.864 | debug | FW Scheduler Connection closed
sainlogic.0 | 2023-04-18 18:57:19.863 | error | Scheduler received data for unkown command 91 subcommand 97
sainlogic.0 | 2023-04-18 18:57:19.863 | debug | Data Command received: 91 subcommand 97
sainlogic.0 | 2023-04-18 18:57:19.862 | debug | FW Scheduler Received data string: ffff5b040261
sainlogic.0 | 2023-04-18 18:57:19.853 | debug | Scheduler connected to weather station run 255,255,11,0,6,8,8,33
sainlogic.0 | 2023-04-18 18:57:19.853 | error | Scheduler received data for unkown command 91 subcommand 97
sainlogic.0 | 2023-04-18 18:57:19.853 | debug | Data Command received: 91 subcommand 97
sainlogic.0 | 2023-04-18 18:57:19.853 | debug | FW Scheduler Received data string: ffff5b040261
sainlogic.0 | 2023-04-18 18:57:19.843 | debug | Scheduler connected to weather station run 255,255,11,0,6,7,7,31
sainlogic.0 | 2023-04-18 18:57:19.843 | error | Scheduler received data for unkown command 91 subcommand 97
sainlogic.0 | 2023-04-18 18:57:19.843 | debug | Data Command received: 91 subcommand 97
sainlogic.0 | 2023-04-18 18:57:19.842 | debug | FW Scheduler Received data string: ffff5b040261
sainlogic.0 | 2023-04-18 18:57:19.833 | debug | Scheduler connected to weather station run 255,255,11,0,6,6,6,29
sainlogic.0 | 2023-04-18 18:57:19.833 | error | Scheduler received data for unkown command 91 subcommand 97
sainlogic.0 | 2023-04-18 18:57:19.833 | debug | Data Command received: 91 subcommand 97
sainlogic.0 | 2023-04-18 18:57:19.832 | debug | FW Scheduler Received data string: ffff5b040261
sainlogic.0 | 2023-04-18 18:57:19.823 | debug | Scheduler connected to weather station run 255,255,11,0,6,5,5,27
sainlogic.0 | 2023-04-18 18:57:19.823 | error | Scheduler received data for unkown command 91 subcommand 97
sainlogic.0 | 2023-04-18 18:57:19.823 | debug | Data Command received: 91 subcommand 97
sainlogic.0 | 2023-04-18 18:57:19.822 | debug | FW Scheduler Received data string: ffff5b040261
sainlogic.0 | 2023-04-18 18:57:19.813 | debug | Scheduler connected to weather station run 255,255,11,0,6,4,4,25
sainlogic.0 | 2023-04-18 18:57:19.813 | error | Scheduler received data for unkown command 80 subcommand 71
sainlogic.0 | 2023-04-18 18:57:19.813 | debug | Data Command received: 80 subcommand 71
sainlogic.0 | 2023-04-18 18:57:19.812 | debug | FW Scheduler Received data string: ffff50120e475731313030415f56322e322e33b9
sainlogic.0 | 2023-04-18 18:57:19.794 | debug | Scheduler connected to weather station run 255,255,80,3,83
sainlogic.0 | 2023-04-18 18:57:19.789 | info | Scheduler pulling for new data

I hope you can help me.
Thanks for the adapter and the great job you do here.

BR.

phifogg, 2023-04-22:
Hi Wulfgar,
it loosk to me as if your weatherstation does not react as expected to the scheduler commands.  Without a proper API doc I can't really tell what is wrong there and would recommend to use the listener mode instead.

If you can find an example on how to 'read' the data returned by the scheduler commands, I can surely add it.
Daniel

Mychel60, 2023-05-03:
Hi all,
I have a GW1100A_V2.2.3 sensor installed. It works under my apple phone. I also installed the sainlogic adapter on IObroker. It has a green status but I cannot retrieve data at all. Does anyone has an idea whats wrong here? Thanks for comments
[image]

phifogg, 2023-05-04:
You need to configure your station to send the data to ioBroker using ECOWITT or WU protocoll. For the adapter this is called 'listener' mode as it listens for updates.

Wulfgar77, 2023-05-04:
> Hi Wulfgar, it loosk to me as if your weatherstation does not react as expected to the scheduler commands. Without a proper API doc I can't really tell what is wrong there and would recommend to use [...]

Ok thanks for the info. I will check with Ecowitt for the API document. If I have something new I will inform you. Thanks for the great adapter, since most of it works.

phifogg, 2023-05-07:
I'll close this issue for now as 'works as designed'. Come back if you still have issues.

closed 2023-05-07 by phifogg

Mychel60, 2023-05-07:
Thanks a lot.  will come back to you later.Sent from my I-Phone

Mychel60, 2024-02-01:
Hi,

I have big problems to bring my Weatherstation *GW1100A_V2.3.*1  up and
running. It should run under IOBroker. I followed the forum, but I could
not solve the issue successfully.

The device is in Listener mode; Scheduler is switched off. I also use the
ECOWITT set up. It works great with my Iphone, but I have absolutely no
access with IOB. Here the settings:
[image: image.png][image: image.png]
[image: image.png]
Do I need "port sharing " in Fritz-Box?

Thanks a lot for your help

Michael

____________________________________________________________________________

Michael Richter
Dipl-Ing(FH)
St.-Gundekar-Str. 9
86650 Wemding
Mobil: +49-172 3576351
Privat: +49-9092-1442
Mail: ***@***.***
www: mr-electroservice.com

phifogg, 2024-02-02:
The images didn't come through though. Sorry. 
Can you put the adapter in debug mode and send some log files?
When in debug, you can also try any browser and hit the URL http://<iobroker>:<port>/<urlprefix>. This should trigger a message like 'invalid listener input' or so.

Mychel60, 2024-02-03:
Thanks for feedback.👍😁 I will come back to you in the coming days.RegdsMichaelSent from my I-Phone

When in debug, you can also try any browser and hit the URL http://:/. This should trigger a message like 'invalid listener input' or so.