phifogg/ioBroker.sainlogic #162: weather.daily.lightning_num

Michel83, 2023-05-13:
Hi, leider wird der neue Datenpunkt nicht aktualisiert heute 8 Blitze aber nicht im Adapter zu sehen.

sainlogic.0.weather.daily.lightning_num

[image]

Grüsse

phifogg, 2023-05-20:
Wäre natürlich ein Log gut wo die Daten kommen, dann kann ich sehen was passiert. Ideal im Debug modus.

Michel83, 2023-05-22:
Hi,

reicht das ?

```
<html><body>
<!--StartFragment-->

2023-05-22 16:47:18.026  - debug: sainlogic.0 (25142) Extracting value for info.last_listener_update(last_listener_update)
--
2023-05-22 16:47:18.026  - debug: sainlogic.0 (25142) Setting value from data for info.softwaretype to WH2650A_V1.7.6
2023-05-22 16:47:18.028  - debug: sainlogic.0 (25142) Setting value from data for weather.current.indoortemp to 22.6
2023-05-22 16:47:18.029  - debug: sainlogic.0 (25142) Target for indoortemp unit is set: °C, using conversion: null
2023-05-22 16:47:18.029  - debug: sainlogic.0 (25142) Setting value from data for weather.current.indoorhumidity to 59
2023-05-22 16:47:18.030  - debug: sainlogic.0 (25142) Setting value from data for weather.current.pressurerel to 1081.8
2023-05-22 16:47:18.030  - debug: sainlogic.0 (25142) Target for pressurerel unit is set: hPa, using conversion: null
2023-05-22 16:47:18.030  - debug: sainlogic.0 (25142) Setting value from data for weather.current.pressureabs to 995
2023-05-22 16:47:18.031  - debug: sainlogic.0 (25142) Target for pressureabs unit is set: hPa, using conversion: null
2023-05-22 16:47:18.033  - debug: sainlogic.0 (25142) Setting value from data for weather.current.outdoortemp to 26.3
2023-05-22 16:47:18.034  - debug: sainlogic.0 (25142) Target for outdoortemp unit is set: °C, using conversion: null
2023-05-22 16:47:18.034  - debug: sainlogic.0 (25142) Setting value from data for weather.current.outdoorhumidity to 48
2023-05-22 16:47:18.035  - debug: sainlogic.0 (25142) Setting value from data for weather.current.winddir to 264
2023-05-22 16:47:18.036  - debug: sainlogic.0 (25142) Setting value from data for weather.current.windspeed to 1.4
2023-05-22 16:47:18.037  - debug: sainlogic.0 (25142) Target for windspeed unit is set: km/h, using conversion: null
2023-05-22 16:47:18.038  - debug: sainlogic.0 (25142) Setting value from data for weather.current.windgustspeed to 3.6
2023-05-22 16:47:18.039  - debug: sainlogic.0 (25142) Target for windgustspeed unit is set: km/h, using conversion: null
2023-05-22 16:47:18.039  - debug: sainlogic.0 (25142) Setting value from data for weather.current.maxdailygust to 9.4
2023-05-22 16:47:18.040  - debug: sainlogic.0 (25142) Target for maxdailygust unit is set: km/h, using conversion: null
2023-05-22 16:47:18.040  - debug: sainlogic.0 (25142) Setting value from data for weather.current.solarradiation to 11811
2023-05-22 16:47:18.043  - debug: sainlogic.0 (25142) Target for solarradiation unit is set: lx, using conversion: null
2023-05-22 16:47:18.044  - debug: sainlogic.0 (25142) Setting value from data for weather.current.uvi to 0
2023-05-22 16:47:18.044  - debug: sainlogic.0 (25142) Setting value from data for weather.current.rain to 0
2023-05-22 16:47:18.045  - debug: sainlogic.0 (25142) Target for rain unit is set: mm, using conversion: null
2023-05-22 16:47:18.045  - debug: sainlogic.0 (25142) Setting value from data for weather.current.dailyrain to 0
2023-05-22 16:47:18.046  - debug: sainlogic.0 (25142) Target for dailyrain unit is set: mm, using conversion: null
2023-05-22 16:47:18.047  - debug: sainlogic.0 (25142) Setting value from data for weather.current.weeklyrain to 0
2023-05-22 16:47:18.048  - debug: sainlogic.0 (25142) Target for weeklyrain unit is set: mm, using conversion: null
2023-05-22 16:47:18.048  - debug: sainlogic.0 (25142) Setting value from data for weather.current.monthlyrain to 97.2
2023-05-22 16:47:18.049  - debug: sainlogic.0 (25142) Target for monthlyrain unit is set: mm, using conversion: null
2023-05-22 16:47:18.049  - debug: sainlogic.0 (25142) Setting value from data for weather.current.yearlyrain to 460.7
2023-05-22 16:47:18.050  - debug: sainlogic.0 (25142) Target for yearlyrain unit is set: mm, using conversion: null
2023-05-22 16:47:18.052  - debug: sainlogic.0 (25142) Setting value from data for weather.current.totalrainin to 460.7
2023-05-22 16:47:18.052  - debug: sainlogic.0 (25142) Target for totalrainin unit is set: mm, using conversion: null
2023-05-22 16:47:18.053  - debug: sainlogic.0 (25142) Setting value from data for weather.current.1.outdoortemp to 21.1
2023-05-22 16:47:18.053  - debug: sainlogic.0 (25142) Target for outdoortemp unit is set: °C, using conversion: null
2023-05-22 16:47:18.054  - debug: sainlogic.0 (25142) Setting value from data for weather.current.1.outdoorhumidity to 62
2023-05-22 16:47:18.054  - debug: sainlogic.0 (25142) Setting value from data for weather.current.2.outdoortemp to 16.9
2023-05-22 16:47:18.055  - debug: sainlogic.0 (25142) Target for outdoortemp unit is set: °C, using conversion: null
2023-05-22 16:47:18.056  - debug: sainlogic.0 (25142) Setting value from data for weather.current.2.outdoorhumidity to 70
2023-05-22 16:47:18.057  - debug: sainlogic.0 (25142) Setting value from data for weather.current.1.soilmoisture to 25
2023-05-22 16:47:18.057  - debug: sainlogic.0 (25142) Setting value from data for weather.current.lightning to 17
2023-05-22 16:47:18.058  - debug: sainlogic.0 (25142) Setting value from data for weather.current.lightning_time to 1684766190
2023-05-22 16:47:18.058  - debug: sainlogic.0 (25142) Setting value from data for weather.info.wh25batt to 0
2023-05-22 16:47:18.058  - debug: sainlogic.0 (25142) Setting value from data for weather.info.1.soilbatt to 1.3
2023-05-22 16:47:18.059  - debug: sainlogic.0 (25142) Setting value from data for info.freq to 868M
[... cut ...] 0","rainratein":"0.000","eventrainin":"0.000","hourlyrainin":"0.000","dailyrainin":"0.000","weeklyrainin":"0.000","monthlyrainin":"3.827","yearlyrainin":"18.138","totalrainin":"18.138","temp1f":"69.98","humidity1":"62","temp2f":"62.42","humidity2":"70","soilmoisture1":"25","lightning_num":"7","lightning":"17","lightning_time":"1684766190","wh65batt":"0","wh25batt":"0","batt1":"0","batt2":"0","soilbatt1":"1.3","wh57batt":"5","freq":"868M","model":"WH2650A"}

<!--EndFragment-->
</body>
</html>
```

phifogg, 2023-05-23:
Steht aber doch da er würde es machen:
sainlogic.0 | 2023-05-22 16:52:12.033 | debug | Setting value from data for weather.current.lightning to 17
Und das geht nicht?

phifogg, 2023-05-23:
Ah.. könnte sein das es von lightning_num überschrieben wird. Könnte sein, dass ich beide auf den selben State schreibe, das wäre blöd.

Michel83, 2023-05-23:
Hi, also der Datenpunkt bleibt leer 

[image]

Michel83, 2023-06-08:
Funktioniert super, Danke und Grüsse