phifogg/ioBroker.sainlogic #167: Several type mismatches.

Grothesk242, 2023-06-02:
``
2023-06-02 23:48:56.298  - warn: sainlogic.0 (709070) Object sainlogic.0.info.model is invalid: Default value has to be type "string" but received type "number"
2023-06-02 23:48:56.298  - warn: sainlogic.0 (709070) This object will not be created in future versions. Please report this to the developer.
2023-06-02 23:48:56.299  - warn: sainlogic.0 (709070) Object sainlogic.0.info.freq is invalid: Default value has to be type "string" but received type "number"
2023-06-02 23:48:56.299  - warn: sainlogic.0 (709070) This object will not be created in future versions. Please report this to the developer.
2023-06-02 23:48:56.300  - warn: sainlogic.0 (709070) Object sainlogic.0.info.last_listener_update is invalid: Default value has to be type "string" but received type "number"
2023-06-02 23:48:56.300  - warn: sainlogic.0 (709070) This object will not be created in future versions. Please report this to the developer.
2023-06-02 23:48:56.741  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.info.model" has to be type "string" but received type "number"
2023-06-02 23:48:56.742  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.info.freq" has to be type "string" but received type "number"
2023-06-02 23:48:56.743  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.info.last_listener_update" has to be type "string" but received type "number"
2023-06-02 23:48:56.979  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.current.dailyrain" has to be type "number" but received type "string"
2023-06-02 23:48:56.984  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.maxvalues.absolut.dailyrain" has to be type "number" but received type "string"
2023-06-02 23:48:56.985  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.maxvalues.daily.dailyrain" has to be type "number" but received type "string"
2023-06-02 23:48:56.986  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.current.dailyrain_piezo" has to be type "number" but received type "string"
2023-06-02 23:48:56.988  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.current.indoorhumidity" has to be type "number" but received type "string"
2023-06-02 23:48:56.989  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.maxvalues.absolut.indoorhumidity" has to be type "number" but received type "string"
2023-06-02 23:48:56.990  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.maxvalues.daily.indoorhumidity" has to be type "number" but received type "string"
2023-06-02 23:48:56.991  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.minvalues.absolut.indoorhumidity" has to be type "number" but received type "string"
2023-06-02 23:48:56.992  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.minvalues.daily.indoorhumidity" has to be type "number" but received type "string"
2023-06-02 23:48:56.992  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.current.indoortemp" has to be type "number" but received type "string"
2023-06-02 23:48:56.993  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.maxvalues.absolut.indoortemp" has to be type "number" but received type "string"
2023-06-02 23:48:56.994  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.maxvalues.daily.indoortemp" has to be type "number" but received type "string"
2023-06-02 23:48:56.995  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.minvalues.absolut.indoortemp" has to be type "number" but received type "string"
2023-06-02 23:48:56.996  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.minvalues.daily.indoortemp" has to be type "number" but received type "string"
2023-06-02 23:48:56.997  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.current.monthlyrain" has to be type "number" but received type "string"
2023-06-02 23:48:56.998  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.maxvalues.absolut.monthlyrain" has to be type "number" but received type "string"
2023-06-02 23:48:56.998  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.maxvalues.daily.monthlyrain" has to be type "number" but received type "string"
2023-06-02 23:48:56.999  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.current.monthlyrain_piezo" has to be type "number" but received type "string"
2023-06-02 23:48:57.000  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.current.outdoorhumidity" has to be type "number" but received type "string"
2023-06-02 23:48:57.001  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.maxvalues.absolut.outdoorhumidity" has to be type "number" but received type "string"
2023-06-02 23:48:57.001  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.maxvalues.daily.outdoorhumidity" has to be type "number" but received type "string"
2023-06-02 23:48:57.002  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.minvalues.absolut.outdoorhumidity" has to be type "number" but received type "string"
2023-06-02 23:48:57.003  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.minvalues.daily.outdoorhumidity" has to be type "number" but received type "string"
2023-06-02 23:48:57.004  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.current.outdoortemp" has to be type "number" but received type "string"
2023-06-02 23:48:57.004  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.maxvalues.absolut.outdoortemp" has to be type "number" but received type "string"
2023-06-02 23:48:57.005  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.maxvalues.daily.outdoortemp" has to be type "number" but received type "string"
2023-06-02 23:48:57.006  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.minvalues.absolut.outdoortemp" has to be type "number" but received type "string"
2023-06-02 23:48:57.007  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.minvalues.daily.outdoortemp" has to be type "number" but received type "string"
2023-06-02 23:48:57.007  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.current.pressureabs" has to be type "number" but received type "string"
2023-06-02 23:48:57.008  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.maxvalues.absolut.pressureabs" has to be type "number" but received type "string"
2023-06-02 23:48:57.009  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.maxvalues.daily.pressureabs" has to be type "number" but received type "string"
2023-06-02 23:48:57.011  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.minvalues.absolut.pressureabs" has to be type "number" but received type "string"
2023-06-02 23:48:57.012  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.minvalues.daily.pressureabs" has to be type "number" but received type "string"
2023-06-02 23:48:57.013  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.current.pressurerel" has to be type "number" but received type "string"
2023-06-02 23:48:57.014  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.maxvalues.absolut.pressurerel" has to be type "number" but received type "string"
2023-06-02 23:48:57.015  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.maxvalues.daily.pressurerel" has to be type "number" but received type "string"
2023-06-02 23:48:57.016  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.minvalues.absolut.pressurerel" has to be type "number" but received type "string"
2023-06-02 23:48:57.017  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.minvalues.daily.pressurerel" has to be type "number" but received type "string"
2023-06-02 23:48:57.018  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.current.rain" has to be type "number" but received type "string"
2023-06-02 23:48:57.019  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.maxvalues.absolut.rain" has to be type "number" but received type "string"
[... cut ...]
2023-06-02 23:48:57.254  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.daily.lightning_num" has to be type "number" but received type "string"
2023-06-02 23:48:57.254  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.current.eventrain_piezo" has to be type "number" but received type "string"
2023-06-02 23:48:57.255  - info: sainlogic.0 (709070) State value to set for "sainlogic.0.weather.current.hourlyrain_piezo" has to be type "number" but received type "string"
``

phifogg, 2023-06-04:
That is a bit weird. Can you post your config? Like is it using ecowitt or weatherunderground?

Grothesk242, 2023-06-04:
I'm using ecowit protocol. Will send a complete restart of the adapter later. The above is from a complete restart of the whole iobroker. Maybe a a dependent adapter hadn't been startet yet?

Grothesk242, 2023-06-05:
Settings are:

```
{
  "_id": "system.adapter.sainlogic.0",
  "common": {
    "name": "sainlogic",
    "version": "0.10.3",
    "title": "Sainlogic Weatherstation",
    "authors": [
      "Fogg <foggch@gmail.com>"
    ],
    "keywords": [
      "weather",
      "sainlogic"
    ],
    "license": "MIT",
    "platform": "Javascript/Node.js",
    "main": "main.js",
    "icon": "sainlogic.png",
    "enabled": true,
    "extIcon": "https://raw.githubusercontent.com/phifogg/ioBroker.sainlogic/master/admin/sainlogic.png",
    "readme": "https://github.com/phifogg/ioBroker.sainlogic/blob/master/README.md",
    "loglevel": "info",
    "mode": "daemon",
    "type": "weather",
    "compact": true,
    "connectionType": "local",
    "dataSource": "push",
    "materialize": true,
    "dependencies": [
      {
        "js-controller": ">=2.0.0"
      }
    ],
    "installedVersion": "0.10.3",
    "host": "chet",
    "logLevel": "info",
    "installedFrom": "iobroker.sainlogic@0.10.3"
  },
  "native": {
    "listener_active": true,
    "bind": "0.0.0.0",
    "port": "45000",
    "path": "/data/report/",
    "listener_protocol": "ecowitt",
    "listener_forward_url": "",
    "scheduler_active": false,
    "ws_address": "192.168.178.22",
    "ws_port": "45000",
    "ws_freq": "30",
    "ws_getfirmware": true,
    "ws_getcurrent": true,
    "ws_getmax": true,
    "ws_getmin": true,
    "ws_getmaxdaily": true,
    "ws_getmindaily": true,
    "unit_rain": "mm",
    "unit_temperature": "°C",
    "unit_pressure": "hPa",
    "unit_windspeed": "km/h",
    "unit_solar": "lx"
  }
}
```

Restarting looks like this:

```
2023-06-05 06:49:20.832  - info: sainlogic.0 (886895) starting. Version 0.10.3 in /opt/iobroker/node_modules/iobroker.sainlogic, node: v18.16.0, js-controller: 5.0.1-alpha.0-20230525-c580eb7e
[... 61 lines cut ...]
2023-06-05 06:49:21.051  - info: sainlogic.0 (886895) State value to set for "sainlogic.0.weather.current.totalrainin" has to be type "number" but received type "string"
2023-06-05 06:49:21.052  - info: sainlogic.0 (886895) State value to set for "sainlogic.0.weather.maxvalues.absolut.totalrainin" has to be type "number" but received type "string"
2023-06-05 06:49:21.053  - info: sainlogic.0 (886895) State value to set for "sainlogic.0.weather.current.yearlyrain_piezo" has to be type "number" but received type "string"
2023-06-05 06:49:21.053  - info: sainlogic.0 (886895) State value to set for "sainlogic.0.weather.current.soilmoisture" has to be type "number" but received type "string"
2023-06-05 06:49:21.054  - info: sainlogic.0 (886895) State value to set for "sainlogic.0.weather.info.soilbatt" has to be type "number" but received type "string"
2023-06-05 06:49:21.055  - info: sainlogic.0 (886895) State value to set for "sainlogic.0.weather.info.wh80batt" has to be type "number" but received type "string"
2023-06-05 06:49:21.056  - info: sainlogic.0 (886895) State value to set for "sainlogic.0.weather.info.wh40batt" has to be type "number" but received type "string"
2023-06-05 06:49:21.057  - info: sainlogic.0 (886895) State value to set for "sainlogic.0.weather.info.wh25batt" has to be type "number" but received type "string"
2023-06-05 06:49:21.058  - info: sainlogic.0 (886895) State value to set for "sainlogic.0.weather.current.tf_ch" has to be type "number" but received type "string"
2023-06-05 06:49:21.058  - info: sainlogic.0 (886895) State value to set for "sainlogic.0.weather.info.tf_batt" has to be type "number" but received type "string"
2023-06-05 06:49:21.059  - info: sainlogic.0 (886895) State value to set for "sainlogic.0.weather.current.tf_co2" has to be type "number" but received type "string"
2023-06-05 06:49:21.060  - info: sainlogic.0 (886895) State value to set for "sainlogic.0.weather.current.humi_co2" has to be type "number" but received type "string"
2023-06-05 06:49:21.061  - info: sainlogic.0 (886895) State value to set for "sainlogic.0.weather.current.pm25_co2" has to be type "number" but received type "string"
2023-06-05 06:49:21.062  - info: sainlogic.0 (886895) State value to set for "sainlogic.0.weather.daily.pm25_24h_co2" has to be type "number" but received type "string"
2023-06-05 06:49:21.062  - info: sainlogic.0 (886895) State value to set for "sainlogic.0.weather.current.pm10_co2" has to be type "number" but received type "string"
2023-06-05 06:49:21.109  - info: sainlogic.0 (886895) State value to set for "sainlogic.0.weather.daily.pm10_24h_co2" has to be type "number" but received type "string"
2023-06-05 06:49:21.110  - info: sainlogic.0 (886895) State value to set for "sainlogic.0.weather.current.co2" has to be type "number" but received type "string"
2023-06-05 06:49:21.222  - info: sainlogic.0 (886895) State value to set for "sainlogic.0.weather.daily.co2_24h" has to be type "number" but received type "string"
2023-06-05 06:49:21.233  - info: sainlogic.0 (886895) State value to set for "sainlogic.0.weather.info.co2_batt" has to be type "number" but received type "string"
2023-06-05 06:49:21.234  - info: sainlogic.0 (886895) State value to set for "sainlogic.0.weather.current.lightning_time" has to be type "number" but received type "string"
2023-06-05 06:49:21.246  - info: sainlogic.0 (886895) State value to set for "sainlogic.0.weather.current.lightning" has to be type "number" but received type "string"
2023-06-05 06:49:21.251  - info: sainlogic.0 (886895) State value to set for "sainlogic.0.weather.daily.lightning_num" has to be type "number" but received type "string"
2023-06-05 06:49:21.253  - info: sainlogic.0 (886895) State value to set for "sainlogic.0.weather.current.eventrain_piezo" has to be type "number" but received type "string"
2023-06-05 06:49:21.254  - info: sainlogic.0 (886895) State value to set for "sainlogic.0.weather.current.hourlyrain_piezo" has to be type "number" but received type "string"
```

mcm1957, 2023-06-16:
Ic can confirm these info / errors:

Adapter 1.10.3
js-controller 4.0.24

running in lister mode

```
2023-06-16 18:12:38.860 - info: sainlogic.0 (25805) starting. Version 0.10.3 in /opt/iobroker/node_modules/iobroker.sainlogic, node: v16.19.0, js-controller: 4.0.24
--
2023-06-16 18:12:38.908 - info: sainlogic.0 (25805) Starting Listener
2023-06-16 18:12:38.958 - info: sainlogic.0 (25805) Creating new data point: info.last_listener_update
2023-06-16 18:12:38.960 - info: sainlogic.0 (25805) Creating new data point: weather.current.dailyrain
2023-06-16 18:12:38.961 - info: sainlogic.0 (25805) Creating new data point: weather.maxvalues.absolut.dailyrain
2023-06-16 18:12:38.962 - info: sainlogic.0 (25805) Creating new data point: weather.maxvalues.daily.dailyrain
2023-06-16 18:12:38.962 - info: sainlogic.0 (25805) Creating new data point: weather.current.dailyrain_piezo
2023-06-16 18:12:38.963 - info: sainlogic.0 (25805) Creating new data point: weather.current.dewpointtemp
2023-06-16 18:12:38.964 - info: sainlogic.0 (25805) Creating new data point: weather.maxvalues.absolut.dewpointtemp
2023-06-16 18:12:38.965 - info: sainlogic.0 (25805) Creating new data point: weather.maxvalues.daily.dewpointtemp
2023-06-16 18:12:38.965 - info: sainlogic.0 (25805) Creating new data point: weather.minvalues.absolut.dewpointtemp
2023-06-16 18:12:38.966 - info: sainlogic.0 (25805) Creating new data point: weather.minvalues.daily.dewpointtemp
2023-06-16 18:12:38.967 - info: sainlogic.0 (25805) Creating new data point: weather.current.indoorhumidity
2023-06-16 18:12:38.967 - info: sainlogic.0 (25805) Creating new data point: weather.maxvalues.absolut.indoorhumidity
2023-06-16 18:12:38.968 - info: sainlogic.0 (25805) Creating new data point: weather.maxvalues.daily.indoorhumidity
2023-06-16 18:12:38.969 - info: sainlogic.0 (25805) Creating new data point: weather.minvalues.absolut.indoorhumidity
2023-06-16 18:12:38.969 - info: sainlogic.0 (25805) Creating new data point: weather.minvalues.daily.indoorhumidity
2023-06-16 18:12:38.969 - info: sainlogic.0 (25805) Creating new data point: weather.current.indoortemp
2023-06-16 18:12:38.970 - info: sainlogic.0 (25805) Creating new data point: weather.maxvalues.absolut.indoortemp
2023-06-16 18:12:38.970 - info: sainlogic.0 (25805) Creating new data point: weather.maxvalues.daily.indoortemp
2023-06-16 18:12:38.970 - info: sainlogic.0 (25805) Creating new data point: weather.minvalues.absolut.indoortemp
2023-06-16 18:12:38.971 - info: sainlogic.0 (25805) Creating new data point: weather.minvalues.daily.indoortemp
2023-06-16 18:12:38.971 - info: sainlogic.0 (25805) Creating new data point: weather.current.monthlyrain
2023-06-16 18:12:38.972 - info: sainlogic.0 (25805) Creating new data point: weather.maxvalues.absolut.monthlyrain
2023-06-16 18:12:38.972 - info: sainlogic.0 (25805) Creating new data point: weather.maxvalues.daily.monthlyrain
2023-06-16 18:12:38.972 - info: sainlogic.0 (25805) Creating new data point: weather.current.monthlyrain_piezo
2023-06-16 18:12:38.973 - info: sainlogic.0 (25805) Creating new data point: weather.current.outdoorhumidity
2023-06-16 18:12:38.973 - info: sainlogic.0 (25805) Creating new data point: weather.maxvalues.absolut.outdoorhumidity
2023-06-16 18:12:38.973 - info: sainlogic.0 (25805) Creating new data point: weather.maxvalues.daily.outdoorhumidity
2023-06-16 18:12:38.974 - info: sainlogic.0 (25805) Creating new data point: weather.minvalues.absolut.outdoorhumidity
2023-06-16 18:12:38.974 - info: sainlogic.0 (25805) Creating new data point: weather.minvalues.daily.outdoorhumidity
2023-06-16 18:12:38.974 - info: sainlogic.0 (25805) Creating new data point: weather.current.outdoortemp
2023-06-16 18:12:38.975 - info: sainlogic.0 (25805) Creating new data point: weather.maxvalues.absolut.outdoortemp
2023-06-16 18:12:38.975 - info: sainlogic.0 (25805) Creating new data point: weather.maxvalues.daily.outdoortemp
2023-06-16 18:12:38.976 - info: sainlogic.0 (25805) Creating new data point: weather.minvalues.absolut.outdoortemp
2023-06-16 18:12:38.976 - info: sainlogic.0 (25805) Creating new data point: weather.minvalues.daily.outdoortemp
2023-06-16 18:12:38.977 - info: sainlogic.0 (25805) Creating new data point: weather.current.pressureabs
2023-06-16 18:12:38.977 - info: sainlogic.0 (25805) Creating new data point: weather.maxvalues.absolut.pressureabs
2023-06-16 18:12:38.977 - info: sainlogic.0 (25805) Creating new data point: weather.maxvalues.daily.pressureabs
2023-06-16 18:12:38.978 - info: sainlogic.0 (25805) Creating new data point: weather.minvalues.absolut.pressureabs
2023-06-16 18:12:38.978 - info: sainlogic.0 (25805) Creating new data point: weather.minvalues.daily.pressureabs
2023-06-16 18:12:38.978 - info: sainlogic.0 (25805) Creating new data point: weather.current.pressurerel
2023-06-16 18:12:38.979 - info: sainlogic.0 (25805) Creating new data point: weather.maxvalues.absolut.pressurerel
2023-06-16 18:12:38.980 - info: sainlogic.0 (25805) Creating new data point: weather.maxvalues.daily.pressurerel
2023-06-16 18:12:38.980 - info: sainlogic.0 (25805) Creating new data point: weather.minvalues.absolut.pressurerel
2023-06-16 18:12:38.980 - info: sainlogic.0 (25805) Creating new data point: weather.minvalues.daily.pressurerel
2023-06-16 18:12:38.981 - info: sainlogic.0 (25805) Creating new data point: weather.current.rain
2023-06-16 18:12:38.981 - info: sainlogic.0 (25805) Creating new data point: weather.maxvalues.absolut.rain
[... cut ...] 426%22%2C%22rainin%22%3A%220.000%22%2C%22dailyrainin%22%3A%220.000%22%2C%22weeklyrainin%22%3A%220.000%22%2C%22monthlyrainin%22%3A%224.461%22%2C%22solarradiation%22%3A%2279.40%22%2C%22UV%22%3A%220%22%2C%22dateutc%22%3A%222023-06-16+16%3A12%3A48%22%2C%22softwaretype%22%3A%22EasyWeatherV1.6.5%22%2C%22action%22%3A%22updateraw%22%2C%22realtime%22%3A%221%22%2C%22rtfreq%22%3A%225%22%7D
2023-06-16 18:12:50.446 - info: sainlogic.0 (25805) Forward response body: success

<!--EndFragment-->
</body>
</html>

mcm1957, 2023-06-16:
In addition the following log
```
2023-06-16 18:12:39.084 - warn: sainlogic.0 (25805) Object sainlogic.0.info.last_listener_update is invalid: Default value has to be type "string" but received type "number"
2023-06-16 18:12:39.085 - warn: sainlogic.0 (25805) This object will not be created in future versions. Please report this to the developer.
```
raises some alarm. Please check wether teh adapter creates the objects still with upcomingjs-.controlle5. It's available as beta (see forum).

phifogg, 2023-07-21:
Thanks @mcm1957. Sounds like initialization code is using a default - probably empty - string. I'll check into that.

phifogg, 2023-07-23:
Closed with 0.10.5

closed 2023-07-23 by phifogg