phifogg/ioBroker.sainlogic #171: Listener received illegal request (GW1100A_V2.2.5)

x-sven, 2023-07-20:
**Describe the bug**  
Error Message in iobroker log:
Listener received illegal request: /weatherstation/updateweatherstation.php?ID=4711&PASSWORD=home&tempf=-9999&humidity=-9999&dewptf=-9999&windchillf=-9999&winddir=-9999&windspeedmph=-9999&windgustmph=-9999&solarradiation=-9999&UV=-9999&indoortempf=77.00&indoorhumidity=48&baromin=29.607&temp2f=79.52&humidity2=45&soilmoisture=0&dateutc=now&softwaretype=GW1100A_V2.2.5&action=updateraw&realtime=1&rtfreq=5

**Versions:**  
 - Adapter version: v0.10.4
 - Node.js: v16.20.1
 - NPM: v8.19.4
 - Operating system: docker

**Additional context**  
Tried both versions, means with and without "?" in all combinations.

[image]

Any ideas how to debug or fix the issue?

THX Sven

phifogg, 2023-07-21:
Hi Sven,
can you post your config on ioB as well?
Thanks,
Daniel

x-sven, 2023-07-21:
You mean this? I hope...

[image]

phifogg, 2023-07-21:
yep, thanks. this look ok. Can you try from a browser to get this URL:

http://<iobrokerip>:1889/weatherstation/updateweatherstation.php?ID=4711&PASSWORD=home&tempf=-9999&humidity=-9999&dewptf=-9999&windchillf=-9999&winddir=-9999&windspeedmph=-9999&windgustmph=-9999&solarradiation=-9999&UV=-9999&indoortempf=77.00&indoorhumidity=48&baromin=29.607&temp2f=79.52&humidity2=45&soilmoisture=0&dateutc=now&softwaretype=GW1100A_V2.2.5&action=updateraw&realtime=1&rtfreq=5

This is basically what your weatherstation should be doing. This should update the data in ioB.

If that works, it is a matter of typo's, questions marks or so on the weatherstation-config.

x-sven, 2023-07-21:
ok, did that with the IP of the iobroker server. Same error in the io log :-/

However, there is a port forwarding proxy in-front of the iobroker. Means, the IP address of the hostname where the weather station is sending to differs from the real IP address the iobroker container sees for itself - just as remark if that matters.

x-sven, 2023-07-21:
...BUT it's only for the wunderground protocol not working, with ecowitt works.

phifogg, 2023-07-21:
The forwarder shouldn't be an issue as long as the whole URL stays intact (all parameters).
Ecowitt works? The difference is that ecowitt transfers the data in the body, not the URL. So there is something wrong with the URL. Which is a bit weird, as the log message you quoted seems fine.

phifogg, 2023-07-23:
Try in the config of the adapter to put only: /weatherstation/updateweatherstation.php

That's how it works for mine.

phifogg, 2023-07-23:
I think I found it... your values are a but bonkers. A lot of them are outside what is expected. The one causing all to fail is winddir which is supposed to be between 0 and 360 but the actual values is -9999 causing an uncaught exception.

x-sven, 2023-07-24:
Thanks for looking at the issue. It's what the (actual) firmware delivers by default if no wind sensor is connected. Hope the report was useful and will close it by this comment. Best regards Sven

closed 2023-07-24 by x-sven